dariuszseweryn/RxAndroidBle #289: Connection error: Can't create handler inside thread that has not called Looper.prepare()

RobLewis, 2017-10-01:
Something I haven't seen before, in practically the simplest possible code
### Summary
Trying to connect to device gives Connection error: "Can't create handler inside thread that has not called Looper.prepare()"
### Library version
`1.4.1`

#### Preconditions
Scan for and locate a BLE device
#### Steps to reproduce actual result
See code
#### Minimum code snippet reproducing the issue
```
void processScanResult( ScanResult scanResult ) {
        Log.d( "MAIN", "Entering processScanResult" );
        bpScanSub.unsubscribe();  // stop scanning
        bpDevice = scanResult.getBleDevice();
        bpRssi = scanResult.getRssi();
        bpName = bpDevice.getName();
        bpMacAddress = bpDevice.getMacAddress();
        Toast.makeText( this, "Found device " + bpName + " with MAC address " + bpMacAddress, Toast.LENGTH_LONG ).show();
        bpConnSub = bpDevice.establishConnection( false )
                //.subscribeOn( Schedulers.io( ) )
                .subscribe(
                        this::onBpConnected,
                        this::onBpConnError
                );
    }
    
    void handleScanError( Throwable scanError ) {
        Log.d( "MAIN", "Scan error: " + scanError.getMessage() );
    }
    
    void onBpConnected( RxBleConnection bpConnection ) {
        Toast.makeText( this, "Connected to device " + bpName, Toast.LENGTH_LONG ).show();
    }
    
    void onBpConnError( Throwable connError ) {
        Log.d( "MAIN", "Connection error: " + connError.getMessage() );
    }
```
#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`
```
09-30 16:59:24.494 32038-32055/name.jacklewis.bloodpressureapp D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=F4:5E:AB:10:DF:5C, mScanRecord=ScanRecord [mAdvertiseFlags=5, mServiceUuids=[00001810-0000-1000-8000-00805f9b34fb], mManufacturerSpecificData={}, mServiceData={}, mTxPowerLevel=0, mDeviceName=A&D_UA-651BLE_10DF5C], mRssi=-62, mTimestampNanos=84940822805792}
09-30 16:59:24.497 32038-32038/name.jacklewis.bloodpressureapp D/BluetoothDevice: mAddress: F4:5E:AB:10:DF:5C
09-30 16:59:24.508 32038-32038/name.jacklewis.bloodpressureapp D/MAIN: Entering processScanResult
09-30 16:59:24.511 32038-32038/name.jacklewis.bloodpressureapp D/BluetoothDevice: mAddress: F4:5E:AB:10:DF:5C
09-30 16:59:24.511 32038-32066/name.jacklewis.bloodpressureapp D/BluetoothAdapter: 651889534: getState(). Returning 12
09-30 16:59:24.513 32038-32066/name.jacklewis.bloodpressureapp D/BluetoothAdapter: 651889534: getState(). Returning 12
09-30 16:59:24.513 32038-32066/name.jacklewis.bloodpressureapp I/BluetoothLeScanner: startRegisteration: mLeScanClients={com.polidea.rxandroidble.internal.operations.ScanOperationApi21$1@2a3878df=android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper@3e98618a} ,callback=com.polidea.rxandroidble.internal.operations.ScanOperationApi21$1@2a3878df
09-30 16:59:24.540 32038-32038/name.jacklewis.bloodpressureapp D/BluetoothAdapter: isEnabled
09-30 16:59:24.570 32038-32066/name.jacklewis.bloodpressureapp D/BluetoothDevice: mAddress: F4:5E:AB:10:DF:5C
09-30 16:59:24.570 32038-32066/name.jacklewis.bloodpressureapp D/BluetoothGatt: connect() - device: F4:5E:AB:10:DF:5C, auto: false
09-30 16:59:24.570 32038-32066/name.jacklewis.bloodpressureapp D/BluetoothGatt: registerApp()
09-30 16:59:24.570 32038-32066/name.jacklewis.bloodpressureapp D/BluetoothGatt: registerApp() - UUID=4881cea5-d1db-4d04-9061-76ce68914cb5
09-30 16:59:24.572 32038-32055/name.jacklewis.bloodpressureapp D/BluetoothGatt: onClientRegistered() - status=0 clientIf=1
09-30 16:59:24.572 32038-32055/name.jacklewis.bloodpressureapp D/BluetoothDevice: mAddress: F4:5E:AB:10:DF:5C
09-30 16:59:24.587 32038-32056/name.jacklewis.bloodpressureapp D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=1 device=F4:5E:AB:10:DF:5C
09-30 16:59:24.587 32038-32056/name.jacklewis.bloodpressureapp D/BluetoothDevice: mAddress: F4:5E:AB:10:DF:5C
09-30 16:59:24.598 32038-32146/name.jacklewis.bloodpressureapp D/MAIN: Connection error: Can't create handler inside thread that has not called Looper.prepare()
09-30 16:59:24.599 32038-32146/name.jacklewis.bloodpressureapp D/BluetoothDevice: mAddress: F4:5E:AB:10:DF:5C
09-30 16:59:24.599 32038-32066/name.jacklewis.bloodpressureapp D/BluetoothManager: getConnectionState()
09-30 16:59:24.600 32038-32066/name.jacklewis.bloodpressureapp D/BluetoothManager: getConnectedDevices
09-30 16:59:24.605 32038-32066/name.jacklewis.bloodpressureapp D/BluetoothDevice: mAddress: F4:5E:AB:10:DF:5C
09-30 16:59:24.607 32038-32066/name.jacklewis.bloodpressureapp D/BluetoothDevice: mAddress: F4:5E:AB:10:DF:5C
09-30 16:59:24.607 32038-32066/name.jacklewis.bloodpressureapp D/BluetoothGatt: cancelOpen() - device: F4:5E:AB:10:DF:5C
09-30 16:59:24.607 32038-32066/name.jacklewis.bloodpressureapp D/BluetoothDevice: mAddress: F4:5E:AB:10:DF:5C
09-30 16:59:24.609 32038-32055/name.jacklewis.bloodpressureapp D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=1 device=F4:5E:AB:10:DF:5C
09-30 16:59:24.609 32038-32055/name.jacklewis.bloodpressureapp D/BluetoothDevice: mAddress: F4:5E:AB:10:DF:5C
09-30 16:59:24.609 32038-32055/name.jacklewis.bloodpressureapp D/BluetoothDevice: mAddress: F4:5E:AB:10:DF:5C
09-30 16:59:24.611 32038-32066/name.jacklewis.bloodpressureapp D/BluetoothGatt: close()
09-30 16:59:24.611 32038-32066/name.jacklewis.bloodpressureapp D/BluetoothGatt: unregisterApp() - mClientIf=1
09-30 16:59:24.658 32038-32067/name.jacklewis.bloodpressureapp D/GraphicBuffer: register, handle(0xa2b24e80) (w:690 h:59 s:704 f:0x1 u:0x000f02)
09-30 16:59:24.658 32038-32067/name.jacklewis.bloodpressureapp I/MaliEGL: [Mali]window_type=1, is_framebuffer=0, errnum = 0
09-30 16:59:24.658 32038-32067/name.jacklewis.bloodpressureapp I/MaliEGL: [Mali]surface->num_buffers=4, surface->num_frames=3, win_min_undequeued=1
09-30 16:59:24.658 32038-32067/name.jacklewis.bloodpressureapp I/MaliEGL: [Mali]max_allowed_dequeued_buffers=3
09-30 16:59:24.659 32038-32067/name.jacklewis.bloodpressureapp D/GraphicBuffer: unregister, handle(0xa2b24e80) (w:690 h:59 s:704 f:0x1 u:0x000f02)
09-30 16:59:24.661 32038-32067/name.jacklewis.bloodpressureapp D/GraphicBuffer: register, handle(0xa2b24e80) (w:690 h:59 s:704 f:0x1 u:0x000f02)
09-30 16:59:24.683 32038-32067/name.jacklewis.bloodpressureapp V/RenderScript: 0xb094da00 Launching thread(s), CPUs 4
09-30 16:59:28.026 32038-32067/name.jacklewis.bloodpressureapp D/OpenGLRenderer: Flushing caches (mode 0)
09-30 16:59:28.027 32038-32067/name.jacklewis.bloodpressureapp D/OpenGLRenderer: Flushing caches (mode 0)
09-30 16:59:28.028 32038-32067/name.jacklewis.bloodpressureapp D/GraphicBuffer: unregister, handle(0xa2b24e80) (w:690 h:59 s:704 f:0x1 u:0x000f02)
09-30 16:59:42.624 32038-32038/name.jacklewis.bloodpressureapp D/FeatureProxyBase: FeatureProxyBase class constructor
09-30 16:59:42.624 32038-32038/name.jacklewis.bloodpressureapp D/FeatureProxyBase: getService(), serviceName = multiwindow_service_v1
```
#### Actual result
"Found device" Toast is shown, then `onBpConnError` is called with message "Can't create handler inside thread that has not called Looper.prepare()"
Note that the commented-out `subscribeOn` did not have any effect. 
#### Expected result
`onBpConnected` called with device connection

RobLewis, 2017-10-01:
Think I found the problem: rookie mistake. 
I needed an `.observeOn( AndroidSchedulers.mainThread() )` to make it possible to display the Toast in `onBpConnected()`. 
(However that doesn't explain why the _first_ (scanning) `Observable` chain was able to display a Toast and didn't have the `.observeOn()` operator.)

dariuszseweryn, 2017-10-01:
Hello @RobLewis 
The library does not use `Handler`s in any way. It is emitting values on arbitarary `Thread` and it is not guaranteed to have a `Looper` (it does not have to be precise when it comes to using a `RxBleConnection`).
Scan related emissions (unlike connection) are not routed to any specific `Scheduler` - they use the default Android callback thread which apparently has a `Looper`. It is an implementation detail an may be changed in the future versions of Android so it should not be relied upon.
`Toast` class rely on being called from a `Thread` that has a `Looper`. To be sure that is the case you need to put `.observeOn(AndroidSchedulers.mainThread()` or any other `Thread` that has a `Looper` before you are using an instance (object) of the `Toast` class.

RobLewis, 2017-10-01:
Thanks for the explanation. It sounds like `Toast` is a bit different from other UI elements, which (we are told) can only be operated on in the main thread. Offhand, are there any use cases where you might want to display a `Toast` from a different thread (with a `Looper`)?

dariuszseweryn, 2017-10-01:
I am not sure how `Toast` works internally- I rarely use it. Best practice would be to use it on the main thread anyway.
> Offhand, are there any use cases where you might want to display a `Toast` from a different thread (with a `Looper`)?

I don't think so.

closed 2017-10-01 by dariuszseweryn