dariuszseweryn/RxAndroidBle #290: Library Source does not match the bytecode for class RxBleConnection

RobLewis, 2017-10-01:
### Summary
Getting the message "Library Source does not match the bytecode for class RxBleConnection" when I command "Go To Declaration" on RxBleConnection in my project. 
### Library version
`1.4.1`

#### Preconditions
I have this in build.gradle: 
compile "com.polidea.rxandroidble:rxandroidble:1.4.1"
#### Steps to reproduce actual result
Use the library in a project
#### Minimum code snippet reproducing the issue
unknown
#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`
n/a (but the Application seems to work OK)
#### Actual result
Getting the message from Android Studio that "Library Source does not match the bytecode for class RxBleConnection" when I command "Go To Declaration" on RxBleConnection in my project.
(but the Application seems to work OK)
#### Expected result
Do not see this message.

dariuszseweryn, 2017-10-01:
Have you tried [suggested stackoverflow solutions](https://stackoverflow.com/questions/39990752/source-code-does-not-match-the-bytecode-when-debugging-on-a-device)?

RobLewis, 2017-10-01:
Yes I looked there but everything I found seemed to involve maven. I will look some more.

dariuszseweryn, 2017-10-01:
How about clean + rebuild?

RobLewis, 2017-10-01:
clean: no
rebuild: no
`rm -rf $HOME/.gradle/caches/`  and rebuild: this gets rid of the message, but now when I command "Go To Declaration" it just decompiles `RxBleConnection.class`. There is an option to "Choose Sources" but I don't know where to find those, if they even exist on my computer. 
Re-launch Android Studio: no change. 
Why is this happening? I've never seen anything like this before.

dariuszseweryn, 2017-10-02:
It looks like a bug in your local build toolchain. The sources are a part of the distribution package of the library.
You can use the `Invalidate caches and restart` option from `Android Studio` menu.
Either way this is not related to the library itself.

Best Regards

dariuszseweryn, 2017-10-03:
Since there are no new posts here I assume that the problem is solved. Feel free to reopen if not.

closed 2017-10-03 by dariuszseweryn

SultonN, 2020-08-18:
Method 'Clicks(com.example.android.miwok.View)' is never used error . can you help me? please