dariuszseweryn/RxAndroidBle #299: Interrupted awaitRelease

mbmc, 2017-10-26:
### Summary

this code
```
rxBleClient.getBleDevice(address)
    .establishConnection(false)
    .flatMap(RxBleConnection::discoverServices)
    .take(1)
    .subscribe(discoveryResult -> {
        // do something
    }, throwable -> Log.d(...)));
```

is causing a
```
RxBle#ConnectionOperationQueue: Interrupted awaitRelease()
     java.lang.InterruptedException
    at java.lang.Object.wait(Native Method)
    at com.polidea.rxandroidble.internal.serialization.QueueSemaphore.awaitRelease(QueueSemaphore.java:15)
    at com.polidea.rxandroidble.internal.serialization.ConnectionOperationQueueImpl$1.run(ConnectionOperationQueueImpl.java:64)
    at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:457)
    at java.util.concurrent.FutureTask.run(FutureTask.java:266)
    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
    at java.lang.Thread.run(Thread.java:764)
```

### Library version

`1.4.2`

I want to be able to close the connection as soon as a specific service is discovered.

dariuszseweryn, 2017-10-26:
Yup, this is just a `verbose` log in the console. It should not cause the library from a normal operation. Does it?

dariuszseweryn, 2017-10-31:
@mbmc Ping?

mbmc, 2017-11-01:
All good, thanks!

closed 2017-11-01 by mbmc