dariuszseweryn/RxAndroidBle #304: Introduce more verbose logging

uKL, 2017-11-02:
Currently, the library mostly logs the state of the operations. This is helpful for debugging the library itself. Additionally, I think that we could have logs with:

- [ ] Scan results
- [ ] Applied scan filters
- [ ] Service discovery result
- [ ] Characteristics UUIDs
- [ ] Hex dump of the data written & read
- [ ] MTU change request details
- [ ] Notification occurrences 
- [ ] Details on checking the adapter, permissions, etc.

RobLewis, 2017-11-02:
Bravo! Fantastic help for debugging. Would these be output to separate files, and/or logcat? I assume there would be an API for choosing options—maybe just a bitfield? 

Maybe include connection state changes, if not already covered.

uKL, 2018-03-16:
Related to #367 on the API level.

dariuszseweryn, 2019-03-22:
Released in `1.9.0`

closed 2019-03-22 by dariuszseweryn