dariuszseweryn/RxAndroidBle #308: NullPointerException throw when unsubscribe is called

hoanglm4, 2017-11-06:
### Summary
Sometimes, when I call unsubscribe, NullPointerException is throw

This is my code:
```
public class MainActivity extends RxAppCompatActivity {
    private static final String TAG = "MainActivity";
    public static final UUID VUZ_CHAR_UUID = UUID.fromString("25A80002-6478-11E6-BDF4-AD3BCC77759F");

    private Observable<RxBleConnection> connectionObservable;
    private PublishSubject<Void> disconnectTriggerSubject = PublishSubject.create();
    private RxBleDevice bleDevice;
    private Subscription mConnectionSubscription;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);
        ButterKnife.bind(this);
        String macAddress = "D0:29:DD:6F:32:DD";
        bleDevice = SampleApplication.getRxBleClient(this).getBleDevice(macAddress);
        connectionObservable = prepareConnectionObservable();
        //noinspection ConstantConditions
        getSupportActionBar().setSubtitle(getString(R.string.mac_address, macAddress));
    }

    private Observable<RxBleConnection> prepareConnectionObservable() {
        return bleDevice
                .establishConnection(false)
                .takeUntil(disconnectTriggerSubject)
                .compose(new ConnectionSharingAdapter())
                .doOnUnsubscribe(() -> Log.d(TAG, "prepareConnectionObservable>> doOnUnsubscribe"));
    }

    private boolean isConnected() {
        return bleDevice.getConnectionState() == RxBleConnection.RxBleConnectionState.CONNECTED;
    }

    @OnClick(R.id.connect)
    public void onConnectToggleClick() {
        if (mConnectionSubscription != null && !mConnectionSubscription.isUnsubscribed()) {
            mConnectionSubscription.unsubscribe();
        }
        if (isConnected()) {
            triggerDisconnect();
        }
        mConnectionSubscription = connectionObservable
                .flatMap(RxBleConnection::discoverServices)
                .flatMap(rxBleDeviceServices -> rxBleDeviceServices.getCharacteristic(VUZ_CHAR_UUID))
                .observeOn(AndroidSchedulers.mainThread())
                .subscribe(
                        characteristic -> {
                            Log.i(getClass().getSimpleName(), "Hey, connection has been established!");
                        },
                        this::onConnectionFailure,
                        this::onConnectionFinished
                );
    }

    @OnClick(R.id.unscription)
    public void disconnectClick() {
        if (mConnectionSubscription != null) {
            mConnectionSubscription.unsubscribe();
        }
        triggerDisconnect();
    }

    private void triggerDisconnect() {
        disconnectTriggerSubject.onNext(null);
    }

    private void onConnectionFailure(Throwable throwable) {
        Log.d(TAG, "onConnectionFailure>> Connection error: " + throwable);
    }

    private void onConnectionFinished() {
        Log.d(TAG, "onConnectionFinished>> Connection finish");
    }

}
```

Log crash:
```
Caused by java.lang.NullPointerException: Attempt to invoke interface method 'void rx.Subscription.unsubscribe()' on a null object reference
       at com.polidea.rxandroidble.internal.serialization.ConnectionOperationQueueImpl.onConnectionUnsubscribed(ConnectionOperationQueueImpl.java:147)
       at com.polidea.rxandroidble.internal.connection.ConnectorImpl$1$2.call(ConnectorImpl.java:68)
       at rx.subscriptions.BooleanSubscription.unsubscribe(BooleanSubscription.java:71)
       at rx.internal.util.SubscriptionList.unsubscribeFromAll(SubscriptionList.java:136)
       at rx.internal.util.SubscriptionList.unsubscribe(SubscriptionList.java:125)
       at rx.Subscriber.unsubscribe(Subscriber.java:98)
       at rx.internal.util.SubscriptionList.unsubscribeFromAll(SubscriptionList.java:136)
       at rx.internal.util.SubscriptionList.unsubscribe(SubscriptionList.java:125)
       at rx.Subscriber.unsubscribe(Subscriber.java:98)
       at rx.internal.util.SubscriptionList.unsubscribeFromAll(SubscriptionList.java:136)
       at rx.internal.util.SubscriptionList.unsubscribe(SubscriptionList.java:125)
       at rx.Subscriber.unsubscribe(Subscriber.java:98)
       at rx.subscriptions.CompositeSubscription.unsubscribeFromAll(CompositeSubscription.java:186)
       at rx.subscriptions.CompositeSubscription.unsubscribe(CompositeSubscription.java:175)
       at rx.internal.operators.OnSubscribeRefCount$3.call(OnSubscribeRefCount.java:164)
       at rx.subscriptions.BooleanSubscription.unsubscribe(BooleanSubscription.java:71)
       at rx.internal.util.SubscriptionList.unsubscribeFromAll(SubscriptionList.java:136)
       at rx.internal.util.SubscriptionList.unsubscribe(SubscriptionList.java:125)
       at rx.Subscriber.unsubscribe(Subscriber.java:98)
       at rx.internal.util.SubscriptionList.unsubscribeFromAll(SubscriptionList.java:136)
       at rx.internal.util.SubscriptionList.unsubscribe(SubscriptionList.java:125)
       at rx.Subscriber.unsubscribe(Subscriber.java:98)
       at rx.internal.util.SubscriptionList.unsubscribeFromAll(SubscriptionList.java:136)
       at rx.internal.util.SubscriptionList.unsubscribe(SubscriptionList.java:125)
       at rx.Subscriber.unsubscribe(Subscriber.java:98)
       at rx.internal.util.SubscriptionList.unsubscribeFromAll(SubscriptionList.java:136)
       at rx.internal.util.SubscriptionList.unsubscribe(SubscriptionList.java:125)
       at rx.Subscriber.unsubscribe(Subscriber.java:98)
       at rx.internal.util.SubscriptionList.unsubscribeFromAll(SubscriptionList.java:136)
       at rx.internal.util.SubscriptionList.unsubscribe(SubscriptionList.java:125)
       at rx.Subscriber.unsubscribe(Subscriber.java:98)
       at rx.internal.subscriptions.SequentialSubscription.unsubscribe(SequentialSubscription.java:180)
       at rx.subscriptions.SerialSubscription.unsubscribe(SerialSubscription.java:36)
       at rx.internal.util.SubscriptionList.unsubscribeFromAll(SubscriptionList.java:136)
       at rx.internal.util.SubscriptionList.unsubscribe(SubscriptionList.java:125)
       at rx.Subscriber.unsubscribe(Subscriber.java:98)
       at rx.internal.util.SubscriptionList.unsubscribeFromAll(SubscriptionList.java:136)
       at rx.internal.util.SubscriptionList.unsubscribe(SubscriptionList.java:125)
       at rx.Subscriber.unsubscribe(Subscriber.java:98)
       at rx.internal.util.SubscriptionList.unsubscribeFromAll(SubscriptionList.java:136)
       at rx.internal.util.SubscriptionList.unsubscribe(SubscriptionList.java:125)
       at rx.Subscriber.unsubscribe(Subscriber.java:98)
       at rx.internal.util.SubscriptionList.unsubscribeFromAll(SubscriptionList.java:136)
       at rx.internal.util.SubscriptionList.unsubscribe(SubscriptionList.java:125)
       at rx.Subscriber.unsubscribe(Subscriber.java:98)
```
### Library version
`1.4.2`

#### Actual result
NullPointerException throw

#### Expected result
Check NULL in https://github.com/Polidea/RxAndroidBle/blob/master/rxandroidble/src/main/java/com/polidea/rxandroidble/internal/serialization/ConnectionOperationQueueImpl.java#L147 or find cause of this bug.

Thanks

dariuszseweryn, 2017-11-13:
Hello @hoanglm4 
I have just pushed a fix for this issue. It should be available in `1.4.3-SNAPSHOT` shortly. Could you give it a try?

closed 2017-11-13 by dariuszseweryn

hoanglm4, 2017-11-13:
Thanks @dariuszseweryn 
I will follow it in fabric crashlytics.

commit 99a18f3a (2017-11-13): Call ConnectionSubscriptionWatcher on Bluetooth Callbacks Scheduler (fixes #308) (#315)

Call ConnectionSubscriptionWatcher on Bluetooth Callbacks Scheduler (fixes #308)
Without specifcation of `.subscribeOn()` and `.unsubscribeOn()` there was a possibility of a race condition when calling `.doOnSubscribe()` and `.doOnUnsubscribe()` which lead to calling `ConnectionOperationQueueImpl.onConnectionUnsubscribed()` before the `.onConnectionSubscribed()` has returned leading to a `NullPointerException`. By default `.doOnSubscribe()` is called on a thread that calls `.subscribe()` and `.doOnUnsubcri [... cut]