dariuszseweryn/RxAndroidBle #310: Smarter setupNotification / setupIndication

simonratner, 2017-11-07:
### Summary

Add method to automatically pick correct {notification/indication} descriptor based on characteristic properties, or make the existing `setupNotification` smarter.

Problem: client must know the properties of the characteristic in question to know which method to call, otherwise the call fails with `BleIllegalOperationException`. However, with automatic discovery, the client can't always know this. 

The *right thing to do* is obvious in most cases---if the characteristic has `PROPERTY_NOTIFY`, subscribe to notifications, if it has `PROPERTY_INDICATE`, subscribe to indications. If it has both property flags, picking notifications is probably a reasonable default (that's what iOS Core Bluetooth does: https://developer.apple.com/documentation/corebluetooth/cbperipheral/1518949-setnotifyvalue).

I notice #239 has more `NotificationSetup` options, so maybe it could include an `auto` option.

### Library version
`1.4.2`

#### Preconditions

Have a peripheral with a characteristic with either `PROPERTY_NOTIFY` or `PROPERTY_INDICATE` set.

#### Steps to reproduce actual result
<br />1. Connect to peripheral.
<br />2. Call `setupNotification` (when characteristic is `PROPERTY_INDICATE`) or call `setupIndication` (when characteristic is `PROPERTY_NOTIFY`).
<br />

#### Minimum code snippet reproducing the issue

```java
rxbleConnection.setupNotification(uuid).subscribe(onNotification, onNotificationFailure);
```

#### Actual result

`BleIllegalOperationException` is thrown.

#### Expected result

Subscribe to the method of notifications / indications that the characteristic supports.

dariuszseweryn, 2017-11-13:
Hello @simonratner 
This is a good idea! And at this moment it should be possible to make appropriate changes just to the `NotificationSetup` class on `feature/issue_213_new_connection_api` branch and make a PR to #239 
The same approach could be used for the `setCCCDescriptor`. Auto would set the descriptor by default if it exists and would not if it does not.

RobLewis, 2017-11-16:
Great idea! (I just spent too much time tracking down a bug in some inherited (non-Rx) Android code, which was due to trying to enable indications for characteristics that only supported notifications.) As for what the default should be when a characteristic supports both, what are the pros and cons of each?

simonratner, 2017-11-17:
When characteristic supports both, pros for defaulting to notifications:
* [_This is a big one_] That's what iOS does, in fact, with Core Bluetooth there is no way to subscribe to indications if a characteristic supports both. As someone who makes the hw side of this equation, consistency is always welcome.
* Notifications are more common, more people know and understand them.
* Notifications are more lightweight on the traffic, as they don't generate an ack.

Cons:
* Defaulting to a less reliable communication method when a device is clearly advertising that it supports a more reliable mode.

RobLewis, 2017-11-17:
@simonratner Good logic, however in my case (low-traffic medical data collection) I'd opt for the most reliable method. So the preference should be user-settable. 

As for the API, how about something like this: 
`Observable<Observable<byte[]>> setupNotificationOrIndication(@NonNull UUID characteristicUuid, @NonNull NotificationSetupMode setupMode, boolean preferIndication);`

The `boolean` would default to false if not specified. 
Might want to review the types of Exceptions generated, and perhaps `setupMode` values?

simonratner, 2017-11-17:
Whether you want to setup indications or notifications if specified in the `setupMode` in the new connection API. If you prefer indications, and your device supports them, just don't use the auto setting?

RobLewis, 2018-02-22:
Am I missing something? The `NotificationSetupMode` enum has options `DEFAULT` and `COMPAT`. The docs don't say what these have to do with choosing indications over notifications.  And the only parameters I see in the `ConnectionSetup` class are `autoConnect`, `suppressOperationCheck`, and `operationTimeout`.