dariuszseweryn/RxAndroidBle #312: RxJava 1.3.1 bug preventing RxAndroidBle from building with ProGuard

mechatronik, 2017-11-09:
### Summary
There is [a bug](https://github.com/ReactiveX/RxJava/issues/5601) in RxJava 1.3.1 (which this lib is dependent on) fixed in [RxJava 1.3.2](https://github.com/ReactiveX/RxJava/pull/5602), which prevented us from building our project with ProGuard enabled. RxJavaProGuardRules author  [recommends](https://github.com/artem-zinnatullin/RxJavaProGuardRules/issues/62) skipping this version of RxJava. When we exclude rxjava bundled with rxandroidble from gradle and replace it with newer version, everything works fine. 

### Library version
1.4.0 and above

#### Actual result
`Warning:rx.internal.schedulers.NewThreadWorker: can't find referenced method 'java.util.concurrent.ConcurrentHashMap$KeySetView keySet()' in library class java.util.concurrent.ConcurrentHashMap`
`Warning:there were 1 unresolved references to library class members.`
`Error:Execution failed for task ':app:transformClassesAndResourcesWithProguardForDebug'`

#### Expected result
Build succeeds

uKL, 2017-11-09:
Good spot! Could you make a pull request bumping rxjava to 1.3.2?

mechatronik, 2017-11-09:
Probably later today

commit d7156a4e (2017-11-10): Update RxJava version to fix #312 (#314)
closed 2017-11-10 by uKL