dariuszseweryn/RxAndroidBle #313: unsubscribe method can not disconnect the connection

Mike-bel, 2017-11-09:
Here is my connect operation codes:

```java
Subscription subscription = Observable.just(mDeviceInfo.getImei())
        .observeOn(Schedulers.computation())
        .map(input1 -> {
          StringBuilder sb = new StringBuilder();
          sb.append(input1.substring(4, 6)).append(":");
          sb.append(input1.substring(6, 8)).append(":");
          sb.append(input1.substring(8, 10)).append(":");
          sb.append(input1.substring(10, 12)).append(":");
          sb.append(input1.substring(12, 14)).append(":");
          sb.append(input1.substring(14, 16));
          return sb.toString();
        })
        .observeOn(Schedulers.io())
        .flatMap(macAddress -> {
          LogUtils.d("connect to mac address: " + macAddress);
          mRxBleDevice = mRxBleClient.getBleDevice(macAddress);
          return mRxBleDevice.establishConnection(false);
        })
        .observeOn(AndroidSchedulers.mainThread())
        .subscribe(
            rxBleConnection -> {
              LogUtils.d("connect result: Connected");
              mRxBleConnection = rxBleConnection;
              handleUnlockChanged(BleUnLockState.TokenRequested, BleUnLockState.Connected);
              handleReconnectSuccess();
              setup();
            },
            throwable -> {
              LogUtils.e("connect result: Failed");
              LogUtils.e(Log.getStackTraceString(throwable));
              handleUnlockFailed(BleUnLockState.Connected, UnlockException.CONNECT_FAILED);
              handleReconnectFailed();
            });
```

Next time before I want to reconnect the device again, I use this code to disconnect:

```java
subscription.unsubscribe();
```

But the reconnection failed:

```java
11-09 18:19:02.425 15244-15244/ble.ubike.com.ble_demo E/BleLib: BleAlreadyConnectedException{macAddress=60:64:05:A9:44:EB}
                                                                    at com.polidea.rxandroidble.internal.RxBleDeviceImpl$1.call(RxBleDeviceImpl.java:80)
                                                                    at com.polidea.rxandroidble.internal.RxBleDeviceImpl$1.call(RxBleDeviceImpl.java:67)
                                                                    at rx.internal.operators.OnSubscribeDefer.call(OnSubscribeDefer.java:46)
                                                                    at rx.internal.operators.OnSubscribeDefer.call(OnSubscribeDefer.java:35)
                                                                    at rx.Observable.unsafeSubscribe(Observable.java:10256)
                                                                    at rx.internal.operators.OperatorMerge$MergeSubscriber.onNext(OperatorMerge.java:248)
                                                                    at rx.internal.operators.OperatorMerge$MergeSubscriber.onNext(OperatorMerge.java:148)
                                                                    at rx.internal.operators.OnSubscribeMap$MapSubscriber.onNext(OnSubscribeMap.java:77)
                                                                    at rx.internal.operators.OperatorObserveOn$ObserveOnSubscriber.call(OperatorObserveOn.java:224)
                                                                    at rx.internal.schedulers.CachedThreadScheduler$EventLoopWorker$1.call(CachedThreadScheduler.java:230)
                                                                    at rx.internal.schedulers.ScheduledAction.run(ScheduledAction.java:55)
                                                                    at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:422)
                                                                    at java.util.concurrent.FutureTask.run(FutureTask.java:237)
                                                                    at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:152)
                                                                    at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:265)
                                                                    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
                                                                    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
                                                                    at java.lang.Thread.run(Thread.java:841)
```

uKL, 2017-11-09:
Hey! This looks like a general usage question. Please ask it on StackOverflow with #rxandroidble tag. Other people may get some help as well :)

closed 2017-11-09 by uKL

Mike-bel, 2017-11-09:
But I can not see any usage problems.