dariuszseweryn/RxAndroidBle #318: Can't connect again to the BLE device after bluetooth off / on

Esperanz0, 2017-11-16:
### Library version
`1.4.2`

#### Steps to reproduce actual result
<br />1. Connect with BLE device.
<br />2. Turn off bluetooth on the phone
<br />3. Turn on bluetooth
<br />4. Try to connect again with the same  BLE device
#### Minimum code snippet reproducing the issue

```
final String bleMac = bleDevice.getMacAddress();
        if (isConnected(bleDevice)) {
            triggerDisconnect(bleDevice);
        } else {
            final Subscription subscription = bleDevice.establishConnection(false)
                    .takeUntil(disconnectTriggerSubject)
                    .flatMap(new Func1<RxBleConnection, Observable<RxBleDeviceServices>>() {
                        @Override
                        public Observable<RxBleDeviceServices> call(RxBleConnection rxBleConnection) {
                            Log.d(TAG, "BLE connection to " + bleMac + " done!");

                            // Add this connection to a map for later use
                            connectionMap.put(bleMac, rxBleConnection);

                            return rxBleConnection.discoverServices();
                        }
                    })
                    .map(new Func1<RxBleDeviceServices, ConnectedDevice>() {
                        @Override
                        public ConnectedDevice call(RxBleDeviceServices rxBleDeviceServices) {
                            Log.d(TAG, "Got service BLE from " + bleMac);

                            BluetoothGattService ngService = null;
                            for (BluetoothGattService service : rxBleDeviceServices.getBluetoothGattServices()) {
                                Log.e(TAG, "ngServices UUID: " + service.getUuid());
                                Log.e(TAG, "ngServices Instance Id: " + service.getInstanceId());
                                if (NG_SERVICE_UUID.equals(service.getUuid())) {
                                    ngService = service;
                                    break;
                                }
                            }

                            // If there are no proper service, just throw an error
                            if (ngService == null) {
                                // try connect again in case of error
                                new Handler(Looper.getMainLooper()).postDelayed(new Runnable() {
                                    @Override
                                    public void run() {
                                        Log.e(TAG, "Try again connect to " + bleMac);
                                        MdsRx.Instance.connect(bleDevice);
                                    }
                                }, CONNCTION_DELAY_MILLIS);
                                return null;
                            }

                            BluetoothGattCharacteristic notifyCharacteristic = ngService.getCharacteristic(NG_NOTIFY_CHARACTERISTIC_UUID);
                            BluetoothGattCharacteristic writeCharacteristic = ngService.getCharacteristic(NG_WRITE_CHARACTERISTIC_UUID);

                            // Setup notifications for incoming data
                            RxBleConnection connection = connectionMap.get(bleMac);
                            Subscription notifySubscription = connection.setupNotification(notifyCharacteristic)
                                    .doOnNext(new Action1<Observable<byte[]>>() {
                                        @Override
                                        public void call(Observable<byte[]> observable) {
                                            Log.d(TAG, "Notifications set, calling bypassConnect()");

                                            // Bypass connect in WB to make it aware of this new device
                                            String wbAddress = addressMap.getOrCreateWbAddress(bleMac);
                                            bleWrapper.bypassConnect(wbAddress);
[... 524 lines cut ...]
 D/BleManager: connecting to D5:1D:A7:76:6F:AE
 D/BleManager: isConnected: RxBleConnectionState{DISCONNECTED}
 D/BleManager: Connection try: 0
 D/RxBle#ClientOperationQueue:   QUEUED ConnectOperation(92429508)
 D/RxBle#ClientOperationQueue:  STARTED ConnectOperation(92429508)
 D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(92429508)
```

#### Actual result
```
D/BluetoothGatt: connect() - device: D5:1D:A7:76:6F:AE, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=75697751-02fe-4dc1-8c65-2e7d90c4ead0
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=5 device=D5:1D:A7:76:6F:AE
D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=133
E/BleManager: new Action1<Throwable> throwable: 
 BleDisconnectedException{bluetoothDeviceAddress='D5:1D:A7:76:6F:AE'}
     at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:76)
     at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:181)
     at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
     at android.os.Binder.execTransact(Binder.java:453)
```
#### Expected result
RxAndroidBle will connect to the device after bluetooth off/on.

dariuszseweryn, 2017-11-21:
Hello @Esperanz0 

I see a `status=133` in your `Actual result`. Why do you think it is a problem with the library?

Esperanz0, 2017-11-22:
@dariuszseweryn Hi.

When I will kill app and run it again its working so I thought it is something with library. 

So if it is no library problem do You have idea how to check where is problem / what failing after bt off/on.

dariuszseweryn, 2017-11-22:
Try scanning the device after adapter off/on

dariuszseweryn, 2017-11-22:
Actually this question should go to the [stackoverflow](https://stackoverflow.com/questions/tagged/rxandroidble) as this is a general usage question or rather an Android BLE stack quirk. Not an issue with the library

dariuszseweryn, 2018-09-04:
I was investigating the issue more deeply recently but unfortunately there is no good solution for the current state of the OS. I will probably create a blog post about the issue