dariuszseweryn/RxAndroidBle #322: BleGattDescriptorException（Caused by:BleGattDescriptorException status=17(0x11)）

Xinima, 2017-11-17:
Hi there,
I am trying to unlock a bluetooth bike lock, I've checked some old issues and can't find a same issue.
I can connect to the device, but when write characteristics to the device, something went wrong, The
exception is happening in this part of code:

  mRxBleConnection.getCharacteristic(READ_DATA_UUID)
        .subscribeOn(AndroidSchedulers.mainThread())
        .observeOn(Schedulers.computation())
        .map(new Func1<BluetoothGattCharacteristic, BluetoothGattDescriptor>() {
          @Override
          public BluetoothGattDescriptor call(
              BluetoothGattCharacteristic bluetoothGattCharacteristic) {
            BluetoothGattDescriptor descriptor =
                bluetoothGattCharacteristic.getDescriptor(DESCRIPTOR_UUID);
            descriptor.setValue(BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE);
            return descriptor;
          }
        })
        .flatMap(new Func1<BluetoothGattDescriptor, Observable<byte[]>>() {
          @Override
          public Observable<byte[]> call(BluetoothGattDescriptor bluetoothGattDescriptor) {
            return mRxBleConnection.writeDescriptor(bluetoothGattDescriptor,
                BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE);
          }
        })
        .observeOn(AndroidSchedulers.mainThread())
        .subscribe(
            new Subscriber<byte[]>() {
              @Override
              public void onCompleted() {

              }

              @Override
              public void onError(Throwable e) {
                LogUtils.e("setup getCharacteristic error");
                LogUtils.e(Log.getStackTraceString(e));
                handleUnlockFailed(BleUnLockState.setupReady, UnlockException.CONNECT_FAILED);
              }

              @Override
              public void onNext(byte[] bytes) {
                LogUtils.d("setup getCharacteristic next");
                mUnlockException = UnlockException.TOKEN_REQUEST_RESOLVE_FAILED;
                write(ByteUtils.hexStringToByteArray(mDeviceInfo.getTokenRequest()),
                    UnlockException.TOKEN_REQUEST_WRITE_FAILED);
              }
            });
And the log is as below:

11-17 16:26:30.243 26865-26865/ble.ubike.com.ble_demo E/BleLib: BleGattDescriptorException{macAddress=50:F1:4A:4B:5B:1C, status=17 (0x11 -> https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h), bleGattOperationType=BleGattOperation{description='DESCRIPTOR_WRITE'}}
                                                                    at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateErrorIfOccurred(RxBleGattCallback.java:242)
                                                                    at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$1200(RxBleGattCallback.java:32)
                                                                    at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$2.onDescriptorWrite(RxBleGattCallback.java:169)
                                                                    at android.bluetooth.BluetoothGatt$1.onDescriptorWrite(BluetoothGatt.java:484)
                                                                    at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:165)
                                                                    at android.os.Binder.execTransact(Binder.java:573)

And I have checked the gatt_api.h file and found that the code 0x11 has two corresponding items:

#define  GATT_INSUF_RESOURCE                 0x11
#define  GATT_RSP_READ_BY_GRP_TYPE           0x11

But I have no idea which one is right, any help will be highly appreciated, thanks a lot.

uKL, 2017-11-17:
Hey,

first, for notifications, you should be using a `setupNotification` API provided by the library.
second, are you sure that your characteristic supports notifications?
third, this is a general usage question. Please follow up on stackoverflow with #rxandroidble tag or contact hello@polidea.com for commercial support.

closed 2017-11-17 by uKL

Xinima, 2017-11-21:
Hi uKL,
I did some research in the last few days, and I found that this problem is caused by writeDescriptor failed, and I successfully unlocked the same lock with my old code which use BluetoothGatt directly, I used the code below:

    BluetoothGattService service = gatt.getService(bltServerUUID);
    BluetoothGattCharacteristic readCharacteristic = service.getCharacteristic(readDataUUID);
    writeCharacteristic = service.getCharacteristic(writeDataUUID);
    gatt.setCharacteristicNotification(readCharacteristic, true);

    BluetoothGattDescriptor descriptor = readCharacteristic.getDescriptor(clientUUID);
    descriptor.setValue(BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE);
    gatt.writeDescriptor(descriptor);

Meanwhile the corresponding code I used with RxAndroidBle is as below:

    mRxBleConnection.getCharacteristic(READ_DATA_UUID)
        .subscribeOn(AndroidSchedulers.mainThread())
        .observeOn(Schedulers.computation())
        .map(new Func1<BluetoothGattCharacteristic, BluetoothGattDescriptor>() {
        @Override
        public BluetoothGattDescriptor call(
            BluetoothGattCharacteristic bluetoothGattCharacteristic) {
                BluetoothGattDescriptor descriptor =
                    bluetoothGattCharacteristic.getDescriptor(DESCRIPTOR_UUID);
            descriptor.setValue(BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE);
            return descriptor;
      }
    })
    .flatMap(new Func1<BluetoothGattDescriptor, Observable<byte[]>>() {
      @Override
      public Observable<byte[]> call(BluetoothGattDescriptor bluetoothGattDescriptor) {
          return mRxBleConnection.writeDescriptor(bluetoothGattDescriptor,
              BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE);
      }
    })

And additionally, I found that on different Android platforms the error is different, on Android 4.4, it reports the error below:

11-21 14:26:51.127 3885-4009/? W/dalvikvm: VFY: unable to resolve virtual method 496: Landroid/bluetooth/BluetoothGattCallback;.onMtuChanged (Landroid/bluetooth/BluetoothGatt;II)V
11-21 14:26:51.157 3885-4079/? D/BluetoothGatt: connect() - device: 50:F1:4A:4B:5B:1C, auto: false
11-21 14:26:51.157 3885-4079/? D/BluetoothGatt: registerApp()
11-21 14:26:51.157 3885-4079/? D/BluetoothGatt: registerApp() - UUID=7d9069e6-d1ff-49d5-b6fc-34d04cade854
11-21 14:26:51.157 3885-3910/? D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
**11-21 14:26:57.657 3885-3909/? D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=5 device=50:F1:4A:4B:5B:1C**
11-21 14:26:57.667 3885-4079/? D/BluetoothManager: getConnectionState()
11-21 14:26:57.667 3885-4079/? D/BluetoothManager: getConnectedDevices
11-21 14:26:57.677 3885-4079/? D/BluetoothGatt: close()
11-21 14:26:57.677 3885-4079/? D/BluetoothGatt: unregisterApp() - mClientIf=5
11-21 14:26:57.677 3885-3885/? E/BleLib: connect result: Failed
11-21 14:26:57.677 3885-3885/? E/BleLib: BleGattException{macAddress=50:F1:4A:4B:5B:1C, status=133 (0x85 -> https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h), bleGattOperationType=BleGattOperation{description='CONNECTION_STATE'}}
                                             at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:78)
                                             at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:168)
                                             at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:71)
                                             at android.os.Binder.execTransact(Binder.java:404)
                                             at dalvik.system.NativeStart.run(Native Method)

And on Android platforms above 4.4, it reports:

11-17 16:26:30.243 26865-26865/ble.ubike.com.ble_demo E/BleLib: BleGattDescriptorException{macAddress=50:F1:4A:4B:5B:1C, status=17 (0x11 -> https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h), bleGattOperationType=BleGattOperation{description='DESCRIPTOR_WRITE'}}
                                                                    at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateErrorIfOccurred(RxBleGattCallback.java:242)
                                                                    at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$1200(RxBleGattCallback.java:32)
                                                                    at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$2.onDescriptorWrite(RxBleGattCallback.java:169)
                                                                    at android.bluetooth.BluetoothGatt$1.onDescriptorWrite(BluetoothGatt.java:484)
                                                                    at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:165)
                                                                    at android.os.Binder.execTransact(Binder.java:573)

Looking forward to your reply, thanks in advance