dariuszseweryn/RxAndroidBle #324: BleGattDescriptorException(writeDescriptor failed, but using BluetoothGatt directly there is no Problem)

Xinima, 2017-11-21:
Hi there,
I am trying to unlock a bluetooth bike lock, but it reports an BleGattDescriptorException. I've checked some old issues and can't find a same issue.I did some research in the last few days, and found that this problem is caused by writeDescriptor failed, and I successfully unlocked the same lock with my old code which use BluetoothGatt directly, I used the code below:

    BluetoothGattService service = gatt.getService(bltServerUUID);
    BluetoothGattCharacteristic readCharacteristic = service.getCharacteristic(readDataUUID);
    writeCharacteristic = service.getCharacteristic(writeDataUUID);
    gatt.setCharacteristicNotification(readCharacteristic, true);

    BluetoothGattDescriptor descriptor = readCharacteristic.getDescriptor(clientUUID);
    descriptor.setValue(BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE);
    gatt.writeDescriptor(descriptor);

Meanwhile the corresponding code I used with RxAndroidBle is as below:

    mRxBleConnection.getCharacteristic(READ_DATA_UUID)
        .subscribeOn(AndroidSchedulers.mainThread())
        .observeOn(Schedulers.computation())
        .map(new Func1<BluetoothGattCharacteristic, BluetoothGattDescriptor>() {
        @Override
        public BluetoothGattDescriptor call(
            BluetoothGattCharacteristic bluetoothGattCharacteristic) {
                BluetoothGattDescriptor descriptor =
                    bluetoothGattCharacteristic.getDescriptor(DESCRIPTOR_UUID);
                descriptor.setValue(BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE);
                return descriptor;
            }
        })
       .flatMap(new Func1<BluetoothGattDescriptor, Observable<byte[]>>() {
       @Override
       public Observable<byte[]> call(BluetoothGattDescriptor bluetoothGattDescriptor) {
             return mRxBleConnection.writeDescriptor(bluetoothGattDescriptor,
                 BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE);
       }
  })

And additionally, I found that on different Android platforms the error is different, on Android 4.4, it reports the error below:

    11-21 14:26:51.127 3885-4009/? W/dalvikvm: VFY: unable to resolve virtual method 496:     
     Landroid/bluetooth/BluetoothGattCallback;.onMtuChanged (Landroid/bluetooth/BluetoothGatt;II)V
    11-21 14:26:51.157 3885-4079/? D/BluetoothGatt: connect() - device: 50:F1:4A:4B:5B:1C, auto: false
    11-21 14:26:51.157 3885-4079/? D/BluetoothGatt: registerApp()
    11-21 14:26:51.157 3885-4079/? D/BluetoothGatt: registerApp() - UUID=7d9069e6-d1ff-49d5-b6fc-   
    34d04cade854
    11-21 14:26:51.157 3885-3910/? D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
    11-21 14:26:57.657 3885-3909/? D/BluetoothGatt: onClientConnectionState() - status=133  
    clientIf=5 device=50:F1:4A:4B:5B:1C
    11-21 14:26:57.667 3885-4079/? D/BluetoothManager: getConnectionState()
    11-21 14:26:57.667 3885-4079/? D/BluetoothManager: getConnectedDevices
    11-21 14:26:57.677 3885-4079/? D/BluetoothGatt: close()
    11-21 14:26:57.677 3885-4079/? D/BluetoothGatt: unregisterApp() - mClientIf=5
    11-21 14:26:57.677 3885-3885/? E/BleLib: connect result: Failed
    11-21 14:26:57.677 3885-3885/? E/BleLib: BleGattException{macAddress=50:F1:4A:4B:5B:1C,    
    status=133 (0x85 ->     
    https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-    
    5.1.0_r1/stack/include/gatt_api.h),    
    bleGattOperationType=BleGattOperation{description='CONNECTION_STATE'}}
    at 
com.polidea.rxandroidble.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleG  
    attCallback.java:78)
    at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:168)
    at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:71)
    at android.os.Binder.execTransact(Binder.java:404)
    at dalvik.system.NativeStart.run(Native Method)

And on Android platforms above 4.4, it reports:

    11-17 16:26:30.243 26865-26865/ble.ubike.com.ble_demo E/BleLib:    
    BleGattDescriptorException{macAddress=50:F1:4A:4B:5B:1C, status=17 (0x11 ->    
    https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-
    5.1.0_r1/stack/include/gatt_api.h), 
    bleGattOperationType=BleGattOperation{description='DESCRIPTOR_WRITE'}}
    at    
com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateErrorIfOccurred(RxBleGatt
    Callback.java:242)
    at   
 com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$1200(RxBleGattCallback.java    
    :32)
    at  com.polidea.rxandroidble.internal.connection.RxBleGattCallback$2.onDescriptorWrite(RxBleGattCallba
    ck.java:169)
    at android.bluetooth.BluetoothGatt$1.onDescriptorWrite(BluetoothGatt.java:484)
    at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:165)
    at android.os.Binder.execTransact(Binder.java:573)

Looking forward to your reply, thanks in advance.

dariuszseweryn, 2017-11-21:
Hello @Xinima 
Thank you for using the library!
There are few things that come to my mind:
1. Your code snippets do not seem to contain all the information about what do you do with the peripheral. In the first snippet there is a reference of `writeCharacteristic` which is not seen on the second snippet
2. Using the library you can setup the notification by subscribing to `RxBleConnection.setupNotification(BluetoothGattCharacteristic)`—you do not have to manually write the CCC Descriptor in most cases
3. The log from 4.4 does not even establish a connection not to mention trying to write the descriptor
4. The log from above 4.4 does not contain any information about the execution before the error
5. `status=17` means that the Gatt has no resources when writing the descriptor—hard to say why without more informations
Anyway—I do not know what is the algorithm for the lock you are using. There is a lack of information on what actually you are doing with the connection. So far it does not seem that there is a bug in the library and as such this question should be posted [on stackoverflow](https://stackoverflow.com/questions/tagged/rxandroidble)

Xinima, 2017-11-21:
Hi @dariuszseweryn，thanks a lot for your reply.
about the questions you mentions above, here are the explanations:
1.I have attach the whole setup method below:

     private void setup() {
        Subscription subscription = mRxBleClient.observeStateChanges()
            .switchMap(
                new Func1<State, Observable<?>>() {
                  @Override
                  public Observable<?> call(State state) {
                    switch (state) {
                      case READY:
                          if (mOnBleLockListener != null) {
                              mOnBleLockListener.onBluetoothEnable(true);
                          }
                      break;

                      case BLUETOOTH_NOT_ENABLED:
                          if (mOnBleLockListener != null) {
                              mOnBleLockListener.onBluetoothEnable(false);
                          }
                      break;

                      default:
                          return Observable.empty();
                    }
                    return Observable.empty();
                  }
              }).subscribe();
        mSubscriptionList.add(subscription);

        Subscription subscriptionConnect = mRxBleDevice.observeConnectionStateChanges()
            .observeOn(AndroidSchedulers.mainThread())
            .subscribe(
                new Subscriber<RxBleConnectionState>() {
                    @Override
                    public void onCompleted() {
                    }

                    @Override
                    public void onError(Throwable e) {
                        LogUtils.e("observeConnectionStateChanges error");
                        LogUtils.e(Log.getStackTraceString(e));
                    }

                   @Override
                   public void onNext(RxBleConnectionState rxBleConnectionState) {
                       LogUtils.d(rxBleConnectionState.toString());
                      if (rxBleConnectionState == RxBleConnectionState.DISCONNECTED) {
                          handleUnlockFailed(BleUnLockState.setupReady, UnlockException.CONNECT_FAILED);
                         if (mOnBleLockListener != null) {
                            mOnBleLockListener.onDisconnect();
                          }
                      }
                   }
               });
         mSubscriptionList.add(subscriptionConnect);

        Subscription subscriptionNotify = mRxBleConnection
            .setupNotification(READ_DATA_UUID, NotificationSetupMode.COMPAT)
            .subscribeOn(AndroidSchedulers.mainThread())
            .doOnNext(new Action1<Observable<byte[]>>() {
                @Override
                public void call(Observable<byte[]> observable) {
                    LogUtils.d("setupNotification done");
                }
             })
            .flatMap(new Func1<Observable<byte[]>, Observable<byte[]>>() {
                @Override
                public Observable<byte[]> call(Observable<byte[]> observable) {
[... 201 lines cut ...]
                                                                        at 
    com.jakewharton.rxrelay.PublishRelay.call(PublishRelay.java:47)
                                                                        at 
com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateStatusError(RxBleGattCallback.java:249)
                                                                        at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateErrorIfOccurred(RxBleGattCallback.java:231) 
                                                                        at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$1100(RxBleGattCallback.java:34) 
                                                                        at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$4.onDescriptorWrite(RxBleGattCallback.java:163) 
                                                                        at android.bluetooth.BluetoothGatt$1$10.run(BluetoothGatt.java:529) 
                                                                        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:711) 
                                                                        at android.bluetooth.BluetoothGatt.-wrap0(Unknown Source:0) 
                                                                        at android.bluetooth.BluetoothGatt$1.onDescriptorWrite(BluetoothGatt.java:525) 
                                                                        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:172) 
                                                                        at android.os.Binder.execTransact(Binder.java:674) 
11-20 15:49:27.415 22120-22120/ble.ubike.com.ble_demo D/BluetoothGatt:      
    setCharacteristicNotification() - uuid: 000036f6-0000-1000-8000-00805f9b34fb enable: false
    11-20 15:49:27.429 22120-22120/ble.ubike.com.ble_demo D/BluetoothManager:    
    getConnectionState()
    11-20 15:49:27.429 22120-22120/ble.ubike.com.ble_demo D/BluetoothManager: 
    getConnectedDevices
    11-20 15:49:27.467 22120-22120/ble.ubike.com.ble_demo D/BluetoothGatt: cancelOpen() - device: 
    50:F1:4A:4B:5B:1C
    11-20 15:49:27.470 22120-22142/ble.ubike.com.ble_demo D/BluetoothGatt: 
    onClientConnectionState() - status=0 clientIf=6 device=50:F1:4A:4B:5B:1C
    11-20 15:49:27.548 22120-22120/ble.ubike.com.ble_demo D/BluetoothGatt: close()
    11-20 15:49:27.549 22120-22120/ble.ubike.com.ble_demo D/BluetoothGatt: unregisterApp() - mClientIf=6

dariuszseweryn, 2017-11-21:
`RxBleLog.setLogLevel(RxBleLog.VERBOSE)` without it there are no logs from the library.
Why do you use `COMPAT` mode of setting the notifications?

Xinima, 2017-11-22:
Hello @dariuszseweryn I have enable the Log printing and the log is as below:

    11-22 09:50:55.446 17034-17295/ble.ubike.com.ble_demo D/BleLib: Retorfit request url:     
    https://api2test.ubike.cn/ls5/v2/ble/Rt9d07lomP/generate-auth?code=B000264806, body: 
    okhttp3.RequestBody$2@3f614d3, header: CLIENT-TOKEN: w80*X&3-=x234M
                                                                geolocation: geo:121.0,31.0,0;mt=0;u=0
    11-22 09:50:55.901 17034-17034/ble.ubike.com.ble_demo D/BleLib: GetTokenResponse{code=0,   
    msg=ok, obj=DeviceInfo{imei=010250F14A4B5B1C, 
    auth=CC6E8EF0E6A2BFC893871E571CA06D7D, address=null, 
    tokenRequest=B7D32028531B4F70E30392206BD52C83}}
    11-22 09:50:55.911 17034-17295/ble.ubike.com.ble_demo D/BleLib: connect to mac address:  
    50:F1:4A:4B:5B:1C
    11-22 09:50:55.961 17034-17303/ble.ubike.com.ble_demo D/RxBle#ClientOperationQueue:   
    QUEUED ConnectOperation(227886569)
    11-22 09:50:55.962 17034-17108/ble.ubike.com.ble_demo D/RxBle#ClientOperationQueue:     
    STARTED ConnectOperation(227886569)
    11-22 09:50:55.970 17034-17304/ble.ubike.com.ble_demo V/RxBle#BleConnectionCompat: 
    Connecting without reflection
    11-22 09:50:55.975 17034-17304/ble.ubike.com.ble_demo D/BluetoothGatt: connect() - device: 
    50:F1:4A:4B:5B:XX, auto: false
    11-22 09:50:55.975 17034-17304/ble.ubike.com.ble_demo D/BluetoothAdapter: 
    isSecureModeEnabled
    11-22 09:50:55.975 17034-17304/ble.ubike.com.ble_demo D/BluetoothGatt: registerApp()
    11-22 09:50:55.976 17034-17304/ble.ubike.com.ble_demo D/BluetoothGatt: registerApp() -  
    UUID=5e5b3431-ef64-47aa-9160-41edc6f7dad8
    11-22 09:50:56.082 17034-17301/ble.ubike.com.ble_demo D/BluetoothGatt: onClientRegistered() - 
    status=0 clientIf=4
    11-22 09:50:56.825 17034-17047/ble.ubike.com.ble_demo D/BluetoothGatt: 
    onClientConnectionState() - status=0 clientIf=4 device=50:F1:4A:4B:5B:XX
    11-22 09:50:56.835 17034-17047/ble.ubike.com.ble_demo D/RxBle#BluetoothGatt: 
    onConnectionStateChange newState=2 status=0
    11-22 09:50:56.839 17034-17108/ble.ubike.com.ble_demo D/RxBle#ClientOperationQueue: 
    FINISHED ConnectOperation(227886569)
    11-22 09:50:56.841 17034-17034/ble.ubike.com.ble_demo D/BleLib: connect result: Connected
    11-22 09:50:56.942 17034-17034/ble.ubike.com.ble_demo D/RxBle#ConnectionOperationQueue:   
    QUEUED ServiceDiscoveryOperation(104381195)
    11-22 09:50:56.943 17034-17302/ble.ubike.com.ble_demo D/RxBle#ConnectionOperationQueue:  
    STARTED ServiceDiscoveryOperation(104381195)
    11-22 09:50:56.946 17034-17304/ble.ubike.com.ble_demo D/BluetoothGatt: discoverServices() - 
    device: 50:F1:4A:4B:5B:XX
    11-22 09:50:57.097 17034-17048/ble.ubike.com.ble_demo D/BluetoothGatt: 
    onClientConnParamsChanged() - Device=50:F1:4A:4B:5B:XX interval=6 status=0
    11-22 09:50:57.314 17034-17048/ble.ubike.com.ble_demo D/BluetoothGatt: onSearchComplete() = 
    Device=50:F1:4A:4B:5B:XX Status=0
    11-22 09:50:57.320 17034-17048/ble.ubike.com.ble_demo D/RxBle#BluetoothGatt: 
    onServicesDiscovered status=0
    11-22 09:50:57.346 17034-17303/ble.ubike.com.ble_demo D/BluetoothGatt: 
    setCharacteristicNotification() - uuid: 000036f6-0000-1000-8000-00805f9b34fb enable: true
    11-22 09:50:57.353 17034-17303/ble.ubike.com.ble_demo D/BleLib: setupNotification done
    11-22 09:50:57.360 17034-17302/ble.ubike.com.ble_demo D/RxBle#ConnectionOperationQueue:    
    FINISHED ServiceDiscoveryOperation(104381195)
    11-22 09:50:57.368 17034-17034/ble.ubike.com.ble_demo D/RxBle#ConnectionOperationQueue:   
    QUEUED DescriptorWriteOperation(90984544)
    11-22 09:50:57.370 17034-17302/ble.ubike.com.ble_demo D/RxBle#ConnectionOperationQueue:  
    STARTED DescriptorWriteOperation(90984544)
    11-22 09:50:57.373 17034-17047/ble.ubike.com.ble_demo D/BluetoothGatt:    
    onClientConnParamsChanged() - Device=50:F1:4A:4B:5B:XX interval=39 status=0
    11-22 09:50:57.475 17034-17047/ble.ubike.com.ble_demo D/RxBle#BluetoothGatt: 
    onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=17
    11-22 09:50:57.485 17034-17302/ble.ubike.com.ble_demo D/RxBle#ConnectionOperationQueue: 
    FINISHED DescriptorWriteOperation(90984544)
    11-22 09:50:57.489 17034-17034/ble.ubike.com.ble_demo E/BleLib: setup getCharacteristic error
    11-22 09:50:57.491 17034-17034/ble.ubike.com.ble_demo E/BleLib: 
    com.polidea.rxandroidble.exceptions.BleGattDescriptorException: GATT exception from MAC 
    address 50:F1:4A:4B:5B:1C, status 17, type BleGattOperation{description='DESCRIPTOR_WRITE'}. 
    (Look up status 0x11 here 
    https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-
    5.1.0_r1/stack/include/gatt_api.h)
    at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateErrorIfOccurred(RxBleGatt   
    Callback.java:242)
[... 127 lines cut ...]
    rx.internal.schedulers.ScheduledAction.run(ScheduledAction.java:55)
                                                                    at 
rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker.run(ExecutorScheduler.java:107)
                                                                    at 
    java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
                                                                    at 
    java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
[... cut ...]
com.polidea.rxandroidble.internal.connection.RxBleGattCallback$2.onDescriptorWrite(RxBleGattCallback.java:169)
                                                                    at 
    android.bluetooth.BluetoothGatt$1.onDescriptorWrite(BluetoothGatt.java:484)
                                                                    at 
    android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:165)
                                                                    at android.os.Binder.execTransact(Binder.java:573)

Xinima, 2017-11-22:
Update:
I found that in class DescriptorWriteOperation's startOperation method the value of final " boolean success = bluetoothGatt.writeDescriptor(bluetoothGattDescriptor);"  is true, 

    @Override
    protected boolean startOperation(BluetoothGatt bluetoothGatt) {
        bluetoothGattDescriptor.setValue(data);

        /*
        * According to the source code below Android 7.0.0 the BluetoothGatt.writeDescriptor() function  
           used
        * writeType of the parent BluetoothCharacteristic which caused operation failure (for instance 
           when
        * setting Client Characteristic Config). With WRITE_TYPE_DEFAULT problem did not occurred.
        * Compare:
        * https://android.googlesource.com/platform/frameworks/base/+/android-
          6.0.1_r74/core/java/android/bluetooth/BluetoothGatt.java#1039
        * https://android.googlesource.com/platform/frameworks/base/+/android-
          7.0.0_r1/core/java/android/bluetooth/BluetoothGatt.java#947
        */
        final BluetoothGattCharacteristic bluetoothGattCharacteristic =    
            bluetoothGattDescriptor.getCharacteristic();
        final int originalWriteType = bluetoothGattCharacteristic.getWriteType();
        bluetoothGattCharacteristic.setWriteType(bluetoothGattCharacteristicDefaultWriteType);

        final boolean success = bluetoothGatt.writeDescriptor(bluetoothGattDescriptor);
        bluetoothGattCharacteristic.setWriteType(originalWriteType);
        return success;
    }
however in the RxBleGattCallback class's onDescriptorWrite callback the status attribute's value is 17, 
        @Override
        public void onDescriptorWrite(BluetoothGatt gatt, BluetoothGattDescriptor descriptor, int status)    
        {
            RxBleLog.d("onDescriptorWrite descriptor=%s status=%d", descriptor.getUuid(), status);
            nativeCallbackDispatcher.notifyNativeDescriptorWriteCallback(gatt, descriptor, status);
            super.onDescriptorWrite(gatt, descriptor, status);

            if (writeDescriptorOutput.hasObservers()
                    && !propagateErrorIfOccurred(writeDescriptorOutput, gatt, descriptor, status, 
            BleGattOperationType.DESCRIPTOR_WRITE)) {
                writeDescriptorOutput.valueRelay.call(new ByteAssociation<>(descriptor, 
                    descriptor.getValue()));
            }
        }

I am a little confused about this, do you have any clue please?

dariuszseweryn, 2017-11-22:
`startOperation()` returned `success=true`. Callback returned `status=17` which is the final result of the operation. If this is a confusion to you I would encourage you to read some tutorials about BLE on Android.
So far it seems that your problem is in the BLE stack of the Android and/or the lock firmware. If you say that using the same phone/lock you were able to make it work- the only thing that comes to my mind is that the timing is bad in sense that the `Connection Parameters` are updated right before the descriptor is being written.
You could try postponing the notification setup/descriptor write a bit so they will trigger after the connection update.
Posting logs from a successful unlock that you have mentioned at the beginning could also be useful to understand the problem with the connection.

Xinima, 2017-11-22:
@dariuszseweryn thanks a lot for your help, here is the log that I used the old code to unlock the same lock with the same phone:

    11-22 16:51:06.351 15730-15730/com.yl.ubike E/UBike: BlueTooth scan Device  
    11-22 16:51:06.352 15730-15730/com.yl.ubike D/BluetoothAdapter: STATE_ON
    11-22 16:51:06.353 15730-15730/com.yl.ubike D/BluetoothAdapter: STATE_ON
    11-22 16:51:06.354 15730-15730/com.yl.ubike D/BluetoothAdapter: STATE_ON
    11-22 16:51:06.355 15730-15730/com.yl.ubike D/BluetoothAdapter: STATE_ON
    11-22 16:51:06.355 15730-15730/com.yl.ubike D/BluetoothLeScanner: Start Scan
    11-22 16:51:06.356 15730-15730/com.yl.ubike D/BluetoothAdapter: STATE_ON
    11-22 16:51:06.356 15730-15730/com.yl.ubike D/BluetoothAdapter: STATE_ON
    11-22 16:51:06.358 15730-15730/com.yl.ubike D/BluetoothAdapter: STATE_ON
    11-22 16:51:06.358 15730-15730/com.yl.ubike D/BluetoothAdapter: STATE_ON
    11-22 16:51:06.465 15730-15742/com.yl.ubike D/BluetoothLeScanner: onClientRegistered() -   
    status=0 clientIf=4 mClientIf=0
    11-22 16:51:09.468 15730-15730/com.yl.ubike E/UBike: BlueTooth find Device~~~~~~~
    11-22 16:51:09.468 15730-15730/com.yl.ubike E/UBike: BlueTooth stop scan
    11-22 16:51:09.471 15730-15730/com.yl.ubike D/BluetoothAdapter: STATE_ON
    11-22 16:51:09.473 15730-15730/com.yl.ubike D/BluetoothAdapter: STATE_ON
    11-22 16:51:09.474 15730-15730/com.yl.ubike D/BluetoothAdapter: STATE_ON
    11-22 16:51:09.475 15730-15730/com.yl.ubike D/BluetoothAdapter: STATE_ON
    11-22 16:51:09.476 15730-15730/com.yl.ubike D/BluetoothAdapter: STATE_ON
    11-22 16:51:09.478 15730-15730/com.yl.ubike D/BluetoothAdapter: STATE_ON
    11-22 16:51:09.478 15730-15730/com.yl.ubike D/BluetoothLeScanner: Stop Scan
    11-22 16:51:09.489 15730-15730/com.yl.ubike D/BluetoothManager: getConnectionState()
    11-22 16:51:09.489 15730-15730/com.yl.ubike D/BluetoothManager: getConnectedDevices
    11-22 16:51:09.502 15730-15730/com.yl.ubike D/BluetoothGatt: connect() - device: 
    50:F1:4A:4B:5B:XX, auto: false
    11-22 16:51:09.502 15730-15730/com.yl.ubike D/BluetoothAdapter: isSecureModeEnabled
    11-22 16:51:09.502 15730-15730/com.yl.ubike D/BluetoothGatt: registerApp()
    11-22 16:51:09.503 15730-15730/com.yl.ubike D/BluetoothGatt: registerApp() - UUID=c221825f-
    1638-4ccc-a01c-3adbb3fc6fa3
    11-22 16:51:09.607 15730-15742/com.yl.ubike D/BluetoothGatt: onClientRegistered() - status=0 
    clientIf=4
    11-22 16:51:12.049 15730-15982/com.yl.ubike D/BluetoothGatt: onClientConnectionState() -  
    status=0 clientIf=4 device=50:F1:4A:4B:5B:XX
    11-22 16:51:12.063 15730-15982/com.yl.ubike D/BluetoothGatt: discoverServices() - device: 
    50:F1:4A:4B:5B:XX
    11-22 16:51:12.064 15730-15982/com.yl.ubike E/UBike: BlueTooth state : STATE_CONNECTED
    11-22 16:51:12.296 15730-15742/com.yl.ubike D/BluetoothGatt: onClientConnParamsChanged() - 
    Device=50:F1:4A:4B:5B:XX interval=6 status=0
    11-22 16:51:12.577 15730-15743/com.yl.ubike D/BluetoothGatt: onSearchComplete() = 
    Device=50:F1:4A:4B:5B:XX Status=0
    11-22 16:51:12.577 15730-15743/com.yl.ubike E/UBike: BlueTooth onServicesDiscovered
    11-22 16:51:12.577 15730-15743/com.yl.ubike D/BluetoothGatt: setCharacteristicNotification() - 
    uuid: 000036f6-0000-1000-8000-00805f9b34fb enable: true
    11-22 16:51:12.581 15730-15743/com.yl.ubike E/UBike: BlueTooth state CONNECT
    11-22 16:51:12.581 15730-15743/com.yl.ubike E/UBike: BlueTooth state CONNECT
    11-22 16:51:12.662 15730-15742/com.yl.ubike D/BluetoothGatt: onClientConnParamsChanged() - 
    Device=50:F1:4A:4B
    5B:XX interval=39 status=0
    11-22 16:51:15.194 15730-15743/com.yl.ubike E/UBike: BlueTooth return data~~~~~~
    11-22 16:51:15.194 15730-15743/com.yl.ubike E/UBike: BlueTooth return data~~~~~~
                                                     
                                                        88B10FDF8C4507A578C95B357706DEE900000000
    11-22 16:51:15.208 15730-15743/com.yl.ubike E/UBike: BlueTooth return data~~~~~~
    11-22 16:51:15.208 15730-15743/com.yl.ubike E/UBike: BlueTooth return data~~~~~~
                                                     
                                                        691460BBA41B5CA1E768FE1C4AAE3EDD00000000
    11-22 16:51:15.828 15730-15742/com.yl.ubike E/UBike: BlueTooth return data~~~~~~
    11-22 16:51:15.828 15730-15742/com.yl.ubike E/UBike: BlueTooth return data~~~~~~
                                                     
                                                        3F100136A768CA656E29335998636A6F00000000
    11-22 16:51:17.194 15730-15743/com.yl.ubike E/UBike: BlueTooth return data~~~~~~
    11-22 16:51:17.195 15730-15743/com.yl.ubike E/UBike: BlueTooth return data~~~~~~
                                                     
                                                        FFA219C3338D7B23C2598D90A3257D9400000000
    11-22 16:51:18.709 15730-15743/com.yl.ubike D/BluetoothGatt: onClientConnParamsChanged() - 
    Device=50:F1:4A:4B:5B:XX interval=318 status=0

    11-22 16:51:21.491 15730-15743/com.yl.ubike E/UBike: BlueTooth return data~~~~~~
    11-22 16:51:21.491 15730-15743/com.yl.ubike E/UBike: BlueTooth return data~~~~~~
                                                     
                                                        DE5A3D7ED083F81A587A166FB74FF8E000000000
    11-22 16:51:21.497 15730-15982/com.yl.ubike E/UBike: BlueTooth return data~~~~~~
[... cut ...]
    11-22 16:53:00.644 15730-15730/com.yl.ubike D/BluetoothGatt: close()
    11-22 16:53:00.649 15730-15730/com.yl.ubike D/BluetoothGatt: unregisterApp() - mClientIf=4
    11-22 16:53:00.656 15730-15730/com.yl.ubike E/UBike: mBluetoothGatt.disconnect()
    11-22 16:53:00.660 15730-15730/com.yl.ubike D/BluetoothGatt: cancelOpen() - device: 
    50:F1:4A:4B:5B:XX

I tried  postponing the setupNotification and writeDescriptor for several hundreds milliseconds, but still no luck, it reported the same error.

dariuszseweryn, 2017-11-22:
In the log above there is no indication that the CCC descriptor is being written. Am I correct?

Xinima, 2017-11-22:
Yes, you are right. Actually I found that in my old code it also reports status=17, but I didn't handle it at all... And it just can unlock...

dariuszseweryn, 2017-11-22:
So this is not an issue with the library but rather with Peripheral Firmware or Android BLE Stack

closed 2017-11-22 by dariuszseweryn

Xinima, 2017-11-22:
yes, my bad, a lot thanks for your help, have a nice day