dariuszseweryn/RxAndroidBle #327: Location Services disabled (code 4) even when Location Services are Activated - Android Pen

ejbp, 2017-11-26:
**Error:** 
com.polidea.rxandroidble.exceptions.BleScanException: Location Services disabled (code 4)

**Facts:**
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/> is in manifest and user accepts the checkSelfPermission.

**Double checked:**

I double checked it in settings and location services are enabled and the app is allowed to such permission.

**Hardware && versions:**
[ro.build.version.release]: [7.1.2]
[ro.build.version.sdk]: [25]
[ro.product.board]: [q8723bs]
[ro.product.brand]: [Amlogic]
[ro.product.cpu.abi]: [arm64-v8a]
[ro.product.cpu.abilist]: [arm64-v8a,armeabi-v7a,armeabi]
[ro.product.cpu.abilist32]: [armeabi-v7a,armeabi]
[ro.product.cpu.abilist64]: [arm64-v8a]
[ro.product.device]: [q8723bs]
[ro.product.firmware]: [00502001]
[ro.product.locale]: [en-US]
[ro.product.manufacturer]: [Amlogic]
[ro.product.model]: [H96 PRO]
[ro.product.name]: [q8723bs]

Hardware [Link](https://www.geekbuying.com/item/H96-Pro-4K-Amlogic-S912-KODI-17-3-TV-Dongle-382407.html?Currency=USD&gclid=EAIaIQobChMI3rqj3Ind1wIVCD8bCh3E8Qm2EAQYASABEgIlE_D_BwE)

I can't figure out the reason of this exception. Any idea?

uKL, 2017-11-27:
Could you make a snippet of code and post what does 
`locationManager.getAllProviders()` will print out?

The library expects either `NETWORK_PROVIDER` or `GPS_PROVIDER`.

ejbp, 2017-11-27:
Result of : 

`Log.d("BluetoothCommManager", "Providers: " + locationManager.getAllProviders());`

is 

`11-27 09:43:59.023 20336-20336/? D/BluetoothCommManager: Providers: [passive, network]`

When I run:

```
scanSubscription = rxBleClient.scanBleDevices(

                    new ScanSettings.Builder()
                            .setScanMode(ScanSettings.SCAN_MODE_BALANCED)
                            // .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY) // change if needed
                            // .setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES) // change if needed
                            .build(),

                    new ScanFilter.Builder().setDeviceName("BTDeviceName").build()

            ).subscribe(

                    scanResult -> {
                        // Process scan result here.

                        Log.d("BluetoothComm", "ScanResult: " + scanResult);

                    },
                    throwable -> {
                        // Handle an error here.
                        Log.d("BluetoothComm", "Bluetooth Scan Error: ");
                        throwable.printStackTrace();
                    }

            );
```

the result is:

```
11-27 09:43:59.028 20336-20336/? D/BluetoothComm: Bluetooth Scan Error: 
11-27 09:43:59.028 20336-20336/? W/System.err: com.polidea.rxandroidble.exceptions.BleScanException: Location Services disabled (code 4)
11-27 09:43:59.028 20336-20336/? W/System.err:     at com.polidea.rxandroidble.internal.scan.ScanPreconditionsVerifierApi18.verify(ScanPreconditionsVerifierApi18.java:31)
11-27 09:43:59.028 20336-20336/? W/System.err:     at com.polidea.rxandroidble.internal.scan.ScanPreconditionsVerifierApi24.verify(ScanPreconditionsVerifierApi24.java:38)
11-27 09:43:59.028 20336-20336/? W/System.err:     at com.polidea.rxandroidble.RxBleClientImpl$1.call(RxBleClientImpl.java:114)
11-27 09:43:59.028 20336-20336/? W/System.err:     at com.polidea.rxandroidble.RxBleClientImpl$1.call(RxBleClientImpl.java:111)
11-27 09:43:59.028 20336-20336/? W/System.err:     at rx.internal.operators.OnSubscribeDefer.call(OnSubscribeDefer.java:46)
11-27 09:43:59.028 20336-20336/? W/System.err:     at rx.internal.operators.OnSubscribeDefer.call(OnSubscribeDefer.java:35)
11-27 09:43:59.028 20336-20336/? W/System.err:     at rx.Observable.subscribe(Observable.java:10352)
11-27 09:43:59.028 20336-20336/? W/System.err:     at rx.Observable.subscribe(Observable.java:10319)
11-27 09:43:59.028 20336-20336/? W/System.err:     at rx.Observable.subscribe(Observable.java:10159)
11-27 09:43:59.028 20336-20336/? W/System.err:     at com.xxx.comm.bluetooth.BluetoothCommManager.scanDevice(BluetoothCommManager.java:74)

//Log started here:
11-27 09:43:59.028 20336-20336/? W/System.err:     at com.xxx.gui.launcher.PairingScene.start(PairingScene.java:83)
11-27 09:43:59.028 20336-20336/? W/System.err:     at com.xxx.LauncherActivity.onResume(LauncherActivity.java:345)
11-27 09:43:59.028 20336-20336/? W/System.err:     at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1269)
11-27 09:43:59.028 20336-20336/? W/System.err:     at android.app.Activity.performResume(Activity.java:6786)
11-27 09:43:59.028 20336-20336/? W/System.err:     at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3407)
11-27 09:43:59.028 20336-20336/? W/System.err:     at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3470)
11-27 09:43:59.028 20336-20336/? W/System.err:     at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2733)
11-27 09:43:59.028 20336-20336/? W/System.err:     at android.app.ActivityThread.-wrap12(ActivityThread.java)
11-27 09:43:59.029 20336-20336/? W/System.err:     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1478)
11-27 09:43:59.029 20336-20336/? W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:102)
11-27 09:43:59.029 20336-20336/? W/System.err:     at android.os.Looper.loop(Looper.java:154)
11-27 09:43:59.029 20336-20336/? W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:6121)
11-27 09:43:59.029 20336-20336/? W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
11-27 09:43:59.029 20336-20336/? W/System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
11-27 09:43:59.029 20336-20336/? W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)
11-27 09:43:59.035 20336-20368/? W/System: ClassLoader referenced unknown path: /data/user_de/0/com.google.android.gms/app_chimera/m/0000000e/n/arm64-v8a
```

uKL, 2017-11-27:
Please share:

`locationManager.isProviderEnabled(LocationManager.NETWORK_PROVIDER);`
`locationManager.isProviderEnabled(LocationManager.GPS_PROVIDER);`

Also, what is your target SDK?

ejbp, 2017-11-27:
Configuration:

```
compileSdkVersion 26
    //buildToolsVersion '26.0.2'
    buildToolsVersion '26.0.2'
    flavorDimensions "default"
    defaultConfig {
        applicationId "com.xxx"
        minSdkVersion 21
        targetSdkVersion 26
        versionCode 0
        versionName "0.2.8"
        multiDexEnabled true
    }
```

note: I've recently upgraded to 26 but I've tested before with 25 which was giving the same error.

I've changed the request permission from ACCESS_COARSE_LOCATION to ACCESS_FINE_LOCATION but it stills gives the same error.

```
Log.d("BluetoothCommManager", "Providers: " + locationManager.getAllProviders());
Log.d("BluetoothCommManager", "isProviderEnabled (Network): " + locationManager.isProviderEnabled(LocationManager.NETWORK_PROVIDER));
Log.d("BluetoothCommManager", "isProviderEnabled (GPS): " + locationManager.isProviderEnabled(LocationManager.GPS_PROVIDER));

11-27 10:00:25.993 21028-21028/? D/BluetoothCommManager: Providers: [passive, network]
11-27 10:00:25.994 21028-21028/? D/BluetoothCommManager: isProviderEnabled (Network): false
11-27 10:00:25.994 21028-21028/? D/BluetoothCommManager: isProviderEnabled (GPS): false
```

```
11-27 10:00:26.003 21028-21028/? D/BluetoothComm: Bluetooth Scan Error: 
11-27 10:00:26.003 21028-21028/? W/System.err: com.polidea.rxandroidble.exceptions.BleScanException: Location Services disabled (code 4)
11-27 10:00:26.003 21028-21028/? W/System.err:     at com.polidea.rxandroidble.internal.scan.ScanPreconditionsVerifierApi18.verify(ScanPreconditionsVerifierApi18.java:31)
11-27 10:00:26.003 21028-21028/? W/System.err:     at com.polidea.rxandroidble.internal.scan.ScanPreconditionsVerifierApi24.verify(ScanPreconditionsVerifierApi24.java:38)
11-27 10:00:26.003 21028-21028/? W/System.err:     at com.polidea.rxandroidble.RxBleClientImpl$1.call(RxBleClientImpl.java:114)
```

[image]
[image]

dariuszseweryn, 2017-11-27:
In the case of this dongle none of the network or gps providers seems to be available. Could you check for the fused and passive?
```
    /**
     * Name of the network location provider.
     * <p>This provider determines location based on
     * availability of cell tower and WiFi access points. Results are retrieved
     * by means of a network lookup.
     */
    public static final String NETWORK_PROVIDER = "network";

    /**
     * Name of the GPS location provider.
     *
     * <p>This provider determines location using
     * satellites. Depending on conditions, this provider may take a while to return
     * a location fix. Requires the permission
     * {@link android.Manifest.permission#ACCESS_FINE_LOCATION}.
     *
     * <p> The extras Bundle for the GPS location provider can contain the
     * following key/value pairs:
     * <ul>
     * <li> satellites - the number of satellites used to derive the fix
     * </ul>
     */
    public static final String GPS_PROVIDER = "gps";

    /**
     * A special location provider for receiving locations without actually initiating
     * a location fix.
     *
     * <p>This provider can be used to passively receive location updates
     * when other applications or services request them without actually requesting
     * the locations yourself.  This provider will return locations generated by other
     * providers.  You can query the {@link Location#getProvider()} method to determine
     * the origin of the location update. Requires the permission
     * {@link android.Manifest.permission#ACCESS_FINE_LOCATION}, although if the GPS is
     * not enabled this provider might only return coarse fixes.
     */
    public static final String PASSIVE_PROVIDER = "passive";

    /**
     * Name of the Fused location provider.
     *
     * <p>This provider combines inputs for all possible location sources
     * to provide the best possible Location fix. It is implicitly
     * used for all API's that involve the {@link LocationRequest}
     * object.
     *
     * @hide
     */
    public static final String FUSED_PROVIDER = "fused";
```

uKL, 2017-11-27:
It's strange because in the menu you have "Device only" which corresponds to GPS, not the network.

In available providers though, you have only "passive" and "network".

What are your options when you press "MODE"?

ejbp, 2017-11-27:
Ok, so without any change on the configuration the result of:

 ```
Log.d("BluetoothCommManager", "Providers: " + locationManager.getAllProviders());
Log.d("BluetoothCommManager", "isProviderEnabled (Network): " + locationManager.isProviderEnabled(LocationManager.NETWORK_PROVIDER));
Log.d("BluetoothCommManager", "isProviderEnabled (GPS): " + locationManager.isProviderEnabled(LocationManager.GPS_PROVIDER));
Log.d("BluetoothCommManager", "isProviderEnabled (PASSIVE_PROVIDER): " + locationManager.isProviderEnabled(LocationManager.PASSIVE_PROVIDER));
Log.d("BluetoothCommManager", "isProviderEnabled (FUSED_PROVIDER): " + locationManager.isProviderEnabled("fused"));

```

is

```
11-27 10:21:14.334 21756-21756/? D/BluetoothCommManager: Providers: [passive, network]
11-27 10:21:14.335 21756-21756/? D/BluetoothCommManager: isProviderEnabled (Network): false
11-27 10:21:14.335 21756-21756/? D/BluetoothCommManager: isProviderEnabled (GPS): false
11-27 10:21:14.336 21756-21756/? D/BluetoothCommManager: isProviderEnabled (PASSIVE_PROVIDER): true
11-27 10:21:14.336 21756-21756/? D/BluetoothCommManager: isProviderEnabled (FUSED_PROVIDER): false
```

Now, changing mode from device (which was the default) to High Accuracy, the result is:

```
11-27 10:32:15.011 22157-22157/? D/BluetoothCommManager: Providers: [passive, network]
11-27 10:32:15.012 22157-22157/? D/BluetoothCommManager: isProviderEnabled (Network): true
11-27 10:32:15.012 22157-22157/? D/BluetoothCommManager: isProviderEnabled (GPS): false
11-27 10:32:15.012 22157-22157/? D/BluetoothCommManager: isProviderEnabled (PASSIVE_PROVIDER): true
11-27 10:32:15.013 22157-22157/? D/BluetoothCommManager: isProviderEnabled (FUSED_PROVIDER): false
```

And now it's working, meaning that I receive the ScanResults. 

My question now is if this condition (about the providers) is really necessary cause, if permission ACCESS_COARSE_LOCATION is allowed, even with mode in "device" mode, the scan works. I know this cause I've tested to do it directly (with the android libs). This is will force to request the user to change this, which is one more barrier to use the app (or even not be able to use if it's a basic user).

Looking forward for your thoughts. Anyway, thank you very much for your help and the Lib :).

uKL, 2017-11-27:
It looks like your provider settings are reversed.
* "High accuracy" should enable GPS
* "Device only" should enable coarse location (network).

Enabling location is required only in certain scenarios where:
* You are not an Android Wear device
* Your target and device SDK is >= Android M

This is to control user's privacy better (you're getting device MAC address, which you can use as a location tip).

ejbp, 2017-11-27:
"This is to control user's privacy better (you're getting device MAC address, which you can use as a location tip)."

Yes, I understand it. What I'm saying is that, using android libs, when I'm in device mode I also receive the ScanResult (Name of device + Mac address).  

But, you're probably right, it may be a device provider issue on this.

Thanks for your help.

Best

uKL, 2017-11-27:
This is to be considered if verification should be a configuration parameter for a scan. @dariuszseweryn What do you think?

dariuszseweryn, 2017-11-27:
That is a good idea.
I think that we should also extend `CheckerLocationProvider` with the `LocationManager.PASSIVE_PROVIDER` check. It is not documented anywhere but so far it seems that it does not matter which provider is available if the user switched on `Location Services`.

uKL, 2017-11-27:
Not sure, AFAIK you cannot disable "passive" at all from a menu setting.

dariuszseweryn, 2017-11-27:
@ejbp Could you check the above?
Could you run all location provider checks when the `Location Services` setting if off?

ejbp, 2017-11-27:
[image]

Result:

```

Log.d("BluetoothCommManager", "Providers: " + locationManager.getAllProviders());
Log.d("BluetoothCommManager", "isProviderEnabled (Network): " + locationManager.isProviderEnabled(LocationManager.NETWORK_PROVIDER));
Log.d("BluetoothCommManager", "isProviderEnabled (GPS): " + locationManager.isProviderEnabled(LocationManager.GPS_PROVIDER));
Log.d("BluetoothCommManager", "isProviderEnabled (PASSIVE_PROVIDER): " + locationManager.isProviderEnabled(LocationManager.PASSIVE_PROVIDER));
Log.d("BluetoothCommManager", "isProviderEnabled (FUSED_PROVIDER): " + locationManager.isProviderEnabled("fused"));

11-27 11:27:42.452 24704-24704/? D/BluetoothCommManager: Providers: [passive, network]
11-27 11:27:42.453 24704-24704/? D/BluetoothCommManager: isProviderEnabled (Network): false
11-27 11:27:42.453 24704-24704/? D/BluetoothCommManager: isProviderEnabled (GPS): false
11-27 11:27:42.454 24704-24704/? D/BluetoothCommManager: isProviderEnabled (PASSIVE_PROVIDER): true
11-27 11:27:42.454 24704-24704/? D/BluetoothCommManager: isProviderEnabled (FUSED_PROVIDER): false
11-27 11:27:42.459 24704-24704/? D/BluetoothComm: Bluetooth Scan Error: 
11-27 11:27:42.459 24704-24704/? W/System.err: com.polidea.rxandroidble.exceptions.BleScanException: Location Services disabled (code 4)
11-27 11:27:42.460 24704-24704/? W/System.err:     at com.polidea.rxandroidble.internal.scan.ScanPreconditionsVerifierApi18.verify(ScanPreconditionsVerifierApi18.java:31)
11-27 11:27:42.460 24704-24704/? W/System.err:     at com.polidea.rxandroidble.internal.scan.ScanPreconditionsVerifierApi24.verify(ScanPreconditionsVerifierApi24.java:38)
11-27 11:27:42.460 24704-24704/? W/System.err:     at com.polidea.rxandroidble.RxBleClientImpl$1.call(RxBleClientImpl.java:114)
11-27 11:27:42.460 24704-24704/? W/System.err:     at com.polidea.rxandroidble.RxBleClientImpl$1.call(RxBleClientImpl.java:111)
11-27 11:27:42.460 24704-24704/? W/Syste
```

dariuszseweryn, 2017-12-28:
Hello @ejbp 

I have another trace on what could work for your case. Would you be so kind and check what does 
`Settings.Secure.getInt(context.getContentResolver(), Settings.Secure.LOCATION_MODE)` return when your location is being switched on and off?

dariuszseweryn, 2018-01-04:
@ejbp Ping? :)
The above info could potentially fix your use case but I would need to confirm that it works as I assume and I do not have access to the same Android Pen.

P.S. Happy New Year!

ejbp, 2018-01-04:
Sorry Darius. I was off during some days and I missed your post. I will try to answer asap.

dariuszseweryn, 2018-01-05:
You have written that you have three different values. Which one corresponds to when the `Location` ([as shown is this picture](https://github.com/Polidea/RxAndroidBle/issues/327#issuecomment-347155184)) is switched `on` and which for `off`?

ejbp, 2018-01-05:
##When location is on && setting is Device Only
Debugger for Settings.Secure.getInt(context.getContentResolver(), Settings.Secure.LOCATION_MODE): value = 1

[image]

##When location is on && setting is Battery Saving
Debugger for Settings.Secure.getInt(context.getContentResolver(), Settings.Secure.LOCATION_MODE): value = 2

[image]

##When location is on && setting is High Accuracy
Debugger for Settings.Secure.getInt(context.getContentResolver(), Settings.Secure.LOCATION_MODE): value = 3

[image]

##When location is off
Debugger for Settings.Secure.getInt(context.getContentResolver(), Settings.Secure.LOCATION_MODE): value = 0

[image]

dariuszseweryn, 2018-01-05:
I think we have a winner. Could you just repeat the below test:
```
Log.d("BluetoothCommManager", "Providers: " + locationManager.getAllProviders());
Log.d("BluetoothCommManager", "isProviderEnabled (Network): " + locationManager.isProviderEnabled(LocationManager.NETWORK_PROVIDER));
Log.d("BluetoothCommManager", "isProviderEnabled (GPS): " + locationManager.isProviderEnabled(LocationManager.GPS_PROVIDER));
Log.d("BluetoothCommManager", "isProviderEnabled (PASSIVE_PROVIDER): " + locationManager.isProviderEnabled(LocationManager.PASSIVE_PROVIDER));
Log.d("BluetoothCommManager", "isProviderEnabled (FUSED_PROVIDER): " + locationManager.isProviderEnabled("fused"));
```
for all cases from your above post? I would like to have those for comparison. Thank you 👍

ejbp, 2018-01-05:
```
Settings.Secure.getInt(context.getContentResolver(), Settings.Secure.LOCATION_MODE): 1
--------------
Providers: [passive, network]
isProviderEnabled (Network): false
isProviderEnabled (GPS): false
isProviderEnabled (PASSIVE_PROVIDER): true
isProviderEnabled (FUSED_PROVIDER): false
```

```
Settings.Secure.getInt(context.getContentResolver(), Settings.Secure.LOCATION_MODE): 2
--------------
Providers: [passive, network]
isProviderEnabled (Network): true
isProviderEnabled (GPS): false
isProviderEnabled (PASSIVE_PROVIDER): true
isProviderEnabled (FUSED_PROVIDER): false
```

```
Settings.Secure.getInt(context.getContentResolver(), Settings.Secure.LOCATION_MODE): 3
--------------
Providers: [passive, network]
sProviderEnabled (Network): true
isProviderEnabled (GPS): false
isProviderEnabled (PASSIVE_PROVIDER): true
isProviderEnabled (FUSED_PROVIDER): false
```

```
Settings.Secure.getInt(context.getContentResolver(), Settings.Secure.LOCATION_MODE): 0
--------------
Providers: [passive, network]
isProviderEnabled (Network): false
isProviderEnabled (GPS): false
isProviderEnabled (PASSIVE_PROVIDER): true
isProviderEnabled (FUSED_PROVIDER): false
```

dariuszseweryn, 2018-01-05:
Perfect! Thank you—your use-case should be addressed in `1.5.0` release :) 👍

dariuszseweryn, 2018-01-10:
@ejbp this is already fixed in the `1.5.0-SNAPSHOT`. Thank you for the help!

closed 2018-01-10 by dariuszseweryn

ejbp, 2018-01-10:
Thanks @dariuszseweryn . You're guys are the best. Looking forward to have the release notes and available to import into gradle.

dariuszseweryn, 2018-01-10:
The official release should happen quite soon. In the meantime you can checkout the [`1.5.0-SNAPSHOT`](https://github.com/Polidea/RxAndroidBle#snapshot)

commit 5b547272 (2018-01-10): Changed resolving Location Services status (closes #327) (#350)

As it turned out it is possible to detect what is the current status of the Location even on devices that are not phones—without checking location providers just with the location mode.