dariuszseweryn/RxAndroidBle #328: Checkout project from GitHub not working in Android Studio 3

RobLewis, 2017-11-27:
### Summary
Creating a new Android Studio 3 project from GitHub site doesn't work

### Library version
`1.4.3`

#### Preconditions
Android Studio 3.0.1

#### Steps to reproduce actual result
<br />1. In the "Welcome to Android Studio" window, select "check out project from Version Control", selecting GitHub
<br />2. For the Git Repository URL, use "https://github.com/Polidea/RxAndroidBle.git" (Test button reports success) 
<br />3. Click "Clone" and select "Yes" to creating a studio project 
<br />4. In the "Import Project" window, leave "Import project from external model" selected, and leave the first "Grade" selected in the window (there are 2 identical listings for "Gradle") 
<br />5. Click "Next", and in the next window, leave the recommended "Use default Gradle wrapper" selected; click "Finish" 
<br />6. Android Studio starts creating a project
#### Minimum code snippet reproducing the issue
N/A
#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`
N/A
#### Actual result
A message window appears: "The modules below are not imported from Gradle anymore. Check those to be removed from the ide project too:" (only RxAndroidBle is shown, with its box checked)
#### Expected result
A working project is created.

oerjanti, 2017-11-28:
After updating Studio from 2.x to 3.0.1 I got it working by updating Gradle version manually in:
File > Project Structure > Project
Gradle version: 4.3
Android plugin version: 3.0.1

I did not get the error about module import but I hope this helps

dariuszseweryn, 2017-12-04:
Closed by mistake

uKL, 2017-12-15:
Thanks for a very descriptive issue.

I've seen the same issue but after I press "Synchronize with Gradle" the problem has disappeared.

It looks like Android Studio has some problems importing projects from repo directly when `rootProject.name = 'rxandroidble-parent'` is specified in `settings.gradle`.

It will work perfectly if you check out your project to a `rxandroidble-parent` directory.

... or just synchronize your project with Gradle again after accepting the dialog.

closed 2017-12-15 by uKL

uKL, 2017-12-15:
Related Android Studio issue created https://issuetracker.google.com/issues/70710141

RobLewis, 2018-03-14:
Update: tried cloning again to a new Android Studio project, with the just-released 1.5.0 version and first got the message "Unsupported Modules Detected: Compilation is not supported for following modules: RxAndroidBle2. Unfortunately you can't have non-Gradle Java modules and Android-Gradle modules in one project." 
Then it logged "Gradle sync started", and a minute later "Project setup started". Then Gradle sync failed with a NPE. Then I tried a Clean and it synced a few more times and eventually seems to have resolved whatever difficulties it had.

irshadbabar, 2018-08-15:
Try to see if you haven't installed the git on your PC. If not then install that would probably resolve your problem.

Afaquejaya, 2019-02-27:
> ### Summary
> 
> Creating a new Android Studio 3 project from GitHub site doesn't work
> ### Library version
> 
> `1.4.3`
> #### Preconditions
> 
> Android Studio 3.0.1
> #### Steps to reproduce actual result
> 
> 
> 1. In the "Welcome to Android Studio" window, select "check out project from Version Control", selecting GitHub
> 
> 2. For the Git Repository URL, use "https://github.com/Polidea/RxAndroidBle.git" (Test button reports success)
> 
> 3. Click "Clone" and select "Yes" to creating a studio project
> 
> 4. In the "Import Project" window, leave "Import project from external model" selected, and leave the first "Gradle" selected in the window (there are 2 identical listings for "Gradle")
> 
> 5. Click "Next", and in the next window, leave the recommended "Use default Gradle wrapper" selected; click "Finish"
> 
> 6. Android Studio starts creating a project
> #### Minimum code snippet reproducing the issue
> 
> N/A
> #### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`
> 
> N/A
> #### Actual result
> 
> A message window appears: "The modules below are not imported from Gradle anymore. Check those to be removed from the ide project too:" (only RxAndroidBle is shown, with its box checked)
> #### Expected result
> 
> A working project is created.

Unable to get Gradle wrapper properties from:
/root/Desktop/Android-App-for-e-notice-board/gradle/wrapper/gradle-wrapper.properties

edgarolijar, 2020-10-31:
> Try to see if you haven't installed the git on your PC. If not then install that would probably resolve your problem.

Maybe he didn't install OS on his PC?

PR #332 (2017-12-04): Added Hot/Cold type to Observable behaviour in Readme
Closes #328
commit bbdac429 (2017-12-04): Added Hot/Cold type to Observable behaviour to Readme (#332)

Closes #328