dariuszseweryn/RxAndroidBle #331: setupNotification cause extra memory allocation

andyalbert, 2017-11-30:
I've noticed a strange behaviour in android profiler after registering for notifications, the java memory keeps increasing so fast allocating hundreds of objects per second (it reached 120 thousand in 3 minutes)!
I've attached an image for heap dump
[image]

dariuszseweryn, 2017-11-30:
Hello,
Do you say that the allocations are strange or the memory allocated is not released?

andyalbert, 2017-11-30:
I mean that the memory allocation keeps increasing till it forces a GC action, is that usual ?

passsy, 2017-11-30:
That's a normal behavior of the JVM. It's not possible to release objects without the GC.

dariuszseweryn, 2017-11-30:
Well, yes- each notification received is packed into an object and emitted. If those objects are deallocated by the GC then everything is ok.

andyalbert, 2017-11-30:
Hmm, so that's just a usual behaviour as the ble device is sending a lot of data, that describes the problem then, Thanks! , but regarding the notification itself, why is it passing `Observable<Observable<byte[]>>` instead of just `Observable<byte[]>`? I thought either the notification should send a stream of data, or either the data(`byte[]`) itself be streamed, but both?

dariuszseweryn, 2017-11-30:
From the Javadoc of `RxBleConnection.setupNotification()`:
```
     * Setup characteristic notification in order to receive callbacks when given characteristic has been changed. Returned observable will
     * emit Observable<byte[]> once the notification setup has been completed. 
```
This means that the emitted `Observable<byte[]>` is an object that will stream the notification byte arrays. But setting up the notification is also an asynchronous process and takes time.

andyalbert, 2017-11-30:
Thanks a lot, it's all clear now.

closed 2017-11-30 by andyalbert