dariuszseweryn/RxAndroidBle #334: Listening to click events from bt keyring.

Neogene, 2017-12-02:
Sorry to bother, I didn't understand reading the docs how to listen to click events sent from a bt device, after connecting to a device to listen to click events (from a bt single key keyring), do I have to:

1) Establish connection with the device after finding using scan devices (RxBleDevice device = scanResult.getBleDevice();
2) doing a FLATMAP for discover services
3) in its .SUBSCRIBE(discoveryResult ->) iterate into services to find the proper characteristic (UUID:0000ffe1-0000-1000-8000-00805f9b34fb)
4) when I find proper characteristic store into a temp var
5) DISCONNECT from device
6)  RECONNECT and this time use setupnotification and listen to that characteristic on DOONNEXT.

if it's correct I receive the following error:

error:com.polidea.rxandroidble.exceptions.BleCannotSetCharacteristicNotificationException: Cannot find client characteristic config descriptor (code 2) with characteristic UUID 0000ffe1-0000-1000-8000-00805f9b34fb

debug log level verbose:

`
201 3981-3981/? I/art: Late-enabling -Xcheck:jni
504 3981-3981/it.leganza.test W/System: ClassLoader referenced unknown path: /data/app/it.leganza.test-1/lib/arm
516 3981-3981/it.leganza.test I/InstantRun: starting instant run server: is main process
612 3981-3981/it.leganza.test W/art: Before Android 4.1, method android.graphics.PorterDuffColorFilter android.support.graphics.drawable.VectorDrawableCompat.updateTintFilter(android.graphics.PorterDuffColorFilter, android.content.res.ColorStateList, android.graphics.PorterDuff$Mode) would have incorrectly overridden the package-private method in android.graphics.drawable.Drawable
.021 3981-4022/it.leganza.test I/Adreno-EGL: <qeglDrvAPI_eglInitialize:379>: EGL 1.4 QUALCOMM build: Nondeterministic_AU_msm8974_LA.BF.1.1.3_RB1__release_AU (I741a3d36ca)
                                                                    OpenGL ES Shader Compiler Version: E031.29.00.00
                                                                    Build Date: 04/04/16 Mon
                                                                    Local Branch: mybranch19053788
                                                                    Remote Branch: quic/LA.BF.1.1.3_rb1.12
                                                                    Local Patches: NONE
                                                                    Reconstruct Branch: NOTHING
.026 3981-4022/it.leganza.test I/OpenGLRenderer: Initialized EGL, version 1.4
.026 3981-4022/it.leganza.test D/OpenGLRenderer: Swap behavior 1
.029 3981-4022/it.leganza.test W/Adreno-ES20: <get_gpu_clk:229>: open failed: errno 13
.109 3981-3981/it.leganza.test W/art: Before Android 4.1, method int android.support.v7.widget.ListViewCompat.lookForSelectablePosition(int, boolean) would have incorrectly overridden the package-private method in android.widget.ListView
.302 3981-3981/it.leganza.test D/ClickManager: ClickManager: initialized.
.328 3981-3981/it.leganza.test D/RxBle#ClientOperationQueue:   QUEUED ScanOperationApi21(29872811)
.328 3981-4027/it.leganza.test D/RxBle#ClientOperationQueue:  STARTED ScanOperationApi21(29872811)
.338 3981-4029/it.leganza.test I/RxBle#QueueOperation: Scan operation is requested to start.
.364 3981-3996/it.leganza.test D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=6 mClientIf=0
.372 3981-4027/it.leganza.test D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(29872811)
.561 3981-3981/it.leganza.test D/ClickManager: startScan() scanResult called: ScanResult{bleDevice=RxBleDeviceImpl{bluetoothDevice=Click Device(FF:FF:70:02:6A:EE)}, rssi=-75, timestampNanos=87366398304967, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=com.polidea.rxandroidble.internal.scan.ScanRecordImplNativeWrapper@a8cb34c}
.563 3981-3981/it.leganza.test D/ClickManager: startScan scanResult FOUND DEVICE: RxBleDeviceImpl{bluetoothDevice=Click Device(FF:FF:70:02:6A:EE)}
.565 3981-4029/it.leganza.test I/RxBle#CancellableSubscription: Scan operation is requested to stop.
.630 3981-4057/it.leganza.test D/RxBle#ClientOperationQueue:   QUEUED ConnectOperation(168559243)
.631 3981-4027/it.leganza.test D/RxBle#ClientOperationQueue:  STARTED ConnectOperation(168559243)
.636 3981-4029/it.leganza.test V/RxBle#BleConnectionCompat: Connecting without reflection
.637 3981-4029/it.leganza.test D/BluetoothGatt: connect() - device: FF:FF:70:02:6A:EE, auto: true
.637 3981-4029/it.leganza.test D/BluetoothGatt: registerApp()
.637 3981-4029/it.leganza.test D/BluetoothGatt: registerApp() - UUID=810c22af-2ad7-44f0-9d4f-956bbde01037
.638 3981-3996/it.leganza.test D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
.640 3981-4027/it.leganza.test D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(168559243)
:21.665 3981-3996/it.leganza.test D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=FF:FF:70:02:6A:EE
:21.667 3981-3996/it.leganza.test D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
:21.697 3981-4057/it.leganza.test D/RxBle#ConnectionOperationQueue:   QUEUED ServiceDiscoveryOperation(2380127)
:21.698 3981-4056/it.leganza.test D/RxBle#ConnectionOperationQueue:  STARTED ServiceDiscoveryOperation(2380127)
:21.703 3981-4029/it.leganza.test D/BluetoothGatt: discoverServices() - device: FF:FF:70:02:6A:EE
:22.253 3981-3997/it.leganza.test D/BluetoothGatt: onSearchComplete() = Device=FF:FF:70:02:6A:EE Status=0
:22.255 3981-3997/it.leganza.test D/RxBle#BluetoothGatt: onServicesDiscovered status=0
:22.263 3981-4057/it.leganza.test D/ClickManager: startScan:discoverServices: com.polidea.rxandroidble.RxBleDeviceServices@18ee52d
:22.263 3981-4057/it.leganza.test D/ClickManager: Service UUID: 00001800-0000-1000-8000-00805f9b34fb
:22.263 3981-4057/it.leganza.test D/ClickManager: ----- characteristic UUID: 00002a00-0000-1000-8000-00805f9b34fb
:22.264 3981-4057/it.leganza.test D/ClickManager: ----- characteristic UUID: 00002a01-0000-1000-8000-00805f9b34fb
:22.265 3981-4057/it.leganza.test D/ClickManager: Service UUID: 0000180f-0000-1000-8000-00805f9b34fb
:22.265 3981-4057/it.leganza.test D/ClickManager: ----- characteristic UUID: 00002a19-0000-1000-8000-00805f9b34fb
:22.265 3981-4057/it.leganza.test D/ClickManager: Service UUID: 00001802-0000-1000-8000-00805f9b34fb
:22.265 3981-4057/it.leganza.test D/ClickManager: ----- characteristic UUID: 00002a06-0000-1000-8000-00805f9b34fb
:22.265 3981-4057/it.leganza.test D/ClickManager: Service UUID: 0000ffe0-0000-1000-8000-00805f9b34fb
:22.266 3981-4057/it.leganza.test D/ClickManager: ----- characteristic UUID: 0000ffe1-0000-1000-8000-00805f9b34fb
:22.266 3981-4057/it.leganza.test D/ClickManager: startScan: found characteristic: android.bluetooth.BluetoothGattCharacteristic@dcf27f3
:22.286 3981-4056/it.leganza.test D/RxBle#ConnectionOperationQueue: FINISHED ServiceDiscoveryOperation(2380127)
:22.289 3981-4057/it.leganza.test I/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (FF:FF:70:02:6A:EE)
:22.290 3981-4056/it.leganza.test D/RxBle#Executors$RunnableAdapter: Terminated.
:22.292 3981-4057/it.leganza.test D/RxBle#ClientOperationQueue:   QUEUED DisconnectOperation(174428550)
[... 10 lines cut ...]
:22.308 3981-4027/it.leganza.test D/RxBle#ClientOperationQueue:  STARTED ConnectOperation(104619256)
:22.312 3981-4029/it.leganza.test V/RxBle#BleConnectionCompat: Connecting without reflection
:22.313 3981-4029/it.leganza.test D/BluetoothGatt: connect() - device: FF:FF:70:02:6A:EE, auto: false
:22.313 3981-4029/it.leganza.test D/BluetoothGatt: registerApp()
:22.313 3981-4029/it.leganza.test D/BluetoothGatt: registerApp() - UUID=cf9199b0-cbf2-440b-92fe-40b111498d70
:22.314 3981-3997/it.leganza.test D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
:22.316 3981-3996/it.leganza.test D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=FF:FF:70:02:6A:EE
:22.316 3981-3996/it.leganza.test D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
:22.319 3981-4027/it.leganza.test D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(104619256)
:22.334 3981-4057/it.leganza.test D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000ffe1-0000-1000-8000-00805f9b34fb enable: true
:22.336 3981-4057/it.leganza.test D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000ffe1-0000-1000-8000-00805f9b34fb enable: false
[... cut ...]
:22.347 3981-4029/it.leganza.test D/BluetoothGatt: cancelOpen() - device: FF:FF:70:02:6A:EE
:22.348 3981-3997/it.leganza.test D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=FF:FF:70:02:6A:EE
:22.348 3981-3997/it.leganza.test D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=0
:22.349 3981-4029/it.leganza.test D/BluetoothGatt: close()
:22.349 3981-4029/it.leganza.test D/BluetoothGatt: unregisterApp() - mClientIf=6
:22.351 3981-4027/it.leganza.test D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(104934100)
`

Neogene, 2017-12-02:
Update, hard coding service and characteristic uuid reduces lines of code but there is no way to listen to notification, it reports always missing the descriptor (analyzing the characteristic there are no descriptor, but using another github project, itrack2, it manages to connect without descriptor without any problem).

`
deviceSubscription = device.establishConnection(false)
                    .flatMap(rxBleConnection -> rxBleConnection.discoverServices())
                    .subscribe(
                            discoveryResult -> {
                                Log.d(TAG, "startScan:discoverServices: "+discoveryResult);

                                discoveryResult.getService(serviceUUID).subscribe(
                                        serviceValue -> {
                                            // Read characteristic value.
                                            Log.d(TAG, "startScan: chars:"+serviceValue);

                                            BluetoothGattCharacteristic characteristic = serviceValue.getCharacteristic(clickCharacteristicUUID);

                                            deviceSubscription.unsubscribe();

                                            deviceSubscription = device.establishConnection(false)
                                                    .flatMap(rxBleConnection -> rxBleConnection.setupNotification(characteristic))
                                                    .doOnNext(notificationObservable -> {
                                                        // Notification has been set up
                                                        Log.d(TAG, "startScan: "+notificationObservable);
                                                    })
                                                    .flatMap(notificationObservable -> notificationObservable) // <-- Notification has been set up, now observe value changes.
                                                    .subscribe(
                                                            bytes -> {
                                                                Log.d(TAG, "startScan: "+bytes);
                                                                // Given characteristic has been changes, here is the value.
                                                            },
                                                            throwable -> {
                                                                Log.d(TAG, "startScan: error:"+throwable);
                                                                // Handle an error here.
                                                            }
                                                    );

                                        },
                                        throwable -> {
                                            // Handle an error here.
                                            Log.d(TAG, "startScan: chars ERROR"+throwable);
                                        }
                                );
`

Error received:

`

12-02 09:45:16.288 3520-3564/it.hartstudio.mcdelivery D/BluetoothGatt: registerApp() - UUID=23fc0b56-36e0-4820-b4f3-f0be67698ed2
12-02 09:45:16.290 3520-3542/it.hartstudio.mcdelivery D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
12-02 09:45:16.292 3520-3541/it.hartstudio.mcdelivery D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=FF:FF:70:02:6A:EE
12-02 09:45:16.292 3520-3541/it.hartstudio.mcdelivery D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
12-02 09:45:16.295 3520-3562/it.hartstudio.mcdelivery D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(104619256)
12-02 09:45:16.319 3520-3582/it.hartstudio.mcdelivery D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000ffe1-0000-1000-8000-00805f9b34fb enable: true
12-02 09:45:16.322 3520-3582/it.hartstudio.mcdelivery D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000ffe1-0000-1000-8000-00805f9b34fb enable: false
12-02 09:45:16.324 3520-3582/it.hartstudio.mcdelivery D/ClickManager: startScan: error:com.polidea.rxandroidble.exceptions.BleCannotSetCharacteristicNotificationException: Cannot find client characteristic config descriptor (code 2) with characteristic UUID 0000ffe1-0000-1000-8000-00805f9b34fb
`

dariuszseweryn, 2017-12-02:
Hello!

Thank you for using the library.

This is a general usage question and as such should be posted on [stackoverflow](https://stackoverflow.com/questions/tagged/rxandroidble) with `rxandroidble` tag—it will be easier to find by other users.

You should also search for `NotificationSetupMode` and related overload of `setupNotification()` function.

Best Regards

closed 2017-12-02 by dariuszseweryn

Neogene, 2017-12-02:
I found another thread/bug here which reported the problem of missing descriptor, seems not so uncommon to have devices which miss this information.

dariuszseweryn, 2017-12-02:
This issue is solved since version `1.2.0`. I have pointed you where is the solution you should look for. It is documented. What are your expectations?