dariuszseweryn/RxAndroidBle #337: advertising scan periodical

GuillaumeBo, 2017-12-13:
### Summary
Hello,
I need to scan ble advertising permanently while i am not connected to my device to find it and know when connect to it (specific product). This scan is realized in foreground service to match 8.0 prerequisite.

To preserve a few battery, i want scan periodically (while respect the Android 7 limit of start amount/time). I see 2 differents implementations :  
 - Start the scan each X seconds and be able to stop it after Y seconds, the wait time will be X-Y.

‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌Scan‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ Wait‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ Scan‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌Wait
|-------------|------|-------------|------|–––> 

|---------X-------->|---------X-------->|-----

|-----Y------>‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ |------Y----->‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌|----

 - Start the scan for X seconds, wait for Y seconds, start the scan for X seconds etc..

‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌Scan‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ Wait‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌ ‌‌ ‌‌ ‌‌ Scan‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌ ‌‌Wait
|--------------|-------|--------------|------|–––> 

|------X----->|--Y-->|------X------>|--Y-->|---

I want to it with Rx style and not with simple timer (i am not an RX expert). I don't know what solution is better/easier to be develop.
This must be transparent for client and must keep the original RxAndroidBle working, subscribe = start the process and unsubscribe = stop the process.

#### Minimum code snippet reproducing the issue
I suppose i have to use "Obersable.interval" but i don't see how to manage this
Any snippet/exemple ?

dariuszseweryn, 2017-12-13:
Hello!

Thank you for using the library.

This is a general usage question and as such should be posted on [stackoverflow](https://stackoverflow.com/questions/tagged/rxandroidble) with `rxandroidble` tag—it will be easier to find by other users.

Best Regards

closed 2017-12-13 by dariuszseweryn

GuillaumeBo, 2017-12-15:
Just for information if anybody needs help about this , thread and answer here : [stackoverflow](https://stackoverflow.com/questions/47790910/rxandroidble-periodic-advertising-scan)