dariuszseweryn/RxAndroidBle #338: unable to parse scan record

yusufonderd, 2017-12-15:
I'm getting unable to parse scan record exception for ble devices and manufacturer data getting is null.

dariuszseweryn, 2017-12-15:
Hello @yusufonderd 

I would like you to give more information about this issue. Preferably conforming to the template. Logs, version of the library is a must—without it I cannot do anything.

Best Regards

dariuszseweryn, 2017-12-27:
Closing due to lack of information and inactivity of the original poster.

When more data will be available—feel free to add that information and reopen this issue.

Best Regards

closed 2017-12-27 by dariuszseweryn

yusufonderd, 2018-01-08:
Hi again. 
I'm facing this issue on version 1.4.3.

        ```
scanSubscription = rxClient.scanBleDevices(scanSettings)
                .subscribe({
                  val manufacturerData =   it.scanRecord.manufacturerSpecificData
                  if(manufacturerData == null){
                          Log.i(TAG,"manufacturerSpecificData is null")
                 }
                 },{
                     it.printStackTrace();
                 })
```
Logs: 

 BluetoothScanningService :  manufacturer data is null

dariuszseweryn, 2018-01-08:
Hello again. We need logs from the library. `RxBleLog.setLogLevel(RxBleLog.VERBOSE)` and logs from the application. Raw bytes of the advertisement could also help to investigate if there was `unable to parse scan record`

smithc42, 2019-10-03:
I think I'm seeing a similar issue. Manufacturer data in the ScanResult is null when I use `CALLBACK_TYPE_FIRST_MATCH` or `CALLBACK_TYPE_FIRST_MATCH or CALLBACK_TYPE_MATCH_LOST`

Manufacturer data is present when using `CALLBACK_TYPE_ALL_MATCHES`

**Library version**
`1.10.1`

**Preconditions**

1. Start scan with scan type call back: `CALLBACK_TYPE_FIRST_MATCH or CALLBACK_TYPE_MATCH_LOST` 
2. Filter by UUID

**Result when using `CALLBACK_TYPE_ALL_MATCHES`:**
```
D/RxBle#ClientOperationQueue: QUEUED   ScanOperationApi21(75312048)
D/RxBle#ClientOperationQueue: STARTED  ScanOperationApi21(75312048)
I/RxBle#ClientOperationQueue: RUNNING  ScanOperationApi21{ANY_MUST_MATCH -> nativeFilters=[BluetoothLeScanFilter [mDeviceName=null, MAC=null, mUuid=6e400001-b5a3-f393-e0a9-e50e24dcca9e, mUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=-1, mManufacturerData=null, mManufacturerDataMask=null]]}
D/RxBle#ScanOperationApi21: No library side filtering —> debug logs of scanned devices disabled
D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(75312048) in 9 ms
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='D6:EF:E5:0B:A3:6D', name=TestDevice}, rssi=-60, timestampNanos=349098893842907, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[03, 19, C0, 00, 02, 01, 06, 11, 09, 42, 75, 7A, 7A, 44, 36, 45, 46, 45, 35, 30, 42, 41, 33, 36, 44, 03, 03, 0A, 18, 11, 07, 9E, CA, DC, 24, 0E, E5, A9, E0, 93, F3, A3, B5, 01, 00, 40, 6E, 08, FF, CF, 07, 42, 7A, 7A, 2A, 04, 00, 00, 00, 00, 00, 00]}
```
**Result when using `CALLBACK_TYPE_FIRST_MATCH or CALLBACK_TYPE_MATCH_LOST`:**
```
CALLBACK_TYPE_FIRST_MATCH
D/RxBle#ClientOperationQueue: QUEUED   ScanOperationApi21(167012866)
D/RxBle#ClientOperationQueue: STARTED  ScanOperationApi21(167012866)
I/RxBle#ClientOperationQueue: RUNNING  ScanOperationApi21{ANY_MUST_MATCH -> nativeFilters=[BluetoothLeScanFilter [mDeviceName=null, MAC=null, mUuid=6e400001-b5a3-f393-e0a9-e50e24dcca9e, mUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=-1, mManufacturerData=null, mManufacturerDataMask=null]]}
I/RxBle#QueueOperation: Scan operation is requested to start.
D/RxBle#ScanOperationApi21: No library side filtering —> debug logs of scanned devices disabled
D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(167012866) in 14 ms
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='D6:EF:E5:0B:A3:6D', name=TestDevice}, rssi=-66, timestampNanos=349821799867788, callbackType=CALLBACK_TYPE_FIRST_MATCH, scanRecord=[03, 19, C0, 00, 02, 01, 06, 11, 09, 42, 75, 7A, 7A, 44, 36, 45, 46, 45, 35, 30, 42, 41, 33, 36, 44, 19, C0, 00]}
```

The scanRecord is significantly shorter in the `CALLBACK_TYPE_FIRST_MATCH or CALLBACK_TYPE_MATCH_LOST` logs.

**Minimum code snippet reproducing the issue:**
```
 val scanSettings = ScanSettings.Builder()
        .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
        .setCallbackType(ScanSettings.CALLBACK_TYPE_FIRST_MATCH or ScanSettings.CALLBACK_TYPE_MATCH_LOST)
        .build()

    val builder = ScanFilter.Builder()

    serviceUUID?.let {
        builder.setServiceUuid(ParcelUuid.fromString(it.uuid))
    }
    return scanBleDevices(scanSettings, builder.build())
```

dariuszseweryn, 2019-10-03:
Hello @smithc42 
This seems interesting. Usually the OS does perform EIR when using `CALLBACK_TYPE_ALL_MATCHES`. It seems that this is not the case in other situations.
Please help me understand the issue better: I wonder what is happening if you would also set the manufacturer ID in the `ScanFilter`

smithc42, 2019-10-04:
Hi @dariuszseweryn, thanks for the reply.

All the below are using the callback: 
- `ScanSettings.CALLBACK_TYPE_FIRST_MATCH or ScanSettings.CALLBACK_TYPE_MATCH_LOST`

**I've tested filtering with `mUuid ` and `mManufacturerId`**
```
D/RxBle#ClientOperationQueue: QUEUED   ScanOperationApi21(230830413)
D/RxBle#ClientOperationQueue: STARTED  ScanOperationApi21(230830413)
I/RxBle#ClientOperationQueue: RUNNING  ScanOperationApi21{ANY_MUST_MATCH -> nativeFilters=[BluetoothLeScanFilter [mDeviceName=null, MAC=null, mUuid=6e400001-b5a3-f393-e0a9-e50e24dcca9e, mUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=1999, mManufacturerData=null, mManufacturerDataMask=null]]}
I/RxBle#QueueOperation: Scan operation is requested to start.
D/RxBle#ScanOperationApi21: No library side filtering —> debug logs of scanned devices disabled
D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(230830413) in 11 ms
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='D6:EF:E5:0B:A3:6D', name=null}, rssi=-62, timestampNanos=599832180999097, callbackType=CALLBACK_TYPE_FIRST_MATCH, scanRecord=[03, 19, C0, 00, 02, 01, 06, 11, 09, 42, 75, 7A, 7A, 44, 36, 45, 46, 45, 35, 30, 42, 41, 33, 36, 44, 19, C0, 00]}
```

The scan record is still missing the manufacturer data. 

**But I also noticed that when the `mUuid` is omitted, filtering by `mManufacturerId` doesn't appear to return any results (maybe I'm using the filter incorrectly?).**
```
D/RxBle#ClientOperationQueue: QUEUED   ScanOperationApi21(99452729)
D/RxBle#ClientOperationQueue: STARTED  ScanOperationApi21(99452729)
I/RxBle#ClientOperationQueue: RUNNING  ScanOperationApi21{ANY_MUST_MATCH -> nativeFilters=[BluetoothLeScanFilter [mDeviceName=null, MAC=null, mUuid=null, mUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=1999, mManufacturerData=null, mManufacturerDataMask=null]]}
I/RxBle#QueueOperation: Scan operation is requested to start.
D/RxBle#ScanOperationApi21: No library side filtering —> debug logs of scanned devices disabled
D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(99452729) in 12 ms
```

**I've also tried filter with `mManufacturerId`, `mManufacturerData` and `mManufacturerDataMask`**
```
D/RxBle#ClientOperationQueue: QUEUED   ScanOperationApi21(88253442)
D/RxBle#ClientOperationQueue: STARTED  ScanOperationApi21(88253442)
I/RxBle#ClientOperationQueue: RUNNING  ScanOperationApi21{ANY_MUST_MATCH -> nativeFilters=[BluetoothLeScanFilter [mDeviceName=null, MAC=null, mUuid=6e400001-b5a3-f393-e0a9-e50e24dcca9e, mUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=1999, mManufacturerData=[52, 50, 55, 97, 55, 97, 55, 48, 50, 52], mManufacturerDataMask=[-1, -1, -1, -1, -1, -1, -1, -1, -1, -1]]]}
I/RxBle#QueueOperation: Scan operation is requested to start.
D/RxBle#ScanOperationApi21: No library side filtering —> debug logs of scanned devices disabled
D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(88253442) in 11 ms
```

**Here's the manufacturer data:**
[image]

dariuszseweryn, 2019-10-25:
It seems that indeed offloaded filtering does not put any data transmitted via Extended Inquiry Response (EIR) and all `0`s are stripped. Additionally it seems that if you would like to use `.setManufacturerData()` you must use `.setManufacturerData(id, byteArrayOf(), byteArrayOf())` if you are about to catch all data that have the `id` of the manufacturer.

[image]

joseffilzmaier, 2020-11-22:
This issue does not seem to be closed, as i have same problem and @dariuszseweryn also seems to have confirmed this problem (if i understand it correctly). Could you please re-open this issue?

angelix, 2021-02-11:
@dariuszseweryn as @joseffilzmaier mentions, this issue still exists and faced the same problem. Could you please re-open this issue?