dariuszseweryn/RxAndroidBle #339: if client is in observable state how to perfume Read and Write operations?

DLeague9, 2017-12-18:
While establishing connection i used `.asObservable()` property to stay connected.
` bleDevice = rxBleDevice;
        rxBleDevice.establishConnection(false)
                .flatMap(RxBleConnection::discoverServices)
                **.asObservable()**
                .subscribe(rxBleDeviceServices -> swapScanResult(rxBleDeviceServices));`
After connection established when i tried to read values it show 
> Already connected to device
`bleDevice.establishConnection(false)
                .asObservable()
        .flatMap(rxBleConnection -> Observable.combineLatest(
                        rxBleConnection.readCharacteristic(gattCharacteristicList.get(6).getUuid()),
                        rxBleConnection.readCharacteristic(gattCharacteristicList.get(7).getUuid()),
                        ReadValuesOnConnection::new
            ))
            .subscribe(
                    readValuesOnConnection -> Log.i("UUid 6 Value: ", readValuesOnConnection.lock+""),
                    throwable -> Log.e("Error", throwable.getMessage())
            );`
Logs:
`I/Bluetooth Enable: True
 I/RxBle#QueueOperation: Scan operation is requested to start.
 I/Scan Results:: 6Q:6C:05:8E:F5:5B
 I/Scan Results:: 6P:6A:05:8E:F8:2X
 I/RxBle#CancellableSubscription: Scan operation is requested to stop.
 W/zygote64: Suspending all threads took: 5.645ms
 I/Characteristics List size:: 33
 E/Error: Already connected to device with MAC address 6Q:6C:05:8E:F5:5B`

i go through the sample project but didn't find any solution. kindly guide me. thanks

dariuszseweryn, 2017-12-18:
Hello!

Thank you for using the library.

This is a general usage question and as such should be posted on [stackoverflow](https://stackoverflow.com/questions/tagged/rxandroidble) with `rxandroidble` tag—it will be easier to find by other users.

Best Regards

closed 2017-12-18 by dariuszseweryn

dariuszseweryn, 2017-12-18:
[stackoverflow question](https://stackoverflow.com/questions/47869791/rxandroidble-if-client-is-in-observable-state-how-to-perform-read-and-write-ope)