dariuszseweryn/RxAndroidBle #340: Transparent Connection

recoverrelax, 2017-12-19:
Hello. I have a scenario which is currently working, but i would like to hear the opinion of the author. The phone scans and successfully connect to a device, but eventually, the device will disconnect and connect later. What is the proper way of handling this case without user interaction?

Currently i have a retryWhen in the establishConnection observable that looks for BleDisconnectedException and resubscribe to it. The solution works great, but i was wondering, do i really need to keep trying to connect, catch the disconnectException and try again. Isn't there a better solution?.

Thanks in advanced for the great library!

dariuszseweryn, 2017-12-19:
Hello @recoverrelax 

Thanks for using the library!

Yes, the approach you have taken is the one that I use myself. 

The vanilla Android API allows for auto-reconnecting behaviour by using the `autoConnect=true` flag. Unfortunately there are some bugs connected with it so it is safer not to use it.

The library allows only for a single connection being established for a single `RxBleDevice.establishConnection()` subscription—so you need to resubscribe. The easiest way is to simply use `.retry()` operator.

I hope this clarifies it for you.

Best Regards

recoverrelax, 2017-12-19:
Hello! Thanks you for the fast reply and feedback.

closed 2017-12-19 by recoverrelax

RobLewis, 2017-12-19:
Great information, thanks. Could the OP share the code that uses `retryWhen()` to look for `BleDisconnectedException`?