dariuszseweryn/RxAndroidBle #342: Can instantiate RxBleClient with last Dagger release

GuillaumeBo, 2017-12-21:
### Summary
I faced an issue yesterday with the new Dagger release (2.14.1). At the RxBleClient creation ( `RxBleClient.create(context);`) the app crash (with 2.13 and less it's ok) 

### Library version
`1.4.3`

#### Logs from the application running with setting 
`E/AndroidRuntime: FATAL EXCEPTION: main
                                                               Process: myapp.id, PID: 23473
                                                               java.lang.NoSuchMethodError: No static method injectMembers(Ldagger/MembersInjector;Ljava/lang/Object;)Ljava/lang/Object; in class Ldagger/internal/MembersInjectors; or its super classes (declaration of 'dagger.internal.MembersInjectors' appears in /data/app/myapp.id-2/base.apk:classe
at com.polidea.rxandroidble.RxBleAdapterStateObservable_Factory.get(RxBleAdapterStateObservable_Factory.java:28)
                                                       at com.polidea.rxandroidble.RxBleAdapterStateObservable_Factory.get(RxBleAdapterStateObservable_Factory.java:10)
                                                       at com.polidea.rxandroidble.RxBleClientImpl_Factory.get(RxBleClientImpl_Factory.java:104)
                                                       at com.polidea.rxandroidble.RxBleClientImpl_Factory.get(RxBleClientImpl_Factory.java:23)
                                                       at dagger.internal.DoubleCheck.get(DoubleCheck.java:47)
                                                       at com.polidea.rxandroidble.DaggerClientComponent.rxBleClient(DaggerClientComponent.java:419)
                                                       at com.polidea.rxandroidble.RxBleClient.create(RxBleClient.java:54)`

uKL, 2017-12-21:
Hey. Thanks for updating us.

This is related to https://github.com/google/dagger/pull/950

We'll let you know when the snapshot will have a workaround.

uKL, 2018-01-04:
https://github.com/Polidea/RxAndroidBle/pull/343

The PR has been merged hence you can expect that the snapshot will work for you soon. Could you test it?

GuillaumeBo, 2018-01-04:
Sure, where can i find this snapshot ?

uKL, 2018-01-04:
https://github.com/Polidea/RxAndroidBle#snapshot

GuillaumeBo, 2018-01-04:
Offcourse i passed on it and missed it...
So i just tryed, it does'nt work

`Rejecting re-init on previously-failed class java.lang.Class<com.polidea.rxandroidble.DaggerClientComponent$1>: java.lang.NoClassDefFoundError: Failed resolution of: Lbleshadow/javax/inject/Provider;
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at com.polidea.rxandroidble.DaggerClientComponent$Builder com.polidea.rxandroidble.DaggerClientComponent.builder() (DaggerClientComponent.java:179)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at com.polidea.rxandroidble.RxBleClient com.polidea.rxandroidble.RxBleClient.create(android.content.Context) (RxBleClient.java:51)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at int my.app.id.ble.BLEConnectionService.onStartCommand(android.content.Intent, int, int) (BLEConnectionService.java:69)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at void android.app.ActivityThread.handleServiceArgs(android.app.ActivityThread$ServiceArgsData) (ActivityThread.java:3476)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at void android.app.ActivityThread.-wrap23(android.app.ActivityThread, android.app.ActivityThread$ServiceArgsData) (ActivityThread.java:-1)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at void android.app.ActivityThread$H.handleMessage(android.os.Message) (ActivityThread.java:1674)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at void android.os.Handler.dispatchMessage(android.os.Message) (Handler.java:105)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at void android.os.Looper.loop() (Looper.java:156)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at void android.app.ActivityThread.main(java.lang.String[]) (ActivityThread.java:6523)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at java.lang.Object java.lang.reflect.Method.invoke!(java.lang.Object, java.lang.Object[]) (Method.java:-2)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at void com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run() (ZygoteInit.java:941)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at void com.android.internal.os.ZygoteInit.main(java.lang.String[]) (ZygoteInit.java:831)
01-04 15:21:56.494 17870-17870/my.app.id I/art: Caused by: java.lang.ClassNotFoundException: Didn't find class "bleshadow.javax.inject.Provider" on path: DexPathList[[zip file "/data/app/my.app.id-1/base.apk"],nativeLibraryDirectories=[/data/app/my.app.id-1/lib/arm64, /data/app/my.app.id-1/base.apk!/lib/arm64-v8a, /system/lib64, /vendor/lib64, /system/vendor/lib64, /product/lib64]]
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at java.lang.Class dalvik.system.BaseDexClassLoader.findClass(java.lang.String) (BaseDexClassLoader.java:56)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String, boolean) (ClassLoader.java:380)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String) (ClassLoader.java:312)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at com.polidea.rxandroidble.DaggerClientComponent$Builder com.polidea.rxandroidble.DaggerClientComponent.builder() (DaggerClientComponent.java:179)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at com.polidea.rxandroidble.RxBleClient com.polidea.rxandroidble.RxBleClient.create(android.content.Context) (RxBleClient.java:51)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at int my.app.id.ble.BLEConnectionService.onStartCommand(android.content.Intent, int, int) (BLEConnectionService.java:69)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at void android.app.ActivityThread.handleServiceArgs(android.app.ActivityThread$ServiceArgsData) (ActivityThread.java:3476)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at void android.app.ActivityThread.-wrap23(android.app.ActivityThread, android.app.ActivityThread$ServiceArgsData) (ActivityThread.java:-1)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at void android.app.ActivityThread$H.handleMessage(android.os.Message) (ActivityThread.java:1674)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at void android.os.Handler.dispatchMessage(android.os.Message) (Handler.java:105)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at void android.os.Looper.loop() (Looper.java:156)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at void android.app.ActivityThread.main(java.lang.String[]) (ActivityThread.java:6523)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at java.lang.Object java.lang.reflect.Method.invoke!(java.lang.Object, java.lang.Object[]) (Method.java:-2)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at void com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run() (ZygoteInit.java:941)
01-04 15:21:56.494 17870-17870/my.app.id I/art:     at void com.android.internal.os.ZygoteInit.main(java.lang.String[]) (ZygoteInit.java:831)
01-04 15:21:56.496 17870-17870/my.app.id I/art: Rejecting re-init on previously-failed class java.lang.Class<com.polidea.rxandroidble.RxBleAdapterStateObservable_Factory>: java.lang.NoClassDefFoundError: Failed resolution of: Lbleshadow/dagger/internal/Factory;
01-04 15:21:56.496 17870-17870/my.app.id I/art:     at com.polidea.rxandroidble.DaggerClientComponent$Builder com.polidea.rxandroidble.DaggerClientComponent.builder() (DaggerClientComponent.java:179)
[... cut ...]
                                                                   at java.lang.reflect.Method.invoke(Native Method) 
                                                                   at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:941) 
                                                                   at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:831) `

The BLEConnectionService.java:69 line =>
`rxBleClient = RxBleClient.create(getApplicationContext());`

uKL, 2018-01-04:
Pretty strange, could you show your build.gradle regarding dependency configuration? Do you use multi dex?

GuillaumeBo, 2018-01-04:
I am using multi dex yes, i have a lot of dependencies...
What part do you want to see?

uKL, 2018-01-04:
This part related to rxandroidble. I'd like to be sure that you didn't exclude transitive dependencies. (or just confirm it)

GuillaumeBo, 2018-01-04:
`android {
    compileSdkVersion 27
    buildToolsVersion "27.0.0"
    defaultConfig {
        applicationId "my.app.id"
        minSdkVersion 21
        targetSdkVersion 27
        versionCode 1
        versionName "1.0"
        testInstrumentationRunner "android.support.test.runner.AndroidJUnitRunner"
        multiDexEnabled true
        renderscriptTargetApi 21
        renderscriptSupportModeEnabled true
    }
    buildTypes {
        debug {
            minifyEnabled false
            buildConfigField "boolean", "debugMode", "true"
        }
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
            buildConfigField "boolean", "debugMode", "false"
        }
    }
    compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
    }
}`
`ext.dagger = "2.13"
    compile "io.reactivex.rxjava2:rxjava:2.1.7"
    compile "io.reactivex.rxjava2:rxandroid:2.0.1"
    compile 'com.trello.rxlifecycle2:rxlifecycle:2.2.1'
    compile 'com.trello.rxlifecycle2:rxlifecycle-components:2.2.1'
    compile "com.github.akarnokd:rxjava2-interop:0.11.2"
    compile "com.google.dagger:dagger:$dagger"
    compile "com.google.dagger:dagger-android:$dagger"
    annotationProcessor "com.google.dagger:dagger-compiler:$dagger"
    annotationProcessor "com.google.dagger:dagger-android-processor:$dagger"

    //Bluetooth
    compile "com.polidea.rxandroidble:rxandroidble:1.5.0-SNAPSHOT"`

And other extra dependencies as google, support, firebase etc...

GuillaumeBo, 2018-01-04:
it's was `ext.dagger = "2.14.1"`  for the test, i missed to rechange for the copy/paste

dariuszseweryn, 2018-01-04:
Yup, indeed there is a problem. We will look on it Tomorrow.

dariuszseweryn, 2018-01-05:
The problem should be fixed now. You can check it by using the newest `1.5.0-SNAPSHOT`. To get rid of the old one you can follow [these instructions](https://discuss.gradle.org/t/how-do-i-always-use-the-latest-version-of-a-changing-module/5314)

GuillaumeBo, 2018-01-05:
It's working with the last Snapshot :)

ened, 2018-01-29:
@uKL You probably already know but the sonatype snapshot repository (https://oss.sonatype.org/content/repositories/snapshots/com/polidea/rxandroidble/rxandroidble/) has a 1.6.0-SNAPSHOT version already, yet that is the incorrect one.

[image]

dariuszseweryn, 2018-01-29:
`1.5.0-SNAPSHOT` ;) Sorry for misunderstanding

closed 2018-02-12 by uKL

iadeelzafar, 2020-11-11:
My project was using `rxandroidble 1.3.3`, everything was working fine. Recently, I upgraded my project to target Android 11 and migrated to Androidx. After that, I started getting this exact same error. I was then able to resolve it after upgrading to rxandroidble `1.5.0`.

My project was not using Dagger explicitly.  Hence, I'm wondering what actually caused this issue to appear? Do you guys have any idea? Any help is appreciated! @dariuszseweryn

iadeelzafar, 2020-11-11:
With more investigation, I found that I was getting this error because I recently migrated Fabric to Firebase Crashlytics. And the latest version of Firebase actually uses `Dagger 2.24` internally. This was overriding my `Dagger 2.8`

So upgrading rxandroidble to `1.5.0` solved the problem.

dariuszseweryn, 2020-11-12:
Happy to hear that. :)