dariuszseweryn/RxAndroidBle #344: observeStateChanges how to use this function

JiWenBo, 2017-12-22:
Subscription flowSubscription = rxBleClient.observeStateChanges()
    .switchMap(state -> { // switchMap makes sure that if the state will change the rxBleClient.scanBleDevices() will unsubscribe and thus end the scan
        switch (state) {

            case READY:
                // everything should work
                return rxBleClient.scanBleDevices();
            case BLUETOOTH_NOT_AVAILABLE:
                // basically no functionality will work here
            case LOCATION_PERMISSION_NOT_GRANTED:
                // scanning and connecting will not work
            case BLUETOOTH_NOT_ENABLED:
                // scanning and connecting will not work
            case LOCATION_SERVICES_NOT_ENABLED:
                // scanning will not work
            default:
                return Observable.empty();
When to trigger READY？（My English is not good ） 我在按钮的点击事件中调用 这个方法，它并不回调READY方法，而是有一个bluetoothAdapter     isEnable 这个提示。

JiWenBo, 2017-12-22:
When I turn off the bluetooth and then open it, it's READY to go back

uKL, 2017-12-22:
Hey @JiWenBo, this seems to be a general usage question and should be asked on StackOverflow (so others may benefit from answers). Please post it there with tag #rxandroidble. For commercial support, you can reach us at hello@polidea.com

closed 2017-12-22 by uKL