dariuszseweryn/RxAndroidBle #349: `RxBleAdapterStateObservable` and `DisconnectionRouter` is leaked memory

hoanglm4, 2018-01-05:
### Summary
`RxBleAdapterStateObservable` and `DisconnectionRouter` is leaked memory

### Library version
`1.4.3`

#### Preconditions
Connect to BLE device is failed

#### Steps to reproduce actual result
<br />1.  connect() method is called => waiting until connection is failed
<br />2.  connect() method is called again
<br />3.  repeat step 1, 2 about 100 times

Please see dumemory capture screen:
[`DisconnectionRouter`](https://www.dropbox.com/s/5xmcehy97p842vs/Screen%2BShot%2B2018-01-05%2Bat%2B11.14.14%2BPM.png?dl=0)
[`RxBleAdapterStateObservable`](https://www.dropbox.com/s/5xmcehy97p842vs/Screen%2BShot%2B2018-01-05%2Bat%2B11.14.14%2BPM.png?dl=0)

#### Minimum code snippet reproducing the issue
``` java
public class CharacteristicOperationExampleActivity extends RxAppCompatActivity {
public static final String EXTRA_CHARACTERISTIC_UUID = "extra_uuid";
private static final UUID characteristicUuid = UUID.fromString("25A80002-6478-11E6-BDF4");

private PublishSubject<Void> disconnectTriggerSubject = PublishSubject.create();
private Observable<RxBleConnection> connectionObservable;
private RxBleDevice bleDevice;

@Override
protected void onCreate(Bundle savedInstanceState) {
    super.onCreate(savedInstanceState);
    setContentView(R.layout.activity_example4);
    ButterKnife.bind(this);
    String macAddress = "D5:F8:47:A6:70:0E";
    bleDevice = SampleApplication.getRxBleClient(this).getBleDevice(macAddress);
    connectionObservable = prepareConnectionObservable();
    //noinspection ConstantConditions
    getSupportActionBar().setSubtitle(getString(R.string.mac_address, macAddress));
    Log.i(getClass().getSimpleName(), "onCreate");
    bleDevice.observeConnectionStateChanges().subscribe(state -> {
        Log.i(getClass().getSimpleName(), "" + state.toString());
    });
}

private Observable<RxBleConnection> prepareConnectionObservable() {
    return bleDevice
            .establishConnection(false)
            .takeUntil(disconnectTriggerSubject)
            .compose(new ConnectionSharingAdapter());
}

@OnClick(R.id.connect)
public void connect() {
    connectionObservable.flatMap(RxBleConnection::discoverServices)
            .flatMap(rxBleDeviceServices -> rxBleDeviceServices.getCharacteristic(characteristicUuid))
            .observeOn(AndroidSchedulers.mainThread())
            .doOnSubscribe(() -> Log.i(getClass().getSimpleName(), "start connect "))
            .subscribe(
                    characteristic -> {
                        Log.i(getClass().getSimpleName(), "Hey, connection has been established!");
                    },
                    this::onConnectionFailure,
                    this::onConnectionFinished
            );
    }
}
```

#### Actual result
`RxBleAdapterStateObservable` and `DisconnectionRouter` is leaked memory. Because [broacastReceiver ](https://github.com/Polidea/RxAndroidBle/blob/master/rxandroidble/src/main/java/com/polidea/rxandroidble/RxBleAdapterStateObservable.java#L50) in RxBleAdapterStateObservale is not unregister
#### Expected result
`RxBleAdapterStateObservable` and `DisconnectionRouter` is not leaked memory

dariuszseweryn, 2018-01-08:
Hello @hoanglm4 
Thank you for the report. Apparently there was an issue with `DisconnectionRouter` that prevented the `RxBleAdapterStateObservable` from being unsubscribed. It should now be fixed with #353 .
You can check out [`1.5.0-SNAPSHOT`](https://github.com/Polidea/RxAndroidBle#snapshot) if that is the case.
Best Regards

closed 2018-01-08 by dariuszseweryn

hoanglm4, 2018-01-08:
Thank you so much!