dariuszseweryn/RxAndroidBle #352: Retry mechanism on some BLE stack errors

martiwi, 2018-01-08:
### Summary
This is a proposed enhancement for discussion.

Today, RxAndroidBle does not support retry mechanism, meaning it has to be handled by upper layer from the application.
An interesting enhancement would be to expose an interface that could be implemented by upper layer if some retry should be applied on conditions.
This could be applied mostly for write operations, and especially long write. 
Example: It could be possible that on some tiny embedded devices, that incoming messages are not handled as fast as it is sent by the smartphone. Embedded device will then respond with a `GATT_INSUF_RESOURCE                 0x11` and instead of just failing the all payload, a simple retry would be good enough to have it working. 

### Library version
`1.4.3`

#### Actual result
Today, the request is canceled, and for a long write, you will have to retry and mostly facing the same issue next time. Thus, the payload can't be sent.

#### Expected result
Upper layer implement a retry interface, and check if for example, BLE stack error code is worth a retry.

dariuszseweryn, 2018-01-08:
Hello @martiwi 
Thank you for the idea. Definitely a nice to have! 👍 
Unfortunately we do not have free workload at the moment so anyone interested could help us out and submit a PR :)

martiwi, 2018-01-18:
@dariuszseweryn I've opened a PR. Please feel free to comment.

dariuszseweryn, 2018-03-12:
It took some time but we have finally merged #357 — should be available in `1.5.0-SNAPSHOT` already and in `1.5.0` in the upcoming days. 🎉  Thank you for your effort 👍

closed 2018-03-12 by dariuszseweryn

martiwi, 2018-03-12:
Hi @dariuszseweryn, thank you!!! I don't see it in the `1.5.0-SNAPSHOT` do you know when it will be available. I checked it from the versio uploaded this morning: `rxandroidble-1.5.0-20180312.055735-7.aar`.

uKL, 2018-03-12:
It should be in the snapshot you've mentioned.

martiwi, 2018-03-12:
Hi @uKL from the source within the archive file, it is not unfortunately. Also this file has been generated before the branch being merged into `master` .

uKL, 2018-03-12:
I merged it at 9:47 UTC
The build ran at 9:57 UTC https://travis-ci.org/Polidea/RxAndroidBle/builds/352249008
This is the artifact rxandroidble-1.5.0-20180312.095700-24.aar
https://i.imgur.com/g5uoxGV.png

I believe your cache is fooling around :)

martiwi, 2018-03-12:
Ok, I've figured out, `https://oss.sonatype.org/content/repositories/snapshots/com/polidea/rxandroidble2` instead of `https://oss.sonatype.org/content/repositories/snapshots/com/polidea/rxandroidble`. My bad! Any idea when this will be part of `rxandroidble2` ? Thanks again for all your hard work guys!

uKL, 2018-03-12:
I suppose tomorrow :) We're getting close to a release od RxJava2. FYI, we have gitter for chats :) https://gitter.im/RxBLELibraries/RxAndroidBle