dariuszseweryn/RxAndroidBle #358: The `createNewLongWriteBuilder` of `RxBleConnectionMock` is incomplete.

streetsofboston, 2018-01-23:
### Summary
Not sure if this is a bug or not, it maybe a feature-request, but it prevented me from properly writing tests for code that executes `createNewLongWriteBuilder` and handles the `Observable` returned by it.

The method `createNewLongWriteBuilder` of `RxBleConnectionMock` returns an `Observable` whose chain will never actually call the method `writeCharacteristic`.

The builder of `createNewLongWriteBuilder` constructs a private member called `bluetoothGattCharacteristicObservable` but this member is never used in the creation of the `Observable` returned by `LongWriteOperationBuilder#build`.

### Library version
`1.4.3`

### Example of how I fixed it locally in my tests
I added code to my tests that `spy` the `RxBleDeviceMock` and `spy` the `RxBleConnectionMock` emitted by the Observable returned by the `establishConnection` method.

Then I used Mockito's `doAnswer` on the emitted `RxBleConnectionMock` to return my own implementation of the `LongWriteOperationBuilder` interface. It is much like the one from the RxBleAndroid's mock-client code itself. Only the method `build()` needed a different implementation.

Here is the modified implementation (in Kotlin) of its `build()` method:
```
override fun build(): Observable<ByteArray> {
    if (bluetoothGattCharacteristicObservable == null) {
        throw IllegalArgumentException("setCharacteristicUuid() or setCharacteristic() needs to be called before build()")
    }

    if (bytes == null) {
        throw IllegalArgumentException("setBytes() needs to be called before build()")
    }

    return letMany(bluetoothGattCharacteristicObservable, bytes) { bleGattCharObs, bytes ->
        val excess = bytes.size % maxBatchSize > 0
        val numberOfBatches = bytes.size / maxBatchSize + if (excess) 1 else 0
        val numberOfBatchesLeft = AtomicInteger(numberOfBatches)

        val batchObs = Observable.fromCallable { numberOfBatchesLeft.get() }
        Observable.zip(batchObs, bleGattCharObs) { chunkIdx, char -> Pair(chunkIdx, char) }
                .concatMap { (idx, char) ->
                    val start = maxBatchSize * (numberOfBatches - idx)
                    val end = min(start + maxBatchSize, bytes.size)
                    connection
                            .writeCharacteristic(char, bytes.sliceArray(start..end))
                            .concatMap { Observable.just(idx) }
                }
                .map { idx -> idx > 0 }
                .compose(writeOperationAckStrategy)
                .repeatWhen { it.takeWhile { numberOfBatchesLeft.decrementAndGet() > 0 } }
                .toCompletable()
                .andThen(Observable.just<ByteArray>(bytes))
    } ?: Observable.empty()
}
```
(The lambda of `letMany` is only called if all its input params are non-null, returning the return-value of the lambda)

The main difference is the inclusion of the `bluetoothGattCharacteristicObservable` in the Observable chain that is returned by the `build` method.

dariuszseweryn, 2018-01-23:
Hello @streetsofboston 

Actually for usage in the tests I would encourage you not to use `MockRxAndroidBle` but just to mock all of the interfaces of `RxAndroidBle` directly. It is easier to do than to wire in into the library mock (which is very neglected at this moment and is not high on the priority list)

streetsofboston, 2018-01-23:
Thank you for the quick response.
Will the `mock-client` library be deprecated?

dariuszseweryn, 2018-01-23:
First of all `mock-client` was created as a simple replacement for the application when the actual peripheral is not physically accessible. It's functionality is limited at best.

I have in mind an idea what could be done with it but it would be a complete overhaul of the current implementation. Unfortunately my time is limited.

Will it be deprecated? In the long run—I hope not. Currently this seems to be the case.

uKL, 2018-03-15:
I think this should be closed in favor of #370. Feel free to reopen if you think otherwise.

closed 2018-03-15 by uKL