dariuszseweryn/RxAndroidBle #363: using CALLBACK_TYPE_FIRST_MATCH setting cause BleScanException: Scan failed because out of hardware resources (code 9)

mohsenoid, 2018-02-09:
In my app, I am using your library to scan for my BLE device once using `CALLBACK_TYPE_ALL_MATCHES` setting and my device `service UUID` filter and everything works nice. After user select which device he wants to connect I save the mac address for next time, but in next time I set the filter on selected device `mac address` and change setting to `CALLBACK_TYPE_FIRST_MATCH` cause I don't want to wait for all results but I get:
BleScanException: Scan failed because out of hardware resources (code 9)

If I change it back to `CALLBACK_TYPE_ALL_MATCHES` it works but wants to know if I am miss understanding and miss using the library or there is another problem.

```
    scanSubscription =
            rxBleClient
                    .scanBleDevices(getSettings(chrisBluetoothDevice), getFilter(chrisBluetoothDevice))
                    .doOnSubscribe {
                        bleState.onNext(BLEStates.Scanning())
                        scanResults = HashMap()
                    }
                    .doOnUnsubscribe({
                        scanSubscription = null
                        bleState.onNext(BLEStates.Idle())
                    })
                    .observeOn(AndroidSchedulers.mainThread())
                    .filter { scanResult ->
                        // filter previously founded devices
                        !scanResults.contains(scanResult.bleDevice.macAddress)
                    }
                    .timeout(BLEConstants.SCAN_TIMEOUT, TimeUnit.MILLISECONDS, Observable.empty())
                    .subscribe(
                            { scanResult ->
                                scanResults[scanResult.bleDevice.macAddress] = scanResult.bleDevice
                            },
                            { throwable ->
                                bleState.onNext(BLEStates.Error(throwable))

                                if (throwable is BleScanException) {
                                    Timber.e(throwable)
                                }
                            },
                            {
                                bleState.onNext(BLEStates.Scanned(chrisBluetoothDevice, scanResults.values.map { bleDevice -> ChrisBluetoothDevice(bleDevice.name, bleDevice.macAddress) }.toTypedArray()))
                            })
```

```
    private fun getFilter(chrisBluetoothDevice: ChrisBluetoothDevice? = null): ScanFilter {
        val builder = ScanFilter.Builder()

        if (chrisBluetoothDevice != null) {
            builder.setDeviceAddress(chrisBluetoothDevice.macAddress)
        } else {
            builder.setServiceUuid(serviceUUID)
        }

        return builder.build()
    }
```

```
    private fun getSettings(chrisBluetoothDevice: ChrisBluetoothDevice? = null): ScanSettings {
        val builder = ScanSettings.Builder()
                .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)

        if (chrisBluetoothDevice != null) {
            // it make sense to use CALLBACK_TYPE_FIRST_MATCH but it cause BleScanException: Scan failed 
            // builder.setCallbackType(ScanSettings.CALLBACK_TYPE_FIRST_MATCH)
            builder.setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
        } else {
            builder.setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
        }
        return builder.build()
    }
```

dariuszseweryn, 2018-02-09:
Hello @mohsenoid 

As you have pasted: `BleScanException: Scan failed because out of hardware resources (code 9)`
It means that the hardware does not have enough resources to start a scan with the specified settings. It is not related to the library but the Android hardware limitations

mohsenoid, 2018-02-09:
Dear @dariuszseweryn 

Thanks for your fast reply,

Do you have any idea why the same phone without any changes works fine when I change setting back to CALLBACK_TYPE_ALL_MATCHES?

closed 2018-02-12 by dariuszseweryn