dariuszseweryn/RxAndroidBle #364: Access to BluetoothGatt

GuillaumeBo, 2018-02-09:
Hello,

Is it possible to add a getter to the BluetoothGatt in the RxBleConnectionImpl ?

In my exemple i need to force clear cache on the unexpected disconnection from my ble device.
I am facing issue on the nexus 6 on this case although with a honor phone there is no issue.

It may be usefull for other things even if i admit that solution bypass the lib...

GuillaumeBo, 2018-02-12:
Thanks you for your answer, i try to use this method, but my operation is not executed cause the device is Disconnected. I need to execute BluetoothGatt operation after the "BluetoothProfile.STATE_DISCONNECTED". It seems you not authorized this, am i wrong ?

`public class ClearBleConnectionCacheCustomOperation implements RxBleCustomOperation<Void> {
    @NonNull
    @Override
    public Observable<Void> asObservable(BluetoothGatt bluetoothGatt, RxBleGattCallback rxBleGattCallback, Scheduler scheduler) throws Throwable {
        return RxJavaInterop.toV1Observable(io.reactivex.Observable.create(emitter -> {
            try {
                Method localMethod = bluetoothGatt.getClass().getMethod("refresh", new Class[0]);
                if (localMethod != null) {
                    boolean bool = (Boolean) localMethod.invoke(bluetoothGatt, new Object[0]);
                    emitter.onComplete();
                }
            } catch (Exception localException) {
                emitter.tryOnError(localException);
            }
        }), BackpressureStrategy.BUFFER);
    }
}`

dariuszseweryn, 2018-02-12:
No, the custom operation may be only executed when the device is connected. So the best idea is to execute it right after the connection.

GuillaumeBo, 2018-02-12:
The operation consist to clear characteristics list cached on the device just after disconnection. 
It's not working when the device is connected...
What is reason to block operation on disconnected device ? BluetoothGatt is still valid at this moment

dariuszseweryn, 2018-02-12:
It is not valid. The `BluetoothGatt` is closed by the library right after the disconnection. 
Why it does not work when the device is connected?

GuillaumeBo, 2018-02-12:
In my old app i made the refresh + close in the "onConnectionStateChange" with BluetoothProfile.STATE_DISCONNECTED status. Maybe we should add task for this moment ?

I just try (after the connection) first time it fails (refresh work) but ble actions right after fail with the error : "GATT exception from MAC address 01:02:03:04:05:06, with type BleGattOperation{description='SERVICE_DISCOVERY'"
on every actions, write, read, notify

On second try it works...

```
in the subscribe of the device connection :
rxBleConnection.queue(new ClearBleConnectionCacheCustomOperation()).subscribe(o -> {
                                rxBleConnection.discoverServices();
                            }, throwable -> {
                            });
```

dariuszseweryn, 2018-02-12:
Try [this](https://github.com/Polidea/react-native-ble-plx/blob/feature/refresh_gatt/android/src/main/java/com/polidea/reactnativeble/utils/RefreshGattCustomOperation.java) for refresh as a first operation after the connection

GuillaumeBo, 2018-02-12:
Thanks for the link, can you explain briefly what your code do ? I do not understand the part pour with the delay and the disconnect listner ?
And when you call it ?

dariuszseweryn, 2018-02-12:
1. Delay as the BLE stack is unstable after `BluetoothGatt.refresh()`—you experience it as the service discovery exception
2. Disconnect listener is to conform to the `CustomOperation` contract. But it should not matter in this situation
3. Right after the connection is established before any other thing. You can see the implementation in the branch the link points

dariuszseweryn, 2018-02-12:
@GuillaumeBo Is there anything else I can help you with?

GuillaumeBo, 2018-02-12:
Ok i see !
Now the issue is that this lengthens a lot the time of connection where is about 100-150ms in my case.
The 1 second is for security without strong time constraints or it's following your tests ?

dariuszseweryn, 2018-02-12:
There is no info on when the stack stabilises so the 1 second is to be on the safe side. Usually a second of delay is not much since it is only needed sporadically (I see no reason to call `BluetoothGatt.refresh()` every time)

GuillaumeBo, 2018-02-12:
Yes i will try to use only when the stack is KO.
For information errors fired that obliges me to refresh the stack are : 
`GATT exception from MAC address 01:02:03:04:05:06, status 1, type BleGattOperation{description='CHARACTERISTIC_WRITE'}. (Look up status 0x01 here https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h)`
and some time
`GATT exception from MAC address 01:02:03:04:05:06, status 3, type BleGattOperation{description='CHARACTERISTIC_WRITE'}. (Look up status 0x03 here https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h)
`

dariuszseweryn, 2018-02-12:
Thanks for the info. Feel free to post your findings here. Do you want to keep this ticket opened? Or is this all and it can be closed for now?

For reference: #53

GuillaumeBo, 2018-02-12:
You can close it, i will post messages if i found any informations about theses bug.
And if anybody has infos about it, feel free to explain us :)

closed 2018-02-12 by dariuszseweryn

GuillaumeBo, 2018-02-12:
@dariuszseweryn a last thing, can we get more details on throwable returned by writeCharacteristic, readCharacteristic etc ? Like errorCode, cause here we have only the string build by the lib and it's no really proper to parse it to obtain the code ?

dariuszseweryn, 2018-02-12:
It is a `BleGattCharacteristicException` which you would know if you would use `exception.toString()` I think. The getters are available.

GuillaumeBo, 2018-02-12:
Perfect, thanks you :)

GuillaumeBo, 2018-02-20:
@dariuszseweryn Hello, i updated the lib to get only rx2 in my app, but i am facing issue. I am not able to make your customOperation for refreshBleGatt work with rx2. I got compile error
"amb(java.lang.Iterable<? extends io.reactivex.ObservableSource<? extends T>>) in Observable cannot be applied to (io.reactivex.Observable<Boolean>, io.reactivex.Observable<Boolean>)

did you know how to fix this ?

dariuszseweryn, 2018-02-23:
I have not yet done anything with the rx2 version—if it is a compile error then probably some interfaces do not match in how you have implemented your operation. I don't have the full context to give you any hint yet

GuillaumeBo, 2018-03-01:
It's compilation/ide error on your file just after replace rx.java1 element to rx.java2 version, i did not make any other modification on it.
the error appear on the "Observable.amb(
                Observable.fromCallable(new Callable<Boolean>() {" bloc

uKL, 2018-03-01:
Please have a look at http://reactivex.io/documentation/operators/amb.html and keep in mind that `amb` in rxjava2 has different parameters than in rxjava1. Use ambWith or provide an iterable.