dariuszseweryn/RxAndroidBle #365: Problems checking out RxAndroidBle with Android Studio

RobLewis, 2018-02-11:
I am trying to check out the current RxAndroidBle repo into a new Android Studio project. 

I get the message "The modules below are not imported from Gradle anymore. Check those to be removed from the ide project too:" 

The only listed module is RxAndroidBle, and its checkbox is already checked. If I uncheck the box, the import appears to go OK, but then I'm warned that the module 'sample' is using the retrolambda plugin (which should probably be removed, no?) 

Android Studio version is 3.0.1 (Mac).

dariuszseweryn, 2018-02-12:
Hello @RobLewis 

Isn't it related to #328 ?
We have created an [issue on the official tracker](https://issuetracker.google.com/issues/70710141)

Currently we have higher priority tasks on our list to take on. Any help is welcome.

piotrek1543, 2018-06-12:
Hi, 
I'm using now the latest version of Android Studio (3.1.3) with `RxAndroidBLE` project, since AS 3.1.0 I haven't noticed any problems.

@RobLewis does problem still occur?

dariuszseweryn, 2019-02-15:
According to the [issue opened on AS tracker](https://issuetracker.google.com/issues/70710141) this should be already fixed. Closing.

closed 2019-02-15 by dariuszseweryn