dariuszseweryn/RxAndroidBle #370: Rework `mockrxandroidble`

dariuszseweryn, 2018-02-12:
### Summary
In the current state the `mockrxandroidble` (`mockclient`) is not very usable both. [It does not work well with unit tests](https://github.com/Polidea/RxAndroidBle#testing) and is not really usable for substitution in a live application due to [lack of possibility to manage state](https://github.com/Polidea/RxAndroidBle/issues/192).

The idea is either to deprecate the `mockclient` completely or to rework it so it could be used for instance with a [server API](#152)

komuros, 2018-02-27:
While I'm waiting the rework I have created a wrapped method to use new scanBleDevice method when i'm not in mock mode

in Debug definition : 
`fun RxBleClient.scanBleDeviceCompat(scanSettings: ScanSettings): Observable<ScanResult> {
    if (DebugModules.mock) {
        return scanBleDevices().flatMap {
            Observable.zip(Observable.just(it.bleDevice), it.bleDevice.establishConnection(false), { device: RxBleDevice, connection: RxBleConnection ->
                Observable.zip(Observable.just(device), connection.discoverServices(), { device, services ->
                    val scanResult = ScanResult(device, 0, 0L, null,
                            ScanRecordImplTwoCompat(services.bluetoothGattServices.map { ParcelUuid(it.uuid) }.toMutableList()))
                    scanResult
                })
            }).flatMap { it }
        }
    } else {
        return scanBleDevices(scanSettings, ScanFilter.empty())
    }
}`

in release : 
`fun RxBleClient.scanBleDeviceCompat(scanSettings: ScanSettings): Observable<ScanResult> {
    return scanBleDevices(scanSettings, ScanFilter.empty())
}`

In my case it works because i don't need in my ScanRecord other thing than ServicesUUId and my ScanFilter is empty.
I post if other person need that.

streetsofboston, 2018-03-12:
I'd love to work on this. It is actually one of my work goals :)
For our own library, we managed to create mocked remote BLE devices to be able to run end-to-end unit test and to monitor/test the BLE-requests received by remote BLE-devices. But it relies heavily on PowerMock and knowledge of the private internal implementation of the `mockclient`.

The code we have is in Kotlin only and not that pretty. 

What would you prefer? 
- Creating/updating the `mockclient` to make it work without the need for PowerMock (mockito only or through dependency injection) or 
- Remove the `mockclient` and modify the actual RxAndroidBle code to remove its dependency on classes such as `BluetoothGattCharacteristic`, etc., and replace those by interfaces whose implementations can be injected.

uKL, 2018-03-16:
Hey, 

that's great!

We're planning future versions, we'll get back to you probably next week with some proposals.

streetsofboston, 2018-03-16:
Sounds good! Thank you!

nateridderman, 2019-01-16:
When you say it doesn't work well with unit tests, do you mean the dependency on Robolectric or are there additional things to be aware of?

dariuszseweryn, 2019-01-16:
`Robolectric` dependency is the issue for unit tests