dariuszseweryn/RxAndroidBle #374: BleConnectionCompat ignores `autoConnect` value passed to it in some situations

budius, 2018-02-20:
### Summary
On BleConnectionCompat.java the `boolean autoConnect` passed to it's `connectGatt` method is only taken into account if calling native implementation if `SDK_INT >= Nougat`

Further down that class there's on lines 60, 67, 70, and 85 calls to connectGatt implementations that simply pass `true` instead.

Maybe that's a specific hack for some broken Android version, but there's no code or commit comment about it. Hence I'm opening a issue, instead of Pull Request.

If it's a real issue, I'll be happy to open a PR

uKL, 2018-02-20:
Thanks for analyzing the code :)

The algorithm is following:
1. For `Android < N` or if `autoconnect == true` => go with a reflection-based setup, this fixes race condition related to a connection setup. Therefore **true** is forcefully passed.

2. Otherwise, use connectGattCompat and
2.1. For `Android M+`, use new API to force connection over Bluetooth LE (the device may support LE and classic BT at the same time)
2.2. For older devices use **TRANSPORT_AUTO**

Feel free to reopen if you find anything else that is related.

closed 2018-02-20 by uKL

budius, 2018-02-20:
Thanks for the quick reply, I guess I missed the `|| !autoConnect` part of the if statement.
So that means that for `autoconnect = false` it's just a matter of using or not the `TRANSPORT_LE` flag.

Thaks

uKL, 2018-02-20:
100% correct, the issue was related to autoConnect = true.