dariuszseweryn/RxAndroidBle #376: Failed to reconnect device after bluetooth disable

TimoPtr, 2018-02-20:
### Summary
Failed to reconnect to a device after switch on the bluetooth. And on every device when bluetooth enable weird exception.

### Library version
`1.4.3`

#### Preconditions
- Honor 8 or Gretel gt6000

#### Steps to reproduce actual result
<br />1. Establish connection
<br />2. Kill bluetooth 
<br />3. try to establish again connection

#### Minimum code snippet reproducing the issue

`    
override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)
        setContentView(R.layout.activity_main)

        val rxBleClient = RxBleClient.create(this)

        val device = rxBleClient.getBleDevice("XX:XX:XX:XX:XX:XX")

        button.setOnClickListener {

            subscriptor = RxJavaInterop.toV2Observable(device?.establishConnection(false)?.doOnNext {
                Log.i(TAG, "connected RX")
            })?.subscribe({
                Log.w(TAG, "connected")
            }, {
                Log.e(TAG, "error",it)
            })

        }
    }`
#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`

First click : 
`02-20 18:22:08.471 8236-8236/com.example.testble D/MAIN: click
02-20 18:22:08.518 8236-8287/com.example.testble D/BluetoothGatt: connect() - device: XXXXXXXXXXXXXXXXXX, auto: false
    registerApp()
02-20 18:22:08.519 8236-8287/com.example.testble D/BluetoothGatt: registerApp() - UUID=b46d98a3-c929-47b1-9737-c94b117d7f53
02-20 18:22:08.536 8236-8248/com.example.testble D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
02-20 18:22:08.760 8236-8248/com.example.testble D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=XXXXXXXXXXXXXXXXXX
02-20 18:22:08.766 8236-8286/com.example.testble I/MAIN: connected RX
02-20 18:22:08.766 8236-8286/com.example.testble W/MAIN: connected
`
Disable bluetooth

When trying to connect after Bluetooth disable : 

`02-20 18:18:07.981 21761-21761/com.example.testble D/MAIN: click
02-20 18:18:08.026 21761-21915/com.example.testble D/BluetoothGatt: connect() - device: XXXXXXXXXXXXXXXXXX, auto: false
02-20 18:18:08.027 21761-21915/com.example.testble D/BluetoothGatt: registerApp()
    registerApp() - UUID=63905fac-2e0b-4cbd-ad6e-87c11750d6b3
02-20 18:18:08.050 21761-21774/com.example.testble D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
02-20 18:18:38.068 21761-21774/com.example.testble D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=7 device=XXXXXXXXXXXXXXXXXX
02-20 18:18:38.095 21761-21774/com.example.testble E/MAIN: error
    com.polidea.rxandroidble.exceptions.BleDisconnectedException: Disconnected from XXXXXXXXXXXXXXXXXX
        at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:76)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:191)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
        at android.os.Binder.execTransact(Binder.java:565)
02-20 18:18:38.291 21761-21766/com.example.testble I/art: Do partial code cache collection, code=29KB, data=26KB
    After code cache collection, code=29KB, data=26KB
    Increasing code cache capacity to 128KB
02-20 18:18:50.343 21761-21761/com.example.testble V/AudioManager: playSoundEffect   effectType: 0
    querySoundEffectsEnabled...
[... 117 lines cut ...]
        at com.polidea.rxandroidble.internal.RxBleDeviceImpl$1.call(RxBleDeviceImpl.java:67)
        at rx.internal.operators.OnSubscribeDefer.call(OnSubscribeDefer.java:46)
        at rx.internal.operators.OnSubscribeDefer.call(OnSubscribeDefer.java:35)
        at rx.Observable.unsafeSubscribe(Observable.java:10256)
        at rx.internal.operators.OnSubscribeDoOnEach.call(OnSubscribeDoOnEach.java:41)
        at rx.internal.operators.OnSubscribeDoOnEach.call(OnSubscribeDoOnEach.java:30)
        at rx.Observable.unsafeSubscribe(Observable.java:10256)
        at hu.akarnokd.rxjava.interop.ObservableV1ToObservableV2.subscribeActual(ObservableV1ToObservableV2.java:37)
        at io.reactivex.Observable.subscribe(Observable.java:10903)
        at io.reactivex.Observable.subscribe(Observable.java:10889)
        at io.reactivex.Observable.subscribe(Observable.java:10818)
        at com.example.testble.MainActivity$onCreate$1.onClick(MainActivity.kt:54)
        at android.view.View.performClick(View.java:5646)
        at android.view.View$PerformClick.run(View.java:22473)
        at android.os.Handler.handleCallback(Handler.java:761)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at android.os.Looper.loop(Looper.java:156)
        at android.app.ActivityThread.main(ActivityThread.java:6523)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:942)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:832)`

Not only on the Honor is that normal ?

#### Expected result

TimoPtr, 2018-02-21:
I'm able to reproduce the bug with other smartphone if I disable in setting Bluetooth scanning (which use to improve location by allowing system apps and services to detest Bluetooth devices at any time). Without the library, I investigate a fix where I first scan an use the objet return in the scan (it failed if I use getRemoteDevice) it works with my custom Gatt connection. I'm trying to do it with RxAndroidBLE. If It's works I will post the fix

TimoPtr, 2018-02-21:
So it's work if we use the RxBleDevice create by scanBleDevice, if using rxBleClient.getBleDevice() it's not working. Can we add something to automatically scan ?

uKL, 2018-02-21:
Hey,

I'm aware that on some devices the BLE device needs to be scanned at least once to co able to connect. This is related to BLE addressing and exchanging address times (@dariuszseweryn knows more).

What kind of error code do you receive?

TimoPtr, 2018-02-21:
Error code 133

uKL, 2018-02-21:
This is not helpful that much. This may be a good idea to introduce some compatibility mode when we'll have a new, more flexible setup API (@dariuszseweryn is working on it in his free time). I think making this a default behavior may not be the best idea.

Until then you can do something so silly like a `retryWhen`
1. If exception is a `BleDisconnectedException`
2. Use client to scan for devices, filtering for our mac address
3. On the first matching scan result, resubscribe to connection

Or you can try "autoConnect" flag, that may work as well.

dariuszseweryn, 2018-02-21:
Hello @TimoPtr 
You can read more on the topic here: #318
Currently we have no solution that is bulletproof...

dariuszseweryn, 2018-02-28:
@TimoPtr 
when do you dispose your `subscriptor`?

TimoPtr, 2018-02-28:
In `onDestroy`

dariuszseweryn, 2018-02-28:
So you have multiple dangling `subscriptors` because you create new here:
```
        button.setOnClickListener {

            subscriptor = RxJavaInterop.toV2Observable(device?.establishConnection(false)?.doOnNext {
                Log.i(TAG, "connected RX")
            })?.subscribe({
                Log.w(TAG, "connected")
            }, {
                Log.e(TAG, "error",it)
            })

        }
```
Try changing to:
```
        button.setOnClickListener {
            subscriptor?.dispose()
            subscriptor = RxJavaInterop.toV2Observable(device?.establishConnection(false)?.doOnNext {
                Log.i(TAG, "connected RX")
            })?.subscribe({
                Log.w(TAG, "connected")
            }, {
                Log.e(TAG, "error",it)
            })

        }
```

dariuszseweryn, 2018-02-28:
I would always dispose if I am no longer interested in a subscription. The library checks if the flow was unsubscribed

TimoPtr, 2018-02-28:
But if the onError is trigger you don't need to ?

dariuszseweryn, 2018-02-28:
Not really sure about it

dariuszseweryn, 2018-02-28:
Either way I expect that if you would call `subscriptor?.dispose()` before creating a new one everything should work just fine

TimoPtr, 2018-02-28:
Hum so I update my test case and I end up with a BleAlreadyConnectedException. The scenario is 
- Connect to a device by its MAC
- Power off the bluetooth (receive a BleDisconnectedException)
- Power on the bluetooh 
- click on the button 
- Receive BleAlreadyConnectedException

Sample : 
```
      val rxBleClient = RxBleClient.create(this)

        val device = rxBleClient.getBleDevice("XXXXXXXX")

        button.setOnClickListener {
            Log.d(TAG, "click")
            button.isEnabled = false
            Log.d(TAG, "clear")
            subscriptor?.dispose()
            subscriptor = null
            subscriptor = RxJavaInterop.toV2Observable(device?.establishConnection(false)?.doOnNext {
                Log.i(TAG, "connected RX")
            })?.observeOn(AndroidSchedulers.mainThread())?.subscribe({
                Log.w(TAG, "connected")
            }, {
                button.isEnabled = true
                Log.e(TAG, "error", it)
            })
        }
 ```

On Galaxy s8

dariuszseweryn, 2018-03-01:
I have tried to reproduce the same issue but failed. I have changed the sample's `ScanActivity` code to:
```
    @OnClick(R.id.scan_toggle_btn)
    public void onScanToggleClick() {
        Log.w(TAG, "click");

        if (scanSubscription != null) scanSubscription.dispose();
        scanSubscription = RxJavaInterop.toV2Observable(
                rxBleClient.getBleDevice(SENSOR_TAG_MAC_ADDRESS_1)
                        .establishConnection(false)
                        .doOnNext(rxBleConnection -> Log.i(TAG, "connected RX"))
        )
                .observeOn(io.reactivex.android.schedulers.AndroidSchedulers.mainThread())
                .subscribe(
                        rxBleConnection -> Log.w(TAG, "connected"),
                        throwable -> Log.e(TAG, "error", throwable)
                );
    }
```
Logs:
```
03-01 10:04:51.240 600-600/com.polidea.rxandroidble.sample W/MAIN: click
03-01 10:04:51.302 600-660/com.polidea.rxandroidble.sample D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(1108935128)
03-01 10:04:51.303 600-616/com.polidea.rxandroidble.sample D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(1108935128)
03-01 10:04:51.308 600-661/com.polidea.rxandroidble.sample V/RxBle#BleConnectionCompat: Connecting without reflection
03-01 10:04:51.314 600-661/com.polidea.rxandroidble.sample D/BluetoothGatt: connect() - device: 5C:31:3E:C1:1F:17, auto: false
03-01 10:04:51.314 600-661/com.polidea.rxandroidble.sample D/BluetoothGatt: registerApp()
03-01 10:04:51.369 600-661/com.polidea.rxandroidble.sample D/BluetoothGatt: registerApp() - UUID=8aeb7178-e681-4534-9a65-93288e33f7ed
03-01 10:04:51.372 600-612/com.polidea.rxandroidble.sample D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
03-01 10:04:51.652 600-613/com.polidea.rxandroidble.sample D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=5C:31:3E:C1:1F:17
03-01 10:04:51.653 600-613/com.polidea.rxandroidble.sample D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
03-01 10:04:51.656 600-660/com.polidea.rxandroidble.sample I/MAIN: connected RX
03-01 10:04:51.658 600-616/com.polidea.rxandroidble.sample D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(1108935128) in 355 ms
03-01 10:04:51.663 600-600/com.polidea.rxandroidble.sample W/MAIN: connected
03-01 10:04:51.979 600-620/com.polidea.rxandroidble.sample D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=5C:31:3E:C1:1F:17
03-01 10:04:51.980 600-620/com.polidea.rxandroidble.sample D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=0
03-01 10:04:51.981 600-620/com.polidea.rxandroidble.sample I/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (5C:31:3E:C1:1F:17)
03-01 10:04:51.989 600-659/com.polidea.rxandroidble.sample D/RxBle#Executors$RunnableAdapter: Terminated.
03-01 10:04:51.990 600-600/com.polidea.rxandroidble.sample E/MAIN: error
                                                                   com.polidea.rxandroidble.exceptions.BleDisconnectedException: Disconnected from 5C:31:3E:C1:1F:17
                                                                       at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:76)
                                                                       at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:168)
                                                                       at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:71)
                                                                       at android.os.Binder.execTransact(Binder.java:404)
                                                                       at dalvik.system.NativeStart.run(Native Method)
03-01 10:04:51.994 600-660/com.polidea.rxandroidble.sample D/RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(1108948264)
03-01 10:04:51.996 600-616/com.polidea.rxandroidble.sample D/RxBle#ClientOperationQueue: STARTED  DisconnectOperation(1108948264)
03-01 10:04:51.996 600-661/com.polidea.rxandroidble.sample D/BluetoothManager: getConnectionState()
03-01 10:04:51.996 600-661/com.polidea.rxandroidble.sample D/BluetoothManager: getConnectedDevices
03-01 10:04:52.001 600-661/com.polidea.rxandroidble.sample D/BluetoothGatt: close()
03-01 10:04:52.001 600-661/com.polidea.rxandroidble.sample D/BluetoothGatt: unregisterApp() - mClientIf=5
03-01 10:04:52.004 600-616/com.polidea.rxandroidble.sample D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(1108948264) in 8 ms
03-01 10:04:57.107 600-600/com.polidea.rxandroidble.sample W/MAIN: click
03-01 10:04:57.122 600-660/com.polidea.rxandroidble.sample D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(1109416080)
03-01 10:04:57.123 600-616/com.polidea.rxandroidble.sample D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(1109416080)
03-01 10:04:57.131 600-661/com.polidea.rxandroidble.sample V/RxBle#BleConnectionCompat: Connecting without reflection
03-01 10:04:57.132 600-661/com.polidea.rxandroidble.sample D/BluetoothGatt: connect() - device: 5C:31:3E:C1:1F:17, auto: false
03-01 10:04:57.132 600-661/com.polidea.rxandroidble.sample D/BluetoothGatt: registerApp()
03-01 10:04:57.133 600-661/com.polidea.rxandroidble.sample D/BluetoothGatt: registerApp() - UUID=4800cb0f-95bb-46bc-b232-85eb6b99904a
[... cut ...]
03-01 10:04:57.403 600-613/com.polidea.rxandroidble.sample D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
03-01 10:04:57.404 600-660/com.polidea.rxandroidble.sample I/MAIN: connected RX
03-01 10:04:57.406 600-616/com.polidea.rxandroidble.sample D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(1109416080) in 283 ms
03-01 10:04:57.406 600-600/com.polidea.rxandroidble.sample W/MAIN: connected
```
I have used `Java` instead of `Kotlin` but it should not make any difference.
I suspect you have a different problem in your code

TimoPtr, 2018-03-01:
The problem after Bluetooth adapter disable, appear only on a specific phone Honor 8. It was working fine with Pixel and S8. I have the same code, so I assume maybe it's a location problem ? I only ask for COARSE permission.

dariuszseweryn, 2018-03-01:
`BleAlreadyConnectedException` is not connected with permissions.
Unfortunately I do not have access to Huawei Honor 8 to test this scenario but judging from the logs nothing should break. I cannot tell anything more... You could either paste the whole activity code or show the full logs with the mac addresses to which you are talking to

TimoPtr, 2018-03-01:
My bad ... In my test application I was on an old version of rxAndroidBLE 1.4.0 and interop 0.10.6 .... So for now I don't have BleAlreadyConnectedException but with the Honor it's just impossible to connect I after Bluetooth adapter off ... Except if I start a scan maybe it's a problem from the manufacturer. I will end up by always scanning before connected ... but it add times ...

dariuszseweryn, 2018-03-01:
Good to hear that it is solved. You have incorrectly stated that the problem is on `1.4.3` version

Closing

closed 2018-03-01 by dariuszseweryn