dariuszseweryn/RxAndroidBle #379: Consider moving from a ConnectionSharingAdapter towards a ReplyingShare

uKL, 2018-02-26:
Due to some issues with ConnectionSharingAdapter let's check if we can avoid it and use the ReplyingShare.

- [ ] Verify if replying share fits our needs
- [ ] Provide a migration paragraph in the readme

RobLewis, 2018-02-26:
I use `ConnectionSharingAdapter`. What are its issues? And I think you may have a typo: isn't it `[ReplayingShare](https://github.com/JakeWharton/RxReplayingShare)`?

streetsofboston, 2018-03-12:
We had some issues with the `ConnectionSharingAdapter` in a multithreaded environment. We wrote an updated version. This updated version would obtain/emit shared connections on one thread.

RobLewis, 2018-03-12:
@streetsofboston I looked briefly at `ReplayingShare` and didn't notice any attention to threading. Do you think it would have the same issues as `ConnectionSharingAdapter`?

streetsofboston, 2018-03-12:
The way `ReplayingShare` is working, is quite differently from the `ConnectionSharingAdapter`. I wouldn't be able to tell you if it had the same issues or not.

The way we fixed it was by `defer`-ing the `establishConnection` Observable and subscribing the deferred Observable on a Scheduler that is single-threaded.

uKL, 2018-03-16:
We are going to deprecate `ConnectionSharingAdapter` and suggest usage of `ReplayingShare` in the future version.

uKL, 2018-03-19:
Updated documentation on a `ReplyingShare`.

closed 2018-03-19 by uKL

RAM512, 2018-05-03:
Can you add ReplyingShare and firstOrError information to Readme.md please?
I'm new in ReactiveX and it's hard for me to understand use cases. For example in [example4_characteristic/CharacteristicOperationExampleActivity](https://github.com/Polidea/RxAndroidBle/blob/master/sample/src/main/java/com/polidea/rxandroidble2/sample/example4_characteristic/CharacteristicOperationExampleActivity.java) you use `.firstOrError()` in `onReadClick()` and `onWriteClick()` but not in `onNotifyClick()`.

dariuszseweryn, 2018-05-13:
@RAM512 `.firstOrError()` is just a handy way to transform a `Observable<T>` to a `Single<T>`. For more information about `Observable` and/or `Single` I can encourage you to check for information directly on `RxJava`.

`ReplyingShare` is a helper (that should not be used imho) for using the same `RxBleConnection` in multiple places in the code while maintaining a single connection established (`RxBleDevice.establishConnection()` when unsubscribed closes the connection).

Best Regards