dariuszseweryn/RxAndroidBle #382: Disconnect is not really effective

GuillaumeBo, 2018-03-07:
I figure on my BLE device log that when i disconnect the smartphone with the unsubscribe/dispose it's not really disconnected.
On the smartphone side, it's marked as disconnected (even on the getConnectionState())
On the ble device it's still marked as connected (the disconnection is really effective when a i turn off bluetooth on the smartphone)
I thought it might be a specificity of my dev devices but when i use NRF Connect, the disconnection on the app works well (on the ble device i see instantly the disconnect event)

I tested with the last version 1.5 rx2 of the lib and with your sample app of last release 1.4.3
same result. 
Have you any information about it ?

dariuszseweryn, 2018-03-08:
`nRF Connect` has a bad habit of intercepting connections. I am not really sure how it is implemented but if it is in the background and a connection is established it holds it so it is not disconnected if the app disconnects. Test when the `nRF Connect` is not working (kill it so no notification is displayed)

dariuszseweryn, 2018-03-08:
Closed by mistake

GuillaumeBo, 2018-03-08:
I tryed with `nRF Connect` totally killed, and it's working ! Thanks you, I did not think that `nRF Connect` was so intrusive !

closed 2018-03-08 by dariuszseweryn