dariuszseweryn/RxAndroidBle #388: How to close gatt connection on writeCharacteristic and retry 2 times on write error?

Lzxe92, 2018-03-11:
```
       Subscription subscription = device.establishConnection(false)
                    .flatMap(rxBleConnection ->
                            rxBleConnection.writeCharacteristic(fromString("525255-0000-1000-8000-252525255"), hexToBytes(data)))
                    .subscribe(

                            characteristicValue -> {

                                Log.e(TAG, "write success " + characteristicValue + " " + device.getMacAddress());

                            },
                            throwable -> {
                                Log.e(TAG, "write error " + throwable);
                            }
                    );
```

I am using the above code to make an connection and write. However i will get " write error com.polidea.rxandroidble.exceptions.BleDisconnectedException: Disconnected from" if i connect to too many devices without disconnecting the previous connection. I am wondering how do i do that?

Also, how do i retry writeCharacteristic  if there is an error in writeCharacteristic ?

dariuszseweryn, 2018-03-11:
Hello!

Thank you for using the library.

This is a general usage question and as such should be posted on [stackoverflow](https://stackoverflow.com/questions/tagged/rxandroidble) with `rxandroidble` tag—it will be easier to find by other users.

Best Regards

closed 2018-03-11 by dariuszseweryn

Lzxe92, 2018-03-11:
do you know the way?

dariuszseweryn, 2018-03-11:
Yes. I suppose you have already asked [a question](https://stackoverflow.com/questions/49223195/how-to-unsubscribe-after-a-successful-writecharacteristic-with-rxandroidble) on Stackoverflow and Rob is already looking on your case