dariuszseweryn/RxAndroidBle #391: v. 1.5.0 compile errors in ClientComponent

RobLewis, 2018-03-14:
### Summary
A fresh clone of the just-released 1.5.0 has numerous "cannot resolve symbol" errors in the `ClientComponent` interface, apparently due to missing dependencies. 

### Library version
`1.5.0`

#### Preconditions
Clone the master repository to a new Android Studio project

#### Steps to reproduce actual result
View the errors in the editor for the interface `com.polidea.rxandroidble2.ClientComponent`

#### Minimum code snippet reproducing the issue
N/A
The `@Module(subcomponents...` in line 102 is in red, as are many of the `@Provides` and `@Named` annotations below it, as well as `Module` and `Binds` annotations. 

#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`
N/A

#### Actual result
Many compile errors

#### Expected result
no compile errors

RobLewis, 2018-03-15:
I checked and other classes have similar problems, including `ClientScope`, `RxBleClient`, `RxBleClientImpl`, and perhaps others.

uKL, 2018-03-15:
I'm unable to reproduce your issue. Please make sure that everything is right with your IDE. 

1. I cloned the repo to a new folder
2. Opened up Android Studio 3.0.1
3. Opened project
4. Built the project

Result:
Project build without issues.

Feel free to reopen with more data on the reproduction.

closed 2018-03-15 by uKL

RobLewis, 2018-03-15:
Perhaps the difference is that I am using Android Studio to clone the project from GitHub, whereas you presumably used git? It could then be a problem with Android Studio. 

Attempting to reproduce it now: 
1. In Android Studio 3.0.1 (Mac), select File | New | Project from Version Control | GitHub and enter the URL (https://github.com/Polidea/RxAndroidBle.git) and click Clone. 
2. Android Studio reports "Unsupported Modules Detected: Compilation is not supported for the following modules: RxAndroidBle2 [the name I gave the Project folder]. Unfortunately you can't have non-Gradle Java modules and Android-Gradle modules in one project." 
3. Then Android Studio logs "Gradle sync started", then "Project setup started", then the sync fails with a Null Pointer Exception. 
4. In the Project view sidebar (set to Android mode), the only folder that has a triangle (indicating it's not empty, I suppose) is `mockrxandroidble`. It contains subfolders `manifests` and `java`. 
5.  I select Build | Rebuild Project. It runs but there is no apparent change. 
6. I try File | Synchronize. Nothing happens. Same with Build | Clean Project. 
7. I try Build | Make Project. I get 200 errors: packages don't exist, or can't find symbol class (such as RxBleConnection). 
8. I close and reopen the project. Now the `rxandroidble` and `sample` folders show their contents, but not `dagger-compiler-shadow` or `dagger-library-shadow`. Opening up classes like `ClientComponent` in the editor gives the errors I originally described, I'm guessing because the dagger folders are apparently invisible. 
9. Closing and reopening the project again doesn't change anything. Neither does Clean Project. 
10. I try again to close the project. This time, it is now missing from the File | Open Recent menu. 

It seems likely that the problem is Android Studio failing for some reason to recognize the two Dagger folders. But I don't know what to do about this. 
I'll try it using git and see what happens.

RobLewis, 2018-03-15:
OK, I cloned the repo (using GitKraken) and opened it in Android Studio. I didn't have to ask it to build. I got some warnings, like removing RetroLambda (did that), remove the buildToolsVersion directive (did that). 
What remains is an error: "android-apt plugin is incompatible with the Android Gradle plugin. Please use 'annotationProcessor' configuration instead."

RobLewis, 2018-03-15:
I found this in the Android docs: 
When using the Android plugin 3.0.0, you must add annotation processors to the processor classpath using the annotationProcessor dependency configuration, as shown below:

```
dependencies {
    ...
    annotationProcessor 'com.google.dagger:dagger-compiler:<version-number>'
}
```
Is this applicable? What do I do?

RobLewis, 2018-03-15:
Think I figured it out: in `rxandroidble` build.gradle: 
Remove the `apply plugin: 'com.neenbedankt.android-apt'`
In `dependencies`, remove the `apt` line and insert
`annotationProcessor 'com.google.dagger:dagger-compiler:2.15'`

This gets rid of the last error; hopefully it works now.

RobLewis, 2018-03-15:
OK, apparently my last clone somehow got an old version of the repo, missing the RxJava2 version. I tried again, using GitKraken to clone the repo—I got the v.2 code, but I'm back to all the unrecognized things like `@Inject`, `@Named`, `@Provides`, `@Module`, `@Component`, `@Scope`, `DaggerClientComponent`, `Lazy`. 

Currently I am only getting these errors in the following items: `ClientComponent, ClientScope, RxBleAdapterStateObservable, RxBleClient,` and `RxBleClientImpl`. There may be others that I haven't noticed.

RobLewis, 2018-03-15:
Success? I'm hesitant to declare victory yet, but I just tried downloading the ZIP file and building the project from that, and it appears to have worked: v. 2 code is present, no errors. 
Mysterious.

RobLewis, 2018-03-16:
@uKL I would like to reopen this but apparently I can't since it was closed by you