dariuszseweryn/RxAndroidBle #392: Exception fired on non authorized Characteristic

GuillaumeBo, 2018-03-15:
I am using the rxble 1.5 with rx2 support
I am trying to handle when my smartphone is writing on characteristic that is not authorized (encrypted).
In theory we should receive a GATT_INSUFFICIENT_AUTHENTICATION or something like this but in my case i receive BleGattCallbackTimeoutException.
This might be due to the Pairing popup that appears on the device (to get access to the service after pairing) and i do not accept it volontary.
This automatic pairing request is made by the OS or the library?
Even if this mecanism of auto pairing request is here, the writeOperation should return an exception about the encrypted issue and not the pairing request ?

uKL, 2018-03-15:
This is auto pairing provided by Android. I've seen devices that were not sending **any** callback from the BLE stack if the popup wasn't accepted. For the library it is transparent and we treat it as a timeout. However, some more logs may be helpful.

The library does not support bonding you when you'd implement it on your side, you could listen for a broadcast that would tell you that the request was rejected by the user.

Hope it helps.

uKL, 2018-03-15:
Related to #35

GuillaumeBo, 2018-03-15:
Yes i implemented the bonding on my side and already with 2 differents devices (huawei and nexus, both Android 7) i got very different comportment... But that is not the issue here.
here are logs, not very helpfull :/
```
03-15 15:10:15.963 30807-30807/fr.pacotte.pa E/BleManager: Connected to 00:00:00:00:00:00
03-15 15:10:17.340 30807-30904/fr.pacotte.pa D/BLEConnectionService: writeCharacteristic on 00:00:00:00:00:00, with uuid : 00010000-0295-7A8B-1410-000000000000, and data : [B@4dd4042
03-15 15:10:17.347 30807-30907/fr.pacotte.pa D/BluetoothGatt: discoverServices() - device: 00:00:00:00:00:00
03-15 15:10:17.503 30807-30819/fr.pacotte.pa D/BluetoothGatt: onSearchComplete() = Device=00:00:00:00:00:00 Status=0
03-15 15:10:47.512 30807-30972/fr.pacotte.pa E/BLEConnectionService: writeCharacteristic 00010000-0295-7A8B-1410-000000000000 Unknown Error, code : GATT exception from MAC address 00:00:00:00:00:00, with type BleGattOperation{description='CHARACTERISTIC_WRITE'}
03-15 15:10:47.514 30807-30972/fr.pacotte.pa E/BleManager: writeCharacteristic failure on "00:00:00:00:00:00", error : GATT exception from MAC address 00:00:00:00:00:00, with type BleGattOperation{description='CHARACTERISTIC_WRITE'}
03-15 15:10:50.574 30807-30820/fr.pacotte.pa D/BluetoothGatt: onClientConnectionState() - status=22 clientIf=6 device=00:00:00:00:00:00
03-15 15:10:50.577 30807-30807/fr.pacotte.pa E/BLEConnectionService: Disconnection from 00:00:00:00:00:00, error : Disconnected from 00:00:00:00:00:00
03-15 15:10:50.580 30807-30907/fr.pacotte.pa D/BleManager: registerDevicesConnectionStatus, new value : DISCONNECTING
03-15 15:10:50.580 30807-30907/fr.pacotte.pa D/BluetoothManager: getConnectionState()
03-15 15:10:50.580 30807-30907/fr.pacotte.pa D/BluetoothManager: getConnectedDevices
03-15 15:10:50.584 30807-30907/fr.pacotte.pa D/BluetoothGatt: close()
```
Maybe OS occults the write operation and returns only the timeout error, but it's not very clean.. we got an error that is not linked to the original operation...

GuillaumeBo, 2018-03-15:
@dariuszseweryn your answer on #277 seems to talk about it ?

dariuszseweryn, 2018-03-16:
Could you paste the above log but with `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`?

GuillaumeBo, 2018-03-16:
Logs in verbose mode : 
```
03-16 10:31:18.638 25897-26006/fr.pacotte.pa D/BLEConnectionService: writeCharacteristic on 00:00:00:00:00:00, with uuid : 00010000-0200-0000-0000-000000000000, and data : [B@7526f0
03-16 10:31:18.645 25897-26006/fr.pacotte.pa D/RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(102613532)
03-16 10:31:18.646 25897-26009/fr.pacotte.pa D/RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(102613532)
03-16 10:31:18.649 25897-26011/fr.pacotte.pa D/BluetoothGatt: discoverServices() - device: 00:00:00:00:00:00
03-16 10:31:18.662 25897-25909/fr.pacotte.pa D/BluetoothGatt: onSearchComplete() = Device=00:00:00:00:00:00 Status=0
03-16 10:31:18.663 25897-25909/fr.pacotte.pa D/RxBle#BluetoothGatt: onServicesDiscovered status=0
03-16 10:31:18.664 25897-26010/fr.pacotte.pa V/RxBle#ServiceDiscoveryOperation: Preparing services description
03-16 10:31:18.688 25897-26010/fr.pacotte.pa V/RxBle#ServiceDiscoveryOperation: --------------- ====== Printing peripheral content ====== ---------------
                                                                                PERIPHERAL ADDRESS: 00:00:00:00:00:00
                                                                                PERIPHERAL NAME: myDevice
                                                                                -------------------------------------------------------------------------
                                                                                
                                                                                Primary Service - Generic Attribute (00001801-0000-1000-8000-00805f9b34fb)
                                                                                Instance ID: 1
                                                                                -> Characteristics:
                                                                                
                                                                                Primary Service - Generic Access (00001800-0000-1000-8000-00805f9b34fb)
                                                                                Instance ID: 64
                                                                                -> Characteristics:
                                                                                	* Device Name (00002a00-0000-1000-8000-00805f9b34fb)
                                                                                	  Properties: [ READ ]
                                                                                	* Appearance (00002a01-0000-1000-8000-00805f9b34fb)
                                                                                	  Properties: [ READ ]
                                                                                
                                                                                Primary Service - Device Information (0000180a-0000-1000-8000-00805f9b34fb)
                                                                                Instance ID: 256
                                                                                -> Characteristics:
                                                                                	* PnP ID (00002a50-0000-1000-8000-00805f9b34fb)
                                                                                	  Properties: [ READ ]
                                                                                
                                                                                Primary Service - Immediate Alert Service 1.1 (00001802-0000-1000-8000-00805f9b34fb)
                                                                                Instance ID: 512
                                                                                -> Characteristics:
                                                                                	* Alert Level (00002a06-0000-1000-8000-00805f9b34fb)
                                                                                	  Properties: [ WRITE_NO_RESPONSE ]
                                                                                
                                                                                Primary Service - Unknown service (000100ff-0200-0000-0000-000000000000)
                                                                                Instance ID: 515
                                                                                -> Characteristics:
                                                                                	* Unknown characteristic (00010000-0200-0000-0000-000000000000)
                                                                                	  Properties: [ WRITE ]
                                                                                	* Unknown characteristic (00010004-0200-0000-0000-000000000000)
                                                                                	  Properties: [ READ WRITE ]
                                                                                --------------- ====== Finished peripheral content ====== ---------------
03-16 10:31:18.694 25897-26010/fr.pacotte.pa D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(177048570)
03-16 10:31:18.697 25897-26009/fr.pacotte.pa D/RxBle#ConnectionOperationQueue: FINISHED ServiceDiscoveryOperation(102613532) in 50 ms
[... cut ...]
03-16 10:31:51.925 25897-26011/fr.pacotte.pa D/BluetoothManager: getConnectionState()
03-16 10:31:51.925 25897-26011/fr.pacotte.pa D/BluetoothManager: getConnectedDevices
03-16 10:31:51.928 25897-26011/fr.pacotte.pa D/BluetoothGatt: close()
03-16 10:31:51.928 25897-26011/fr.pacotte.pa D/BluetoothGatt: unregisterApp() - mClientIf=6
03-16 10:31:51.929 25897-26011/fr.pacotte.pa D/BleManager: registerDevicesConnectionStatus, new value : DISCONNECTED
03-16 10:31:51.931 25897-25937/fr.pacotte.pa D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(205977003) in 9 ms
```

uKL, 2018-03-16:
Well, that doesn't say much. Call callback wasn't executed at all so the timeout occurred.

GuillaumeBo, 2018-03-16:
The 30sec timeout is due to my non action or cancel the pairing modal (same log for theses 2 case) yes

uKL, 2018-03-16:
How would you see it working? Just imagine your perfect solution :)

GuillaumeBo, 2018-03-16:
The ideal solution for me would the writeCharacteristic Single triggers onError with something like BleNotAuthorizedException or BleInsuffisantAuthorizationException (refer to [Gatt API](https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h) ) or simple BleEncryptedCharacteristicException instead of the timeoutException, no?

uKL, 2018-03-16:
When user rejects or instantly when the user is asked to make a bonding?

dariuszseweryn, 2018-03-16:
Maybe the library could monitor Bonding Changes and if it would find one it would not trigger the timeout?

uKL, 2018-03-16:
I was thinking about it. Pausing the timeout and an onError when the user rejects.

GuillaumeBo, 2018-03-16:
Only when user reject or ignore the bonding process. If user accept bonding, i think the write operation is done like normal case.
Or to go further a param that allow either the "auto bonding request" or not

uKL, 2018-03-16:
We have no control if the system starts auto bonding or not. We can only request (with a classic BT scanning hack) the notification to be displayed in a form of a dialog, not a notification.

I've seen some phones that were failing the write after positive bonding, requiring it to be retried - so this is not as straightforward as one can think.

I think for the MVP we could try to listen for bonding related broadcasts and emit meaningful errors. I'll think about it.

GuillaumeBo, 2018-03-16:
Ok :)
Bad to hear that "some phones that were failing the write after positive bonding, requiring it to be retried". There are still a lot of work to do :(

uKL, 2018-03-16:
Yes, that's mostly the case why we didn't have bonding support from the beginning, it's tricky.

Let's follow up in #35.

closed 2018-03-16 by uKL