dariuszseweryn/RxAndroidBle #394: Documentation of establishConnection() exception & feature request

RobLewis, 2018-03-16:
Looking at the code for RxAndroidBle2, it appears that if `.establishConnection()` is called on a device that is already connected, a `BleAlreadyConnectedException` is thrown. This should probably be added to the Javadoc. 

I don't see any way to access an existing connection to a device. Could a method to retrieve it be added? Are there reasons not to have this? Would it make sense to have a method like `.establishOrReturnConnection()` that returns a connection if one already exists, or else establishes a new one?

dariuszseweryn, 2018-03-16:
Hello,
The error is [documented in Javadoc](https://github.com/Polidea/RxAndroidBle/blob/master/rxandroidble/src/main/java/com/polidea/rxandroidble2/exceptions/BleAlreadyConnectedException.java). My stance is that having all potential errors that may be emitted from a particular `Observable` would clutter the description and would not provide help. This kind of error is usually easy to spot during an early stage of the development process

It was a deliberate decision not to provide this kind of a function as the communication patterns are usually stateful (sending a request, waiting for a response) and if two places in the code would execute requests at the same time the responses could get mixed which would lead to hard to spot bugs.
If the user will use `.share()` or `.publish()` or `RxReplayingShare` then they are aware that the connection is being shared. Hopefully...

uKL, 2018-03-16:
Anyway, I agree that we could add `@throws BleAlreadyConnectedException` with some description to the `establishConnection` method.

commit 6b7b23b4 (2018-03-16): [#394] Added missing exception documentation, emitted by the establishConnection method. (#396)

FIxes #394
closed 2018-03-16 by uKL