dariuszseweryn/RxAndroidBle #398: Stopping device scanning and immediately connecting to device causes onClientConnParamsChanged() calls that appear to disrupt communications

NiallBegley, 2018-03-16:
### Summary

### Library version
`1.5.0`

#### Preconditions
Android app that starts scanning immediately after booting up and displays to the user a list of devices found.  In this scenario, the only devices shown are known microcontrollers running custom firmware.  Initial BLE scanning ends after 5 seconds.  User can re-initiate scanning if desired.

Problem seen on Samsung Note 4 running Android 6.0.1

#### Steps to reproduce actual result
<br />1. Scan for devices
<br />2. Before 5 second scanning timeout is reached, when the user selects a device from the list to connect to, unsubscribe from scan Subscription
<br />3. Immediately connect to device and begin writing data

#### Minimum code snippet reproducing the issue
Starting scan:

 ```
scanSubscription = rxBleClient.scanBleDevices(
                new ScanSettings.Builder()
                         .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                        .build(), new ScanFilter.Builder().setManufacturerData(Constants.BLE_MANUFACTURER_ID, Constants.BLE_MANUFACTURER_DATA).build()
        ).subscribe(
                        scanResult -> {
                            //Custom code to populate list adapter
                        },
                        throwable -> {
                            throwable.printStackTrace();
                        }
                );
```

User selects BLE device:

```
        scanSubscription.unsubscribe();

         //This activity *immediately* starts connecting to the BLE device
          Intent myIntent = new Intent(MainActivity.this, ConnectedActivity.class);
            myIntent.putExtra("address", bleobject.getAddress());
            myIntent.putExtra("name", bleobject.getName());
            MainActivity.this.startActivityForResult(myIntent, CONNECTED_ACTIVITY);
```

#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`
See attached file.  I've tried to put some comments in there to make what is happening more clear.  Search for "NIALL"
[log.txt](https://github.com/Polidea/RxAndroidBle/files/1820376/log.txt)

#### Actual result
When immediately connecting to the device after stopping the subscription, there's about a 70% chance (depends on timing) that something happens causing the follow lines to show up and put a halt to all communication.  Note that the connection isn't severed, but any pending writes or reads are thrown out:

```
03-16 13:19:38.371 3598-3659/x.com.yD/BluetoothGatt: onClientConnParamsChanged() - Device=00:A0:50:65:24:91 interval=39 status=0
03-16 13:19:39.011 3598-3609/x.com.y D/BluetoothGatt: onClientConnParamsChanged() - Device=00:A0:50:65:24:91 interval=6 status=0
03-16 13:19:39.311 3598-3609/x.com.yD/BluetoothGatt: onClientConnParamsChanged() - Device=00:A0:50:65:24:91 interval=39 status=0
```

I've found that putting in a 500 ms delay between the time the subscription is cancelled and actually starting BLE communications reliably avoids this issue.  Obviously this isn't ideal, as that amount of time could (and does) vary by device.

The key seems to not start communications until the following lines appear in the log:

```
3-16 13:19:37.421 6292-6338/? D/BtGatt.GattService: [GSIM LOG]: getCallingProcessName: x.com.y
03-16 13:19:37.421 6292-6338/? D/BtGatt.GattService: [GSIM LOG]: getCallingProcessName: x.com.y
03-16 13:19:37.421 6292-6827/? E/BtGatt.GattService: [GSIM LOG]: gsimLogHandler: x.com.y, msg: MESSAGE_STOP_SCAN
03-16 13:19:37.421 6292-6827/? D/BtGatt.GattService: [GSIM LOG]: x.com.y@LowLatency, R time : 5 => 5
03-16 13:19:37.421 6292-6827/? D/BtGatt.GattService: [GSIM LOG]: x.com.y@LowLatency, A time : 5 => 5
03-16 13:19:37.421 6292-6827/? D/BtGatt.GattService: [GSIM LOG]: remove : x.com.yat  BLE_SCAN_ACTUAL_DB
03-16 13:19:37.421 6292-6827/? D/BtGatt.GattService: [GSIM LOG]: remove : x.com.yat  BLE_SCAN_REQUESTED_DB
```
#### Expected result
I would expect that unsubscribing from the scanning subscription and then connecting to my device immediately afterwards would have no detrimental affect on anything.  I've found that *not* cancelling the subscription and just jumping immediately into connecting avoids this particular problem, however, I don't really want my application to be scanning for BLE devices the entire time it is open.

dariuszseweryn, 2018-03-16:
Hello,

I have followed the log you have attached. Could you elaborate more on what is the actual problem, what would you expect to happen next? In the logs you have attached I see that the library is finishing the Long Write Op normally and there is no other action on it (no further requests nor any errors)

The connection params dance is an implementation of a particular BLE stack which is beyond the control of the library (or Android API)

Are you sure sure that your peripheral is working correctly?

uKL, 2018-03-19:
Do you observe it on a Huawei or Samsung? I've seen similar issues before, especially on BLE devices with complex characteristics.

NiallBegley, 2018-03-19:
@uKL I have limited devices to test on but I've only seen it on my old Samsung Note 4.  Haven't seen it on a co-workers Nexus 6 or OnePlus Three.  I'm fairly new to BLE so i'm a little inexperienced, but I don't believe my GATT characteristics are particularly complex.

@DariuszAniszewski Sure, let me elaborate a little bit more.  The interaction between the phone and the microprocessor for this project is supposed to be a fairly long back and forth authentication process.  The phone has to perform a long write to the microprocessor to send some data for the microprocessor to authenticate and, when done, the microprocessor then has to send a similar amount of data back to the phone.  The phone and the microprocessor then do a few more writes and reads back and forth to generate a shared secret before the microprocessor sends over an encrypted payload.  Once the phone receives the payload the connection is automatically severed.

So you'll see in the log file that the first time I perform this flow (when it worked correctly) I wait a few seconds after the BLE scanning stopped before I initiated the connection to the microprocessor.  Once the connection is made the phone performs a long write (line 4771).  In this project I have to write out a quick 5 bytes indicating the length of the outgoing message to the microprocessor so it knows how much to expect, which is why you see a 5 byte write at line 4783 followed by the full long write at 4821.  After that you can see at line 4917 that the microprocessor sends back a message indicating it has finished processing what the phone had previously sent and we're free to move on to the next step.  I won't go through every step here, but you get the idea.  We go back and forth a few more times before you see the payload is sent and the connection is automatically disconnected (5598).

Now, on the second run, I select the device to connect to from the phone *before* the BLE scanning is complete and you see that I unsubscribe from the scanning subscription before immediately launching into the same flow as described above (line 7095).  Even though I've requested the scanning stop and I've initiated the connection with the microprocessor, you can see that the BLE scanning appears to still be happening (7098, 7248).  Scanning appears to actually stop around line 7307, the first write out is requested at 7569 and starting at line 7668 you can see what I believe is causing an issue - some sort of change in what I assume is the connection interval:

```
03-16 14:09:11.812 15563-15574/xyz.com.securitytagreader D/BluetoothGatt: onClientConnParamsChanged() - Device=00:A0:50:65:24:91 interval=6 status=0
03-16 14:09:12.182 6292-6754/? D/BtGatt.GattService: onClientConnParamsChanged() - clientIf=9 address=00:A0:50:65:24:91, interval=39status=0
```

After this happens the connection is completely dead in the water because it appears to have interrupted  the previous write out which I never see arrive on the microprocessor side.  I should quickly point out that this interrupt happened at this point in the back and forth flow but it can literally happen at any other point with similar results, it's all dependent on the timing of when the scan stops in the background.

Regarding your comment about the peripheral causing this issue: that definitely occurred to me but based on the connection working fine with other phones and the fact that waiting for scanning to truly finish before initiating the connection seems to correct the issue makes me lean towards thinking it is an issue on the phone side of things.

Finally to summarize a little bit, I'm not particularly expecting a fix to this issue since I'm feeling increasingly like it's specific to this phone but I figured I'd put it out there in case there is a more reliable way to detect the end of the scanning and/or if my hypothesis about that being the root of the issue is even plausible.

DariuszAniszewski, 2018-03-19:
cc: @dariuszseweryn :)

uKL, 2018-03-23:
Unfortunately, I do not have any solution for you yet. I've seen similar issues (on Huawei) and from the BLE Snoop Log, downloaded from the phone I was able to see that just after the connection is established, Android tries to do a service discovery, traversing all the services, characteristics and descriptors. 

If I did any write operation just after during that period, the connection was no longer operational...

`onClientConnParamsChanged` is automatically triggered by Android and we have limited control over it.

It would be hard to get any ideas without the device to test on. Maybe @dariuszseweryn will have some more thoughts.

toaderandrei, 2018-03-27:
@uKL , is this the same issue I had?(ref: Huawei P9).

uKL, 2018-03-27:
Not really, as far as I remember your case was related to establishing the connection, not the connection becoming unresponsive.

toaderandrei, 2018-03-27:
Well, technically yes, however there was a sequence of commands, one was connect, then discover services. Our cases was that connect was somewhat successful(in the sense no exception was received immediately) then we were doing the discover services call and after 20 seconds the connection was dropped.

uKL, 2018-03-27:
Were you receiving a positive connection callback? AFAIK no, correct me if I'm wrong.

toaderandrei, 2018-03-27:
Well, we were getting an exception after 20 seconds.

uKL, 2018-06-04:
We suspect that this problem may be on a peripheral side or the Android side. We ran out of ideas. If you'd like to debug more, feel free. We'd be happy to introduce a fix if you open a PR.

closed 2018-06-04 by uKL