dariuszseweryn/RxAndroidBle #403: scanBleDevices using with kotlin cause compile error

pfaffenrodt, 2018-03-21:
### Disclaimer
Before you will post a new issue to the RxAndroidBle library consider if you are filling a bug/feature request or a general question 
about:
* Usage of RxJava
* Usage of Bluetooth Low Energy
* Usage of RxAndroidBle
* Weird behaviour of the peripheral you are using
* Other not directly related to a new feature request or a bug

If any of the above points seems to match your use case — consider using Google or creating a question on www.stackoverflow.com with 
a tag `rxandroidble` where it will be easier to access by other people with similar questions. Issues that are not bugs or feature requests
will be closed.

### Summary
Using scanBleDevices with Kotlin we have an compile error.

I think its because both "scanBleDevices " methods expecting variable number of arguments.

Deleting the already deprecated method will fix this issue.

### Library version
1.5.0 (that uses rxjava 2)

#### Preconditions
kotlin_version = '1.2.30'
implementation "com.polidea.rxandroidble2:rxandroidble:1.5.0"
#### Steps to reproduce actual result
1. Try to write code to scan devices
#### Minimum code snippet reproducing the issue
val rxBleClient by lazy { RxBleClient.create(this) }
val scanSettings : ScanSettings = ScanSettings.Builder().build()
rxBleClient.scanBleDevices(scanSettings, ScanFilter.empty())

#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`
> None of the following functions can be called with the arguments supplied.
> scanBleDevices(ScanSettings!, vararg ScanFilter!) defined in com.polidea.rxandroidble2.RxBleClient
> scanBleDevices(vararg UUID!) defined in com.polidea.rxandroidble2.RxBleClient

#### Actual result
Can not compile.
#### Expected result
It should compile.

pfaffenrodt, 2018-03-21:
Sorry. Wrong import of ScanSettings.
I am so stupid.

closed 2018-03-21 by pfaffenrodt

uKL, 2018-03-21:
Glad you did it :D

standinga, 2018-10-05:
Had the same problem:) because of auto imported by Android Studio:) This is why I like my api classes prefixed.

honey-system, 2018-12-21:
> Sorry. Wrong import of ScanSettings.
> I am so stupid.

You should have written how you solved the problem!