dariuszseweryn/RxAndroidBle #404: Crash due to LocationServicesOkObservableApi23 somewhere has null

warko-san, 2018-03-21:
Hi guys! Thank you for that awesome lib you've created! 
I was migrating from com.polidea.rxandroidble:rxandroidble:1.4.3 to com.polidea.rxandroidble2:rxandroidble:1.5.0 recently and faced troubles on APIs > 22. Seems that inside a library there is some nulls passed to observables, but RxJava2 does not allow this. So I can't use it any more. 
The thing is crash sometimes happens when I simply try to obtain RxBleClient instance.
I would really appreciate your help. Thanks in advance! 
### Summary

### Library version
`1.5.0`

#### Logs from the application (crash report)
`03-21 19:53:04.561 29366-29533/com.vue E/AndroidRuntime: FATAL EXCEPTION: RxComputationThreadPool-1
                                                         Process: com.vue, PID: 29366
                                                         java.lang.NullPointerException: Attempt to invoke interface method 'boolean io.reactivex.internal.fuseable.SimpleQueue.offer(java.lang.Object)' on a null object reference
                                                             at io.reactivex.internal.operators.observable.ObservableSwitchMap$SwitchMapInnerObserver.onNext(ObservableSwitchMap.java:372)
                                                             at io.reactivex.internal.operators.observable.ObservableMap$MapObserver.onNext(ObservableMap.java:64)
                                                             at com.polidea.rxandroidble2.internal.util.LocationServicesOkObservableApi23.subscribeActual(LocationServicesOkObservableApi23.java:35)
                                                             at io.reactivex.Observable.subscribe(Observable.java:12005)
                                                             at io.reactivex.internal.operators.observable.ObservableMap.subscribeActual(ObservableMap.java:33)
                                                             at io.reactivex.Observable.subscribe(Observable.java:12005)
                                                             at io.reactivex.internal.operators.observable.ObservableSwitchMap$SwitchMapObserver.onNext(ObservableSwitchMap.java:127)
                                                             at io.reactivex.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver.drain(ObservableConcatMap.java:464)
                                                             at io.reactivex.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver.onSubscribe(ObservableConcatMap.java:324)
                                                             at io.reactivex.internal.operators.observable.ObservableFromArray.subscribeActual(ObservableFromArray.java:30)
                                                             at io.reactivex.Observable.subscribe(Observable.java:12005)
                                                             at io.reactivex.internal.operators.observable.ObservableConcatMap.subscribeActual(ObservableConcatMap.java:54)
                                                             at io.reactivex.Observable.subscribe(Observable.java:12005)
                                                             at io.reactivex.internal.operators.observable.ObservableSwitchMap.subscribeActual(ObservableSwitchMap.java:51)
                                                             at io.reactivex.Observable.subscribe(Observable.java:12005)
                                                             at io.reactivex.internal.operators.observable.ObservableSkip.subscribeActual(ObservableSkip.java:29)
                                                             at io.reactivex.Observable.subscribe(Observable.java:12005)
                                                             at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.subscribeInner(ObservableFlatMap.java:165)
                                                             at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.onNext(ObservableFlatMap.java:139)
                                                             at io.reactivex.internal.observers.DeferredScalarDisposable.complete(DeferredScalarDisposable.java:82)
                                                             at io.reactivex.internal.operators.single.SingleToObservable$SingleToObservableObserver.onSuccess(SingleToObservable.java:73)
                                                             at io.reactivex.internal.operators.single.SingleMap$MapSingleObserver.onSuccess(SingleMap.java:64)
                                                             at io.reactivex.internal.operators.observable.ObservableCountSingle$CountObserver.onComplete(ObservableCountSingle.java:83)
                                                             at io.reactivex.internal.operators.observable.ObservableTakeWhile$TakeWhileObserver.onNext(ObservableTakeWhile.java:86)
                                                             at io.reactivex.internal.operators.observable.ObservableInterval$IntervalObserver.run(ObservableInterval.java:83)
                                                             at io.reactivex.internal.schedulers.ScheduledDirectPeriodicTask.run(ScheduledDirectPeriodicTask.java:38)
                                                             at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:423)
                                                             at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:278)
                                                             at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:154)
                                                             at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:270)
                                                             at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
                                                             at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
                                                             at java.lang.Thread.run(Thread.java:818)`

#### Actual result
App crashes on start

#### Expected result
App does not crash on start

uKL, 2018-03-21:
Thanks, I'll have a look at it.

warko-san, 2018-03-21:
@uKL I forgot to mention that ACCESS_COARSE_LOCATION previously was granted

michaelnimbs, 2018-03-22:
I have exactly the same issue and it happens if I subscribe to observeStateChanges().
The NPE happens here for me: https://github.com/Polidea/RxAndroidBle/blob/506d98cd662fb81c9450c1574b28701539329071/rxandroidble/src/main/java/com/polidea/rxandroidble2/internal/util/LocationServicesOkObservableApi23.java#L35

`locationProviderOk` is true

uKL, 2018-03-23:
@warko-san @michaelnimbs What is your RxJava2 version in a build.gradle?

uKL, 2018-03-23:
I'm seeing this issue with RxJava 2.1.11 (https://github.com/ReactiveX/RxJava/commit/8a6bf14fc9a61f7c1c0016ca217be02ca86211d2) this commit changed how the queue is initialized. For now, I suggest staying with RxJava 2.1.10. I'll try to find out why it happens and submit an issue to RxJava repo.

michaelnimbs, 2018-03-23:
I can confirm this.
Works with RxJava 2.1.10 and breaks with 2.1.11.

uKL, 2018-03-23:
I'll work around this. The issue is that some of our custom Observables do not call `onSubscribe` on an Observer in a specific order. According to a [contract](http://reactivex.io/documentation/contract.html) the onSubscribe call is optional. What's more
```
An Observable may begin issuing notifications to an observer immediately after the Observable receives a Subscribe notification from the observer.
```
which assumes that the Observable may start emitting within the `subscribeActual` method.

@dariuszseweryn I'm pushing it directly to master without a PR due to criticalness. Please have a look once you're back. :)

uKL, 2018-03-23:
@michaelnimbs Please retest with the most recent snapshot (currently building) and the newest RxJava.

closed 2018-03-23 by uKL

michaelnimbs, 2018-03-23:
I guess this would be 1.6.0-SNAPSHOT and not 1.5.0-SNAPSHOT?

uKL, 2018-03-23:
https://oss.sonatype.org/content/repositories/snapshots/com/polidea/rxandroidble2/rxandroidble/1.6.0-SNAPSHOT/

Correct.

michaelnimbs, 2018-03-23:
Works with 1.6.0-SNAPSHOT and RxJava 2.1.11, thanks!

ipodib, 2018-03-29:
When 1.6.0 will be released, since fix that exist in 1.6.0 is critical for me, but I don't want to have SNAPSHOT version. Any thoughts?

thuytrinh, 2018-05-17:
I also encountered this crash today w/ RxAndroidBle `1.50` and RxJava `2.1.13`:

```
05-17 07:59:52.448 32740-493/bike.cobi.app.debug E/AndroidRuntime: FATAL EXCEPTION: RxComputationThreadPool-1
    Process: bike.cobi.app.debug, PID: 32740
    java.lang.NullPointerException: Attempt to invoke interface method 'boolean io.reactivex.internal.fuseable.SimpleQueue.offer(java.lang.Object)' on a null object reference
        at io.reactivex.internal.operators.observable.ObservableSwitchMap$SwitchMapInnerObserver.onNext(ObservableSwitchMap.java:372)
        at io.reactivex.internal.operators.observable.ObservableMap$MapObserver.onNext(ObservableMap.java:64)
        at com.polidea.rxandroidble2.internal.util.LocationServicesOkObservableApi23.subscribeActual(LocationServicesOkObservableApi23.java:35)
        at io.reactivex.Observable.subscribe(Observable.java:12025)
        at io.reactivex.internal.operators.observable.ObservableMap.subscribeActual(ObservableMap.java:33)
        at io.reactivex.Observable.subscribe(Observable.java:12025)
        at io.reactivex.internal.operators.observable.ObservableSwitchMap$SwitchMapObserver.onNext(ObservableSwitchMap.java:127)
        at io.reactivex.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver.drain(ObservableConcatMap.java:464)
        at io.reactivex.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver.onSubscribe(ObservableConcatMap.java:324)
        at io.reactivex.internal.operators.observable.ObservableFromArray.subscribeActual(ObservableFromArray.java:30)
        at io.reactivex.Observable.subscribe(Observable.java:12025)
        at io.reactivex.internal.operators.observable.ObservableConcatMap.subscribeActual(ObservableConcatMap.java:54)
        at io.reactivex.Observable.subscribe(Observable.java:12025)
        at io.reactivex.internal.operators.observable.ObservableSwitchMap.subscribeActual(ObservableSwitchMap.java:51)
        at io.reactivex.Observable.subscribe(Observable.java:12025)
        at io.reactivex.internal.operators.observable.ObservableSkip.subscribeActual(ObservableSkip.java:29)
        at io.reactivex.Observable.subscribe(Observable.java:12025)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.subscribeInner(ObservableFlatMap.java:165)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.onNext(ObservableFlatMap.java:139)
        at io.reactivex.internal.observers.DeferredScalarDisposable.complete(DeferredScalarDisposable.java:82)
        at io.reactivex.internal.operators.single.SingleToObservable$SingleToObservableObserver.onSuccess(SingleToObservable.java:73)
        at io.reactivex.internal.operators.single.SingleMap$MapSingleObserver.onSuccess(SingleMap.java:64)
        at io.reactivex.internal.operators.observable.ObservableCountSingle$CountObserver.onComplete(ObservableCountSingle.java:83)
        at io.reactivex.internal.operators.observable.ObservableTakeWhile$TakeWhileObserver.onNext(ObservableTakeWhile.java:86)
        at io.reactivex.internal.operators.observable.ObservableInterval$IntervalObserver.run(ObservableInterval.java:83)
        at io.reactivex.internal.schedulers.ScheduledDirectPeriodicTask.run(ScheduledDirectPeriodicTask.java:38)
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:423)
        at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:278)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:270)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
        at java.lang.Thread.run(Thread.java:818)
```

Here comes a sample which I was able to reproduce the issue:

```kotlin
fun scan(): Observable<ScanResult> = rxBleClient.scanBleDevices(
        ScanSettings.Builder()
                .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                .setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
                .build()
)

fun scanWithOptionalTimeout(shouldTimeout: Boolean): Observable<ScanResult> = when {
    shouldTimeout -> scan().timeout(ScanTimeoutInSecs, TimeUnit.SECONDS, schedulerFactory.io)
    else -> scan()
}

val scanSubscription = rxBleClient.observeStateChanges()
        .observeOn(schedulerFactory.io)
        .filter { it == RxBleClient.State.READY }
        .switchMap {
            scanWithOptionalTimeout(shouldTimeout = autoStop)
                    .map<ScanState> { DeviceDiscovered(it) }
                    .onErrorResumeNext { error: Throwable ->
                        when (error) {
                            is TimeoutException -> Observable.just(Timeout)
                            else -> Observable.error(error)
                        }
                    }
        }
        .startWith(AwaitingToBeReady)
        .doOnNext { Log.d(TAG, "scanState -> $it") }
        .subscribeBy(onNext = {
            when (it) {
                is DeviceDiscovered -> printScanResult(it.scanResult)
            }
        })

// ...
// Then later we decide to dispose `scanSubscription`,
// The crash will happen.
scanSubscription.dispose()
```

thuytrinh, 2018-06-05:
Hey @uKL, we encountered again this crash w/ v1.6.0:

```
Android: 8.1.0
Manufacturer: Google
Model: Pixel
Date: Tue Jun 05 15:38:18 GMT+02:00 2018

java.lang.NullPointerException: Attempt to invoke interface method 'boolean io.reactivex.internal.fuseable.SimpleQueue.offer(java.lang.Object)' on a null object reference
	at io.reactivex.internal.operators.observable.ObservableSwitchMap$SwitchMapInnerObserver.onNext(ObservableSwitchMap.java:372)
	at io.reactivex.internal.operators.observable.ObservableMap$MapObserver.onNext(ObservableMap.java:64)
	at com.polidea.rxandroidble2.internal.util.LocationServicesOkObservableApi23.subscribeActual(LocationServicesOkObservableApi23.java:54)
	at io.reactivex.Observable.subscribe(Observable.java:12051)
	at io.reactivex.internal.operators.observable.ObservableMap.subscribeActual(ObservableMap.java:33)
	at io.reactivex.Observable.subscribe(Observable.java:12051)
	at io.reactivex.internal.operators.observable.ObservableSwitchMap$SwitchMapObserver.onNext(ObservableSwitchMap.java:127)
	at io.reactivex.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver.drain(ObservableConcatMap.java:464)
	at io.reactivex.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver.onSubscribe(ObservableConcatMap.java:324)
	at io.reactivex.internal.operators.observable.ObservableFromArray.subscribeActual(ObservableFromArray.java:30)
	at io.reactivex.Observable.subscribe(Observable.java:12051)
	at io.reactivex.internal.operators.observable.ObservableConcatMap.subscribeActual(ObservableConcatMap.java:54)
	at io.reactivex.Observable.subscribe(Observable.java:12051)
	at io.reactivex.internal.operators.observable.ObservableSwitchMap.subscribeActual(ObservableSwitchMap.java:51)
	at io.reactivex.Observable.subscribe(Observable.java:12051)
	at io.reactivex.internal.operators.observable.ObservableSkip.subscribeActual(ObservableSkip.java:29)
	at io.reactivex.Observable.subscribe(Observable.java:12051)
	at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.subscribeInner(ObservableFlatMap.java:165)
	at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.onNext(ObservableFlatMap.java:139)
	at io.reactivex.internal.observers.DeferredScalarDisposable.complete(DeferredScalarDisposable.java:82)
	at io.reactivex.internal.operators.single.SingleToObservable$SingleToObservableObserver.onSuccess(SingleToObservable.java:73)
	at io.reactivex.internal.operators.single.SingleMap$MapSingleObserver.onSuccess(SingleMap.java:64)
	at io.reactivex.internal.operators.observable.ObservableCountSingle$CountObserver.onComplete(ObservableCountSingle.java:83)
	at io.reactivex.internal.operators.observable.ObservableTakeWhile$TakeWhileObserver.onNext(ObservableTakeWhile.java:86)
	at io.reactivex.internal.operators.observable.ObservableInterval$IntervalObserver.run(ObservableInterval.java:83)
	at io.reactivex.internal.schedulers.ScheduledDirectPeriodicTask.run(ScheduledDirectPeriodicTask.java:38)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:457)
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:307)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:302)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
	at java.lang.Thread.run(Thread.java:764)
```

Apparently you tried to fix that by moving `context.registerReceiver()` to the bottom in https://github.com/Polidea/RxAndroidBle/commit/1502ed62b4587627bfff402574e5b1739482bc5a. But https://github.com/Polidea/RxAndroidBle/commit/d3197ebf15044d8369763c385bea982253fcb0ff moved it back, thus causing regression right?

thuytrinh, 2018-06-05:
@uKL If that's right, I'm wondering we can compromise like you guys did in `RxBleAdapterStateObservable`:

```java
         observer.onSubscribe(Disposables.fromAction(new Action() {
             @Override
             public void run() throws Exception {
                 try {
                     context.unregisterReceiver(receiver);
                 } catch (IllegalArgumentException exception) {
                     Log.d(TAG, "The receiver is already not registered.");
                 }
             }
         }));
         context.registerReceiver(receiver, createFilter());
```

thuytrinh, 2018-06-07:
@uKL Would you like me to open a new pull request?

thuytrinh, 2018-06-07:
@uKL I tried to revert back to your original change like below and also ignored the `IllegalStateException`:

```java
        observer.onSubscribe(Disposables.fromAction(new Action() {
            @Override
            public void run() throws Exception {
                try {
                    context.unregisterReceiver(broadcastReceiver);
                } catch (IllegalStateException ignored) {
                }
            }
        }));
        observer.onNext(locationProviderOk);
        context.registerReceiver(broadcastReceiver, new IntentFilter(LocationManager.MODE_CHANGED_ACTION));
```

However, the following test failed:

```groovy
    def "should not register receiver again after unregister"() {
        given:
        mockLocationServicesStatus.isLocationProviderOk() >> isLocationProviderOkResult

        when:
        objectUnderTest.take(1).test()

        then:
        1 * contextMock.unregisterReceiver(_)

        then:
        0 * contextMock.registerReceiver(_, _)

        where:
        isLocationProviderOkResult << [true, false]
    }
```

That means `registerReceiver()` is still called again and thus we ended up having a receiver never being unregistered. Am I missing something?

thuytrinh, 2018-06-07:
@uKL I was tempted to check whether there's any subscriber before invoking `context.registerReceiver()` in the `subscribeActual()` body. However, apparently there isn't any way to perform that check. If the `Observable.create()` is used to create the `Observable`, I can check via the [`ObservableEmitter`](http://reactivex.io/RxJava/javadoc/io/reactivex/ObservableEmitter.html).

I'm wondering implementing `subscribeActual()` like in `LocationServicesOkObservableApi23` is a proper way to create the `Observable` in this case, rather than using `Observable.create()`?

> If you don't write an operator, you don't have to worry about subscribeActual(). – akarnokd

In https://stackoverflow.com/q/46250673/2563009. `LocationServicesOkObservableApi23` doesn't seem to create a new RxJava operator, so I guess we can replace it with the `Observable.create()` approach, right? BTW `akarnokd` is @akarnokd.

thuytrinh, 2018-06-07:
@uKL Here is my take on the `Observable.create()` approach:

```java
@TargetApi(Build.VERSION_CODES.KITKAT)
public class LocationServicesOkObservableApi23Factory {
    private final Context context;
    private final LocationServicesStatus locationServicesStatus;

    @Inject
    LocationServicesOkObservableApi23Factory(
            final Context context,
            final LocationServicesStatus locationServicesStatus) {
        this.context = context;
        this.locationServicesStatus = locationServicesStatus;
    }

    public Observable<Boolean> get() {
        return Observable.create(new ObservableOnSubscribe<Boolean>() {
            @Override
            public void subscribe(final ObservableEmitter<Boolean> emitter) throws Exception {
                final boolean initialValue = locationServicesStatus.isLocationProviderOk();
                final BroadcastReceiver broadcastReceiver = new BroadcastReceiver() {
                    @Override
                    public void onReceive(Context context, Intent intent) {
                        final boolean newValue = locationServicesStatus.isLocationProviderOk();
                        emitter.onNext(newValue);
                    }
                };
                emitter.setCancellable(new Cancellable() {
                    @Override
                    public void cancel() throws Exception {
                        try {
                            context.unregisterReceiver(broadcastReceiver);
                        } catch (IllegalStateException ignored) {
                        }
                    }
                });
                if (!emitter.isDisposed()) {
                    emitter.onNext(initialValue);
                    context.registerReceiver(broadcastReceiver, new IntentFilter(LocationManager.MODE_CHANGED_ACTION));
                }
            }
        }).distinctUntilChanged();
    }
}
```

Wdyt?

akarnokd, 2018-06-07:
Calling  onSubscribe is mandatory in RxJava 2.x. Also the preferred way of creating bridges to no-rx sources is Observable.create() which has safeguards guarding against this type of mistakes.

dariuszseweryn, 2018-06-07:
In RxJava 1.x there were helpers like `OnSubscribeCreate`. Unfortunately now they are unavailable (as far as I know) and the only way of creating a subclass of an `Observable` is the current approach. Creating a factory makes it also more cumbersome to use in the dependency injection, but it is not a show-stopper.

The thing is that we have two constraints we have to follow:
1. The `observer.onSubscribe()` has to be called before `observer.on*(any other)`
2. The `context.registerReceiver()` has to be called before `context.unregisterReceiver()`

The `context.unregisterReceiver()` is bound to `observer.onSubscribe()` so we have a chicken and an egg problem. What should be used is probably the [defferred cancellation](https://github.com/ReactiveX/RxJava/wiki/Writing-operators-for-2.0#deferred-cancellation) — correct me if I am wrong.

Or maybe there is some other solution?

akarnokd, 2018-06-07:
http://reactivex.io/RxJava/2.x/javadoc/io/reactivex/Observable.html#create-io.reactivex.ObservableOnSubscribe-

dariuszseweryn, 2018-06-07:
...that allows subclassing? ;)

akarnokd, 2018-06-07:
Why would you subclass it?

dariuszseweryn, 2018-06-07:
Convenience I suppose:
- While using RxJava 1 I could do something like:
```java
class MyFancyObservable extends Observable<MyType> {
  MyFancyObservable(SomeDependency someDependency) {
    super(new OnSubscribeCreate<>(new Action<Emitter<MyType>>() {
      // ... do my fancy stuff like I was using Observable.fromEmitter() but still having a subclass
    }));
  }
}
```
The convenience was that I could then just inject `MyFancyObservable` instead of `Observable<MyType>` and wire it to a specific key in case there are many observables of that type.

The other reason is to learn on what should I be aware in the case of RxJava 2.

commit 1502ed62 (2018-03-23): Make sure that onSubscribe is called before any emission is made by a custom Observable. Otherwise we may expect unexpected behaviors like a NPE in a switchMap.

Fixes #404