dariuszseweryn/RxAndroidBle #416: Configuration change - new feature

kosancicivan, 2018-04-12:
Maybe you could make that connection survive configuration change

dariuszseweryn, 2018-04-12:
Hello @kosancicivan 

Thank you for the idea. We have evaluated it at the beginning of the project and decided to concentrate on handling the BLE stack the best possible way.

Configuration changes are easy to overcome once you will separate the connection lifecycle from the activity lifecycle (which is usually what you want to do anyway). You can do it by using a `Service` or creating a different lifecycle scope which will suit your needs.

This topic is not directly related to BLE and there are different approaches depending on the specific use-case.

Best Regards

closed 2018-04-12 by dariuszseweryn

kosancicivan, 2018-04-13:
Best Regards