dariuszseweryn/RxAndroidBle #418: observeConnectionStateChanges

GuillaumeBo, 2018-04-20:
I am facing issue, the observeConnectionStateChanges seems to not emit anything when the screen is off. I am using rxBle in a service so the device is not necessarily active. 
Is it a wanted behaviour by devteam ?

version : last 1.6 snapshot rx2

dariuszseweryn, 2018-04-20:
There is no code related to the UI in the library. Are you sure that the app does not get suspended and/or service killed?

GuillaumeBo, 2018-04-20:
Yes i am sure, my routine advertise scan is still on, the connection is effective.
If i deploy from AndroidStudio device On -> it's works
if i deploy from AS device screen off -> no connection events emitted but everything else works well

dariuszseweryn, 2018-04-20:
Could you share a sample code which I could copy-paste to recreate the issue?
Also please write your setup (phone model/ os/ etc)

GuillaumeBo, 2018-04-27:
Sorry i was busy, it's to difficult to extract this part of the code, but after many tests, the issue come from my code cause it appears randomly even if app is waked up.
You can close the ticket :)

closed 2018-04-27 by dariuszseweryn