dariuszseweryn/RxAndroidBle #422: Multiple Activities connections

LinkOut, 2018-04-26:
Hello, i'm correctly using RxAndroidBle to send/receive data... the problem becomes when i try to connect again from a new activity...

This is the scenario:
I do BLE operations on my MainActivity, then i open a new one, from the old activity i diconnect. If i try to connect back it gives me errors, and if the second activity is closed, the Main wont connect...

I'm connecting on the onResume method... any advice on how to accomplish this task?  (I have different part of the app (activities) wich needs to comunicate to the device via BLE)

Thanks

ardmn, 2018-04-29:
@LinkOut  can u show stackTrace of error when ypu try "connect back " and whant does "Main wont connect.." means ? Can u show RxAndroidBle logs ?

LinkOut, 2018-05-02:
Thanks for the answer, i've fixed this issue by abstracting the communication, so every activity can access one instance of the comunication layer.

Thanks

closed 2018-05-02 by LinkOut