dariuszseweryn/RxAndroidBle #423: MissingBackpressureException when scan

ardmn, 2018-04-28:
### Summary
Sometimes I have MissingBackpressureException when scaning devices on sgs 7 (Android 6)

### Library version
RxAndroidBle `1.4.3`
rxJava `1.3.5`

#### Steps to reproduce actual result
 Start device scaning

#### Minimum code snippet reproducing the issue
```java

rxBleClient.scanBleDevices(
                new ScanSettings.Builder()
                        .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                        .build(),
                new ScanFilter.Builder().setServiceUuid(new ParcelUuid(MY_UUID)).build()
        ).observeOn(schedulers.mainThread())
                .onBackpressureDrop()
                .sample(2, TimeUnit.SECONDS)
                .onBackpressureDrop()
                .subscribe(
                        rxBleScanResult -> {
                            Log.d(TAG, rxBleScanResult.toString());

                        },
                        throwable -> {
                            // Handle an error here.
                        }
                );

```

#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`
```
rx.exceptions.MissingBackpressureException:
      at rx.internal.util.RxRingBuffer.rx.internal.util.RxRingBuffer getSpscInstance()(Unknown Source) 
                                       void onNext(java.lang.Object)
      at rx.internal.operators.OperatorMerge$MergeSubscriber.void removeInner(rx.internal.operators.OperatorMerge$InnerSubscriber)(Unknown Source) 
                                                             void queueScalar(rx.internal.operators.OperatorMerge$InnerSubscriber,java.lang.Object)
                                                             void requestMore(long)
                                                             void tryEmit(java.lang.Object)
      at rx.internal.operators.OperatorMerge$MergeSubscriber.void onNext(rx.Observable)(Unknown Source)
                                                             void onError(java.lang.Throwable)
                                                             void addInner(rx.internal.operators.OperatorMerge$InnerSubscriber)
                                                             void tryEmit(rx.internal.operators.OperatorMerge$InnerSubscriber,java.lang.Object)
                                                             void emitScalar(rx.internal.operators.OperatorMerge$InnerSubscriber,java.lang.Object,long)
                                                             void emitScalar(java.lang.Object,long)
      at rx.internal.operators.OperatorMerge$InnerSubscriber.void onNext(java.lang.Object)(Unknown Source)
      at rx.internal.operators.OnSubscribeMap$MapSubscriber.void onNext(java.lang.Object)(Unknown Source)
      at rx.internal.operators.OperatorUnsubscribeOn$1.void onNext(java.lang.Object)(Unknown Source)
      at rx.internal.operators.OnSubscribeCreate$NoneEmitter.void onNext(java.lang.Object)(Unknown Source)
      at rx.internal.util.ObserverSubscriber.void onNext(java.lang.Object)(Unknown Source)
      at rx.observers.SafeSubscriber.void onNext(java.lang.Object)(Unknown Source)
      at rx.internal.operators.OperatorUnsubscribeOn$1.void onNext(java.lang.Object)(Unknown Source)
      at rx.internal.operators.OperatorSubscribeOn$SubscribeOnSubscriber.void onNext(java.lang.Object)(Unknown Source)
      at rx.internal.operators.OnSubscribeCreate$NoneEmitter.void onNext(java.lang.Object)(Unknown Source)
      at com.polidea.rxandroidble.internal.operations.ScanOperationApi21$1.void onScanResult(int,android.bluetooth.le.ScanResult)(Unknown Source)
      at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:413)
      at android.os.Handler.handleCallback(Handler.java:739)
      at android.os.Handler.dispatchMessage(Handler.java:95)
      at android.os.Looper.loop(Looper.java:158)
      at android.app.ActivityThread.main(ActivityThread.java:7229)
      at java.lang.reflect.Method.invoke(Native Method)
      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)

```

ardmn, 2018-05-10:
Have any idei to solve this ?

dariuszseweryn, 2018-05-10:
Hi,
I did not yet had time to dig into this. What if you would put `.observeOn(schedulers.mainThread())` just before `.subscribe()`?

ardmn, 2018-05-10:
@dariuszseweryn Ok, thanks, I check this.

ardmn, 2018-05-18:
@dariuszseweryn I have find reproduce scenario and I will give you feedback today or tomorrow

ardmn, 2018-05-18:
My steps to reproduce on samsung galaxy s 7:
1) Restart smatphon 
2) Open application and start for scaning with SCAN_MODE_LOW_LATENCY and UUID filter

@dariuszseweryn  Your advice has help me. In first testing scenario I put .observeOn(schedulers.mainThread()) just before .subscribe() . - it works fine :) without MissingBackpressureException. Then I had tested more variants and I tink that .onBackpressureDrop() must be before observeOn and may be after scanBleDevices Observable.

Full error log without proguard shrinking:
```java
rx.exceptions.MissingBackpressureException
    at rx.internal.util.RxRingBuffer.onNext(RxRingBuffer.java:325)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.queueScalar(OperatorMerge.java:379)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.tryEmit(OperatorMerge.java:361)
    at rx.internal.operators.OperatorMerge$InnerSubscriber.onNext(OperatorMerge.java:846)
    at rx.internal.operators.OnSubscribeMap$MapSubscriber.onNext(OnSubscribeMap.java:77)
    at rx.internal.operators.OperatorUnsubscribeOn$1.onNext(OperatorUnsubscribeOn.java:52)
    at rx.internal.operators.OnSubscribeCreate$NoneEmitter.onNext(OnSubscribeCreate.java:173)
    at rx.internal.util.ObserverSubscriber.onNext(ObserverSubscriber.java:34)
    at rx.observers.SafeSubscriber.onNext(SafeSubscriber.java:134)
    at rx.internal.operators.OperatorUnsubscribeOn$1.onNext(OperatorUnsubscribeOn.java:52)
    at rx.internal.operators.OperatorSubscribeOn$SubscribeOnSubscriber.onNext(OperatorSubscribeOn.java:74)
    at rx.internal.operators.OnSubscribeCreate$NoneEmitter.onNext(OnSubscribeCreate.java:173)
    at com.polidea.rxandroidble.internal.operations.ScanOperationApi21$1.onScanResult(ScanOperationApi21.java:60)
    at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:413)
    at android.os.Handler.handleCallback(Handler.java:739)
    at android.os.Handler.dispatchMessage(Handler.java:95)
    at android.os.Looper.loop(Looper.java:158)
    at android.app.ActivityThread.main(ActivityThread.java:7229)
    at java.lang.reflect.Method.invoke(Native Method)
    at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
```

But I have UI lags with device scaning on UI thread... :)

dariuszseweryn, 2018-05-18:
So this is the correct behaviour. One must put `.onBackpressure*()` and `.observeOn()` in correct order. Basically `MissingBackpressureException`. See [`.observeOn()` Javadoc](http://reactivex.io/RxJava/1.x/javadoc/rx/Observable.html#observeOn-rx.Scheduler-):

> This operator honors backpressure from downstream and expects it from the source Observable. Violating this expectation will lead to MissingBackpressureException. This is the most common operator where the exception pops up; look for sources up the chain that don't support backpressure, such as interval, timer, {code PublishSubject} or BehaviorSubject and apply any of the onBackpressureXXX operators before applying observeOn itself.

This is not the problem with the library but the usage of `RxJava`. Just rethink what you want and what you need to do and then dispatch the work accordingly among available `Scheduler`s.

Best Regards

closed 2018-05-18 by dariuszseweryn