dariuszseweryn/RxAndroidBle #424: No ScanResult for devices previously discovered

akram-aq1, 2018-05-02:
Running 1.5.0 on RXJava2 here and I'm running into a peculiar issue where some ESP32-based peripherals are no longer found by my app during scanning.

Phone is running Android 7, but I have the same problems with one running 8.1 (first is a Xiaomi Redmi Note 4, second is a Nexus 6P).

For context, the app collects a list of these peripherals and then allows a user to connect via GATT to a custom service we defined that has a variety (<10) of characteristics (both read-only and read/write.

At some point of using the app during testing, perhaps after a write of some kind, I have discovered that I can no longer find these devices via an unfiltered scan (e.g.: ScanSettings.Builder().build()).

Strangely enough, I can read/write from the various characteristics of devices that do not provide scan results any longer. This all works just fine.

Despite my app failing to observe any of these devices, the nRFConnect app finds the devices quite happily, and can connect to the service we defined also.

Is this some sort of android caching magick? I've had a glance around the Internet but apart from various nebulous complaints of scanning issues, I can't see anything obvious.

akram-aq1, 2018-05-02:
Thanks for the feedback, jreehuis.

1. Very familiar with this 'feature': it's not to blame, I can power down the phone leave it for a few hours and bring it back up, and it won't see the device. I can use a different device, and the same problem. I start to believe that it is actually the ESP32 or some quirk of how I'm using the scanning subsystem, since a phone that NEVER saw the device can't see it via my app AFTER the ESP peripherals get in this state. Does not explain why nrfConnect sees them, though!

2. These scans are for less than 30s at a time and are manually started/stopped, so again, I do not think this is related.

I think I'm going to have to make a couple test apps that use the native scanner to see if that works in place of rxandroidble.

akram-aq1, 2018-05-03:
Turns out this was a bug in our code, have proven things are OK via the sample app. Sorry for the misreport!

closed 2018-05-03 by akram-aq1

freecats, 2018-09-19:
> Turns out this was a bug in our code, have proven things are OK via the sample app. Sorry for the misreport!

Hi,  @akram-aq1 
Can you share detail of this bug and solution, we are running into this trap too.
Thanks in advance。

akram-aq1, 2018-09-22:
Hola @freecats,

The problem I was having was not the undocumented Google feature - it was simply that I had naively cached a set of ScanResult records and my code was ignoring discovered devices in that cache (oops).

As for 'fixing' the 'feature': unfortunately I don't know of a true fix. Right now we wait for the problem to occur and warn the user that they can't scan until time X in the future. 

Cheers,

Akram

freecats, 2018-09-22:
> Hola @freecats,
>
> The problem I was having was not the undocumented Google feature - it was simply that I had naively cached a set of ScanResult records and my code was ignoring discovered devices in [...]

Thanks  for sharing!