dariuszseweryn/RxAndroidBle #434: LongWriteOperation attempts to write a 0-byte value at the end

braddodson, 2018-05-24:
### Summary
Long write operations split a write into a series of smaller packet writes. In the RxAndroidBle2 code, the behavior of this changed and now a zero byte value is written at the end of the series. This causes an issue on our device and causes the next characteristic write operation to fail.

I was able to fix the issue by changing the takeUntil on this line:
https://github.com/Polidea/RxAndroidBle/blob/eeb4664e7821c08312d05ed70fe5ce677b5418d1/rxandroidble/src/main/java/com/polidea/rxandroidble2/internal/operations/CharacteristicLongWriteOperation.java#L204
into a takeWhile and flipping the predicate. This matches the logic in the 1.x source for RxAndroidBle.

I can submit a pull request for that if desired.
It might be possible to work around the issue using a writeOperationAckStrategy, although I haven't tried.

### Library version
1.5.0

#### Minimum code snippet reproducing the issue

connection.createNewLongWriteBuilder()
                  .setCharacteristic(characteristic)
                  .setBytes(request.toByteArray())
                  .setMaxBatchSize(
                      BluetoothConstants.DEFAULT_MTU - BluetoothConstants.MTU_RESERVED_BYTES)
                  .build()
                  .ignoreElements()
                  .blockingAwait();

dariuszseweryn, 2018-05-25:
Hello @braddodson 

Thanks for the report. As far as I see we should have this use-case covered in [the tests](https://github.com/Polidea/RxAndroidBle/blob/master/rxandroidble/src/test/groovy/com/polidea/rxandroidble2/internal/operations/OperationCharacteristicLongWriteTest.groovy) unless there is a bug in there as well.

Feel free to check it out and provide a failing test case that could be fixed

uKL, 2018-06-04:
Closing due to inactivity. Feel free to reopen.

closed 2018-06-04 by uKL