dariuszseweryn/RxAndroidBle #436: java.lang.IllegalArgumentException: Receiver not registered: com.polidea.rxandroidble2.internal.util.LocationServicesOkObservableApi23

thuytrinh, 2018-05-29:
### Summary

```
Android: 6.0
Manufacturer: HTC
Model: HTC One_M8

java.lang.IllegalArgumentException: Receiver not registered: com.polidea.rxandroidble2.internal.util.LocationServicesOkObservableApi23$1@2defaaf
	at android.app.LoadedApk.forgetReceiverDispatcher(LoadedApk.java:793)
	at android.app.ContextImpl.unregisterReceiver(ContextImpl.java:1203)
	at android.content.ContextWrapper.unregisterReceiver(ContextWrapper.java:576)
	at com.polidea.rxandroidble2.internal.util.LocationServicesOkObservableApi23$2.run(LocationServicesOkObservableApi23.java:49)
	at io.reactivex.disposables.ActionDisposable.onDisposed(ActionDisposable.java:30)
	at io.reactivex.disposables.ActionDisposable.onDisposed(ActionDisposable.java:19)
	at io.reactivex.disposables.ReferenceDisposable.dispose(ReferenceDisposable.java:43)
	at io.reactivex.internal.observers.BasicFuseableObserver.dispose(BasicFuseableObserver.java:152)
	at io.reactivex.internal.disposables.DisposableHelper.setOnce(DisposableHelper.java:81)
	at io.reactivex.internal.operators.observable.ObservableSwitchMap$SwitchMapInnerObserver.onSubscribe(ObservableSwitchMap.java:346)
	at io.reactivex.internal.observers.BasicFuseableObserver.onSubscribe(BasicFuseableObserver.java:66)
	at com.polidea.rxandroidble2.internal.util.LocationServicesOkObservableApi23.subscribeActual(LocationServicesOkObservableApi23.java:46)
	at io.reactivex.Observable.subscribe(Observable.java:12025)
	at io.reactivex.internal.operators.observable.ObservableMap.subscribeActual(ObservableMap.java:33)
	at io.reactivex.Observable.subscribe(Observable.java:12025)
	at io.reactivex.internal.operators.observable.ObservableSwitchMap$SwitchMapObserver.onNext(ObservableSwitchMap.java:127)
	at io.reactivex.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver.drain(ObservableConcatMap.java:464)
	at io.reactivex.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver.onSubscribe(ObservableConcatMap.java:324)
	at io.reactivex.internal.operators.observable.ObservableFromArray.subscribeActual(ObservableFromArray.java:30)
	at io.reactivex.Observable.subscribe(Observable.java:12025)
	at io.reactivex.internal.operators.observable.ObservableConcatMap.subscribeActual(ObservableConcatMap.java:54)
	at io.reactivex.Observable.subscribe(Observable.java:12025)
	at io.reactivex.internal.operators.observable.ObservableSwitchMap.subscribeActual(ObservableSwitchMap.java:51)
	at io.reactivex.Observable.subscribe(Observable.java:12025)
	at io.reactivex.internal.operators.observable.ObservableSkip.subscribeActual(ObservableSkip.java:29)
	at io.reactivex.Observable.subscribe(Observable.java:12025)
	at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.subscribeInner(ObservableFlatMap.java:165)
	at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.onNext(ObservableFlatMap.java:139)
	at io.reactivex.internal.observers.DeferredScalarDisposable.complete(DeferredScalarDisposable.java:82)
	at io.reactivex.internal.operators.single.SingleToObservable$SingleToObservableObserver.onSuccess(SingleToObservable.java:73)
	at io.reactivex.internal.operators.single.SingleMap$MapSingleObserver.onSuccess(SingleMap.java:64)
	at io.reactivex.internal.operators.observable.ObservableCountSingle$CountObserver.onComplete(ObservableCountSingle.java:83)
	at io.reactivex.internal.operators.observable.ObservableTakeWhile$TakeWhileObserver.onNext(ObservableTakeWhile.java:86)
	at io.reactivex.internal.operators.observable.ObservableInterval$IntervalObserver.run(ObservableInterval.java:83)
	at io.reactivex.internal.schedulers.ScheduledDirectPeriodicTask.run(ScheduledDirectPeriodicTask.java:38)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:423)
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:278)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:154)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:270)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
	at java.lang.Thread.run(Thread.java:818)
```

### Library version
`1.6.0-SNAPSHOT`

#### Actual result

It threw `IllegalArgumentException` and thus crashed the app.

#### Expected result

It should not throw `IllegalArgumentException`.

dariuszseweryn, 2018-05-29:
Hello,

Could you add a code snippet and / or reproduction scenario?

thuytrinh, 2018-05-29:
@dariuszseweryn Sure. I'm trying to get more information on our QA side.

dariuszseweryn, 2018-05-29:
Additionally when from this `1.6.0-SNAPSHOT` version was? 4 days ago we have fixed a similar crash in commit d3197ebf15044d8369763c385bea982253fcb0ff

The code that is using this observable can also be useful if this has something to do with threading.

thuytrinh, 2018-05-30:
@DariuszAniszewski I think https://github.com/Polidea/RxAndroidBle/commit/d3197ebf15044d8369763c385bea982253fcb0ff has fixed this crash as well. Is there any way to determine which snapshot version is being used? Apparently all snapshot versions of `1.6.0` have the exact same `1.6.0-SNAPSHOT`.

dariuszseweryn, 2018-05-30:
It's @dariuszseweryn ;)
Hmmm... Apparently (if it is a clean build with no caches) then it is possible to determine from the timestamp. Anyway I have just released `1.6.0` so there will be no more confusion about `1.6.0-SNAPSHOT` ;)
Closing in this situation

closed 2018-05-30 by dariuszseweryn