dariuszseweryn/RxAndroidBle #439: Lost connection on android upper than Lollipop

jumpstreet13, 2018-06-08:
I am trying to connect to remote ble device, read some information and write some information and all works fine on lollipop devices, but on upper versions device is being disconnected with error code 8 (remote device is out of range) and i can not access ble services, but my phone and remote device is on the same table (i tried it with Samsung galaxy s6, Meizu Pro 6, Lg Nexus 5x)

dariuszseweryn, 2018-06-08:
Hello,

Why do you think it is an issue with the library?

jumpstreet13, 2018-06-08:
Hello, 
Honestly, i can not identify, is it issue with library, or with android sdk, or with remote ble device. Maybe in newer versions of android they changed package exchanging timeout and this is the cause of the problem. Just wanted ask if you had something similar in your experience ?

dariuszseweryn, 2018-06-08:
Never had this kind of an issue. It seems to be a problem with implementation on the Peripheral side. It may be related to using [`BluetoothDevice.connectGatt(context, autoConnect, bluetoothGattCallback, TRANSPORT_LE`)](https://github.com/Polidea/RxAndroidBle/blob/master/rxandroidble/src/main/java/com/polidea/rxandroidble2/internal/util/BleConnectionCompat.java#L93)

jumpstreet13, 2018-06-08:
Thank you so much for advice, i will research :)

dariuszseweryn, 2018-06-08:
You're welcome. Feel free to report back if you will find something. I am closing for now.

Best Regards

closed 2018-06-08 by dariuszseweryn