dariuszseweryn/RxAndroidBle #443: DeadSystemException  Add `onError` handling.

ardmn, 2018-06-14:
Hello I got interesting crash  report :
``java.lang.IllegalStateException: Exception thrown on Scheduler.Worker thread. Add `onError` handling.``
https://developer.android.com/reference/kotlin/android/os/DeadSystemException

I inspected my code and didn't find subscricption without error handling. Can it happends from library ? Thanks. 

### Library version
`1.4.3`

#### Steps to reproduce actual result
Don't know

#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`

``
java.lang.IllegalStateException: Exception thrown on Scheduler.Worker thread. Add `onError` handling.
    at rx.internal.schedulers.ScheduledAction.run(Unknown Source)
    at rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker.run(Unknown Source)
    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
    at java.lang.Thread.run(Thread.java:762)
Caused by: e.b.f: android.os.DeadSystemException
    at rx.internal.util.InternalObservableUtils$ErrorNotImplementedAction.call(Unknown Source)
                                                                       call
    at rx.internal.util.InternalObservableUtils$ErrorNotImplementedAction.call(Unknown Source)
                                                                       call
    at rx.internal.util.ActionSubscriber.onError(Unknown Source)
    at rx.observers.SafeSubscriber._onError(Unknown Source)
    at rx.observers.SafeSubscriber.onError(Unknown Source)
    at rx.internal.operators.NotificationLite.next(Unknown Source)
                                           completed
                                           error
                                           accept
    at rx.internal.operators.OperatorReplay$UnboundedReplayBuffer.next(Unknown Source)
                                                               error
                                                               replay
    at rx.internal.operators.OperatorReplay$InnerProducer.request(Unknown Source)
                                                       index
    at rx.Subscriber.add(Unknown Source)
                  request
                  setProducer
    at rx.Subscriber.add(Unknown Source)
                  request
                  setProducer
    at rx.internal.operators.OperatorReplay$7.call(Unknown Source)
                                           call
    at rx.internal.operators.OperatorReplay$7.call(Unknown Source)
                                           call
    at rx.Observable.create(Unknown Source)
                  unsafeCreate
                  lift
                  compose
                  combineLatest
                  combineLatest
                  concat
                  concat
                  defer
                  from
                  from
                  fromCallable
                  interval
                  just
                  merge
                  merge
                  timer
                  buffer
                  collect
                  concatMap
                  delay
                  doOnCompleted
[... 594 lines cut ...]
                  observeOn
                  observeOn
                  observeOn
                  reduce
                  subscribe
                  subscribe
                  unsafeSubscribe
                  subscribe
                  subscribeOn
                  timeout
                  window
    at rx.internal.operators.OperatorSubscribeOn$SubscribeOnSubscriber.onError(Unknown Source)
                                                                    call
                                                                    setProducer
    at rx.android.schedulers.LooperScheduler$ScheduledAction.run(Unknown Source)
    at android.os.Handler.handleCallback(Handler.java:751)
    at android.os.Handler.dispatchMessage(Handler.java:95)
    at android.os.Looper.loop(Looper.java:154)
    at android.app.ActivityThread.main(ActivityThread.java:6692)
    at java.lang.reflect.Method.invoke(Native Method)
    at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1468)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1358)
Caused by: android.os.DeadSystemException
    ... 84 more
``

dariuszseweryn, 2018-06-15:
Hello @ardmn 

I have looked thought the `master-rxjava1` branch for every `.subscribe()` statements. Most of them are only passing the `onError()` to the observers/emitters. Otherwise they either have `onError()` handling or are subscribed to `Observable`'s that are not able to emit errors.

It seems very unlikely that this is caused by the library.

dariuszseweryn, 2018-06-15:
Btw. the newest `RxJava1` library version is `1.5.0`

dariuszseweryn, 2019-02-15:
Revisiting — I still have no idea what has happened. I have no scenario for reproduction and I do not see other reports about this issue. It seems to be a marginal problem and I do not have time to investigate it further. Closing for now.

Feel free to add more info when available or when other reports will be available

closed 2019-02-15 by dariuszseweryn