dariuszseweryn/RxAndroidBle #448: Comment on Wiki "FAQ: Cannot connect" item

RobLewis, 2018-06-21:
There doesn't seem to be a way to comment directly on the [Wiki](https://github.com/Polidea/RxAndroidBle/wiki/FAQ:-Cannot-connect) so I'm noting this here: 
At least [one developer](https://github.com/prasanthmurugan) reported [here](https://github.com/googlesamples/android-BluetoothLeGatt/issues/44) that simply stopping scanning before connecting wasn't enough to connect successfully. Their solution was to insert a 500 ms delay after stopping the scan. (The linked item includes other things people have tried, some successfully. As is often the case it seems to be highly device-dependent.)

uKL, 2018-06-22:
This is correct, in my case 250ms was enough for all devices that I had tested on. Worth noting, thanks!

dariuszseweryn, 2018-06-25:
I have added a [`Connect right after scanning`](https://github.com/Polidea/RxAndroidBle/wiki/FAQ:-Cannot-connect#connect-right-after-scanning) paragraph

closed 2018-06-27 by dariuszseweryn