dariuszseweryn/RxAndroidBle #450: I created a simple example of the use of RxBleCustomOperation

RobLewis, 2018-06-26:
Recently I had a need to retrieve the Bluetooth GATT object from Android. Since `RxAndroidBle` doesn't offer this as a standard operation, I tried implementing it as a Custom Operation. 

For less-advanced programmers like me, `RxBleCustomOperation` isn't the easiest thing to understand. But I think I've got it, and I created [a Gist](https://gist.github.com/RobLewis/fc20bde8ce71c7da2040d04b2ba3c156) to explain how it works. Hope it may benefit someone else. You might consider adding it to the Wiki.

Lloyd0577, 2018-09-16:
hello!,how to receive notify data by RxBleCustomOperation

dariuszseweryn, 2019-02-16:
I have finally added [a Wiki page](https://github.com/Polidea/RxAndroidBle/wiki/RxBleCustomOperation) for this. Thank you @RobLewis !

closed 2019-02-16 by dariuszseweryn